SpenceKonde/megaTinyCore #798: Attiny214 DAC stuck on 0.55V reference

dr-rockzo, 2022-09-12:
As the title says, i can't seem to change the DAC reference from internal 0.55V. I tried all the options except for the external reference and the full scale output remains 0.548V as measured.

I'm away from the computer currently, but will post my test code in the morning, just incase it's a me problem.

dr-rockzo, 2022-09-13:
`#define debug_led PIN_PA4
#define dac_out PIN_PA6

void setup() {
  pinMode(debug_led, OUTPUT);
  pinMode(dac_out, OUTPUT);
  analogReference(INTERNAL2V5);

}

void loop() {
  heartbeat();
  analogWrite(dac_out,128);
}

void heartbeat(){
  static unsigned long lastMillis;
  if(millis()-lastMillis > 500){
      digitalWrite(debug_led, !digitalRead(debug_led));
      lastMillis = millis();
  }    
}`

As promised, the test code. I expected to see 1.25V and the DVM shows 0.275V.

jvasileff, 2022-09-13:
Try this instead, which should work:

```cpp
    DACReference(INTERNAL2V5);
```

dr-rockzo, 2022-09-13:
*facepalms*
Thank you! I figured it was a me problem.

closed 2022-09-13 by dr-rockzo

SpenceKonde, 2022-09-13:
Yeah, the modern parts let you independently set the reference voltages for each peripheral. on the full-size part (16k and 32k 1-series parts got a bunch of bonus features - I think because they wanted to test them under real world conditions before releasing them on the DX-series), the t1614, (and the rest of 16k/32k 1-series) can independently set the voltage reference for a whopping 5 peripherals! 
ADC0 (the one the core uses)
DAC0 (the only one that can output a voltage)
* _ADC1 (not used by the core unless you enable it, see the analog reference_
* _DAC1 (DACREF for AC1, no output pin)_
* _DAC2 (DACREF for AC2, no output pin)_

May be a record for the AVR architecture (italic items above aren't present on parts with less than 1614k of flash. 

Also, Using a 214? You are going to LOVE the 2.6.0 update, which I can finally post (just have to wait for phone to charge enough to receive the 2FA for the file host). It drops the flash footprint of Serial by hundreds of bytes even though I added features to it.

dr-rockzo, 2022-09-13:
Thanks for the info and good news! 

I was planning to use a  414 or 814, but supply shortages and what not.

SpenceKonde, 2022-09-14:
Note that the 214/414/814 (as well as the 416/816/417/817) are kind of a screwing though none is quite as much of a screwing as the 204 vs the 404 - they are priced the same from the manufacturer!  The reason I say it's a screwing is that well, where is the biggest price increment for the 1-series? The same place as it is on the 0-series and 2-series: when you move from 4k to 8k. 

The premium on a 1-series 16k part over an 8k one is only a penny higher than for a 2-series 16k over the 8k 2-series. 

This in spite of the fact that the  814 has 1 ADC, 1 AC, no external vref, and 512b of ram, while the 1614 has 2 ADCs 3 ACs external vref and 2k of RAM. This is a very rare example of AVRs where more than just the  memory size changes when you step up or down the flash size ladder within a family. I can't think of any other examples. 
Memory sizes (expressed as flash kb/sram bytes) 
* 0-series 2/128, 4/256, 8/512, 16/1024
* 1-series 2/128, 4/256, 8/512, 16/2048, 32/2048
* 2-series No 2k, 4/512, 8/1024, 16/2048, 32/3072
So why da fuk does the price jump when going from 4k to 8k instead of 8k to 16k?

There are no differences in specs of 0-series or 2-series parts other than memory; anything a 1604 can do, if the code can be made small enough (in terms of both flash and ram), can be done with a 204 (though why you'd bother shrinking code size once it fit in 4k is a mystery, certainly not for the price incentive). Same with the 2-series. 

MFG prices are:
| 0-series | 1-series | 2-series |
|----------|----------|----------|
|   0.61     |   0.63     |    n/a     | 
|   0.61     |   0.65     |   0.68    | 
|   0.73     |   0.77     |   0.80    |
|   0.77     |    0.86    |   0.88    |

The pricing curves for all three series seems to have been set as a function of the flash size; it's as if the guy deciding the prices knew nothing about the feature set other than that the 2-series is on balance better than the 1-series (I put it like since the 1 series has some features that were dropped in the 2-series, while the 2-series has 2x the CCLs, the event system isn't batshit crazy, and the ADC blows the doors off every modern AVR (it's the one we'll get inthe EA series on full size chips). Considering the sizable number of features added, you'd expect a significantly larger than average premium. We do not see that. Odd isn't it?