SpenceKonde/megaTinyCore #800: 2.6 extra "avr" in toolchain for Mac

ObviousInRetrospect, 2022-09-14:
```
fork/exec /Users/user/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino5/bin/avr-g++: no such file or directory
Error compiling for board ATtiny412/402/212/202.
```

looks like there is an extra avr in the path:
/Users/user/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino5/bin doesn't exist but /Users/user/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino5/avr/bin does.

/Users/user/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino5/ contains only "avr"

Since /Users/user/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino5/avr/avr exists I am assuming an extra avr got put in the tar

SpenceKonde, 2022-09-14:
Fixed in Azduino6. There are no scripts to do the toolchain work. I have the official arduino build script, and just add references to the atpacks and added a line to wget them, and I have a build system in AWS EC2 that I deploy, compile with, and then snag the result from and shut down (it's an xlarge, so it's not the sort of thing you leave up when you don't need it - but I don't have a machine with the right OS and all the dependancies installed, and renting one for an hour every few months to compile the toolchain with is cheap). That gets me only an x64-linux one though. 

But it turns out that nothing is changing in the compiler itself - it's just getting includes and device-spec files added and changed. That doesn't change the executable binaries, but it does change the precompiled .a and .o files that the toolchain packasge includes. And that's the only script I know that generates those. I don't know how to make things cross compile for different platforms, and when investigated, a big scary monster with far too many tentacles, mouths, and eyes roared at me. That's a fight I didn't want to start, so I looked for an alternate path.

All the Arduino7 toolchain packages are extracted (on linux, unless it's for windows), as is the new Azduino build. From the Azduino build I then delete /bin. /avr/bin, /libexec, /lib/*.* (but not directories, except for /lib/gcc/avr/7.3.0/plugins, /(arch), and /avr/bin (yes, there's /bin and /avr/bin. No they don't contain the same files. No, I don't know why, but if you find out let me know), and then copy it into each of the other architectures, replacing duplicate files. Then I recompress them. There are text files that I copy and paste commands out of, and one that I copy and paste regex'es out of to generate the json. but it is not scripted, and never goes smoothly.

closed 2022-09-14 by SpenceKonde

jvasileff, 2022-09-15:
Sorry if this is a dumb question (I'm fairly new to this space), but:

Any reason to not just start with the latest toolchains from Microchip ([12 May 2022](https://www.microchip.com/en-us/tools-resources/develop/microchip-studio/gcc-compilers)) and copy in the includes and precompiled .a and .o files from the [packs](http://packs.download.atmel.com)? I tried doing this as a quick test on a 3226, and it seemed to work fine.

For reference, the version from the Microchip download:

```
 % /tmp/avrx/avr8-gnu-toolchain-darwin_x86_64/bin/avr-g++ --version
avr-g++ (AVR_8_bit_GNU_Toolchain_3.7.0_518) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

and from azduino6:
```
% ~/lib/avr-azduino6/bin/avr-g++ --version
avr-g++ (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

SpenceKonde, 2022-09-15:
Because I don't think all the files that are needed are in the packs..... maybe they are and I don't know where. I'm not fully sure what the required list of files actually is. This is all voodoo being done with incomplete understanding. I don't really like farting around weith the toolchain I'm largely doing it because I have found a way that works, albeit with painful manual steps (which would be even more painful there because they're organized differently in the packs :-/ )

jvasileff, 2022-09-15:
> I'm not fully sure what the required list of files actually is

Yeah, to be fair, I relied on your azduino as a reference, along with compiler errors to help determine what was missing.

I did see a script called `xfer-pack` [here](https://www.avrfreaks.net/forum/building-gcc83-avr-libc-linux-w-atpack-hooks) that I think matches what I did. If I get some time, I may try to put up a GitHub repo that uses GitHub actions to assemble clean toolchains - which may or may not be useful to you down the road.