SpenceKonde/megaTinyCore #801: 2.6 comparator library

ObviousInRetrospect, 2022-09-14:
Got this on a sketch that included had a gratuitous #include <Comparator.h>
```
In file included from /Users/user/Documents/Arduino/soilmoist3224b/soilmoist3224b.ino:19:0:
/Users/user/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.0/libraries/Comparator/src/Comparator.h:323:5: error: 'in_comparator' does not name a type; did you mean 'Comparator'?
     in_comparator::p::inputP_t     input_p        = in_comparator::p::in0;
     ^~~~~~~~~~~~~
     Comparator
/Users/user/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.0/libraries/Comparator/src/Comparator.h:324:5: error: 'in_comparator' does not name a type; did you mean 'Comparator'?
     in_comparator::n::inputN_t     input_n        = in_comparator::n::in0;
     ^~~~~~~~~~~~~
     Comparator

exit status 1
Error compiling for board ATtiny3224/1624/824/424 w/Optiboot.
```

when I commented that out I got the USART error again (on a sketch that does use Serial)

ObviousInRetrospect, 2022-09-15:
I think #809 fixes this

SpenceKonde, 2022-09-16:
Thannnnkk youuuuu

closed 2022-09-16 by SpenceKonde