SpenceKonde/megaTinyCore #805: downloading tool DxCore:avr-gcc@7.3.0-atmel3.6.1-azduino5 gives 404

ArminJo, 2022-09-15:
[Here ](https://github.com/Arduino-IRremote/Arduino-IRremote/actions/runs/3057068764/jobs/4931850358)I get the following error:
Error during install: Error downloading tool DxCore:avr-gcc@7.3.0-atmel3.6.1-azduino5: Server responded with: 404 Not Found

ObviousInRetrospect, 2022-09-15:
This was done intentionally to prevent installation: https://github.com/SpenceKonde/megaTinyCore/discussions/795#discussioncomment-3644565

ObviousInRetrospect, 2022-09-15:
you might want to pin to 2.5.11 for now as a workaround

SpenceKonde, 2022-09-15:
Yes, the archive was mispackaged. Until 2.6.1 is released, please continue to use 2.5.11. 

You probably don't want a version of the core that can't compile anything, so you're probably not looking for 2.6.0 - which in the end doesnt matter much because I had forgotten to commit a change that fixed uploads (there had been a disastrous diff merge that copied a bunch more stuff that I intended, and I hadn't realized it until after I'd realized 2.6.0 was a lemon anyway.

closed 2022-09-15 by SpenceKonde