SpenceKonde/megaTinyCore #819: possible 2.6(.1) bug?

ObviousInRetrospect, 2022-09-17:
Getting the following when I try to upload with DxCore after installing 2.6.0 via boards manager and manually fixing the azduino5 location:

```
java.io.IOException: Cannot run program "/Users/user/Library/Arduino15/packages/DxCore/hardware/megaavr/1.4.10/tools/python3/python3": error=2, No such file or directory
```

I am not 100% confident that this was caused by the install but before you push the boards manager json could you upload a copy of the new one with a new file name (http://drazzy.com/package_drazzy.com_index_TEST.json), launch a portable arduino, install DXCore and MTC 2.5.11, and then do the upgrade to MTC and verify DXC can still upload with serialupdi?

I'm happy to run that test if you can point me at that unreleased JSON.

SpenceKonde, 2022-09-18:
Hm, - I don't see how it could have had such an effect,.... 

I will try to reproduce (though I'm on windows, so things may behave differently). I am suspicious that this could have happened for some other reason likely related to the battles with 2.6.0. Uninstall and reinstall of DxCore should fix it.

SpenceKonde, 2022-09-21:
Closing because cannot reproduce.

closed 2022-09-21 by SpenceKonde