SpenceKonde/megaTinyCore #839: Maximum Wire BUFFER_LENGTH is 255, NOT 256

specternecter, 2022-09-30:
It took me hours to figure out what was happening, but I can't seem to figure out how to fix it in the library. I have an I2C device that sends 256 byte packets to a slave. I'm using an ATTINY3226 as the slave and I modified the boards.txt file to allow me to select a 256 byte BUFFER_LENGTH. However, Wire.onReceive() never returned any data. So I got out an ESP32 and started sending packets to find out why I wasn't getting anything. It turns out that Wire.onReceive() only works if you send **LESS** than 256 bytes. You can send 3 bytes or 55 bytes, or 255 bytes but you can not send 256 bytes. If you send 256 bytes, Wire.onReceive() seems to overflow somewhere and the function will never run. I suspect the issue is similar to if you were to run the for() loop in my attached code with i set as uint8_t instead of uint16_t. In some cases uint8_t works but in others it can just run on forever because the size of incoming is greater than the max uint8_t value of 255. I really need the ATTINY3226 as the slave due to it's very impressive UART capabilities, but I also really need that final byte. Is there a simple fix to get all 256 bytes?

`#include "Wire.h"

uint8_t incoming[256];

void receiveEvent(int len) {
  for(uint16_t i = 0; i < sizeof(incoming); i++){
    incoming[i] = Wire.read();
    Serial1.print(incoming[i], HEX); Serial1.print(" ");
  }
  Serial1.println();
}

void setup() {
  Wire.begin(0x69);
  Wire.onReceive(receiveEvent); /* register receive event */
  Serial1.begin(921600);
}

void loop() {
  // null
}`

SpenceKonde, 2022-10-01:
I don't think this is inconsistent with our design intent. I'll make sure the to correct the comment, and ideally #error if someone tries to set it to larger than 255. The issue is the same as the one that caused problems back when the new Wire library went in: The first position in the buffer is occupied by the slave address. We initially failed to account for that, and so the 32-byte buffer was only 31. Since almost every piece of arduino code in the wild takes as a given that the wire buffer is 32 bytes, things that used it at the limit broke; that was quickly corrected once the problem was recognized - but trying to go beyond 255 doesn't seem like an appropriate use of development time. 256 byte I2C payloads seems to be more than is warranted by the 

Remember that your payload is 8 times larger than the wire library's normal maximum on the vast majority of Arduino devices - if I'm not mistaken, this would be a BFD to increase,  there are places where we pass values around within the library assuming that a byte can hold the number of bytes of payload - so the penalty is particularly tough here.This seems like a lot of work for a change that basically one person will use.  I recall getting pushback from someone on my position of 130 byte maximum (which I wanted in order to be able to do a full 128 byte page + slave address + 2 byte memory address = 131 (BUFFER_LENGTH = 130) when writing to I2C EEPROMs, since the modern AVRs are a bit thin on EEPROM compared to classics, and since 128b pages are the maximum found in I2C eeproms in the wild (24-series eeproms only support up to 4mbit on one bus, using up to three bits of the I2C address as the MSBs of the memory address, ie, 512kbyte) - and that's only enabled parts with 4k+ of ram (32k Dx and Ex-series parts)

MX682X has done the majority of the work enabling the dramatic improvements to Wire since 2.3.x; unless @MX682X disagrees, I'm inclined to just correct the comment and mark this no-plans-to-change

MX682X, 2022-10-01:
As Spence mentioned already, the arduino ecosystem established a standard of 32 bytes of a standard length. With this in mind and to keep things simple, the code is limited to uint8_t buffer sizes. So, defining a size of 256 equals 0. (The drawbacks of 8-bit MCUs)
Serial solved it with this:
```
#if (SERIAL_TX_BUFFER_SIZE > 256)
  typedef uint16_t tx_buffer_index_t;
#else
  typedef uint8_t  tx_buffer_index_t;
#endif
#if (SERIAL_RX_BUFFER_SIZE > 256)
  typedef uint16_t rx_buffer_index_t;
#else
  typedef uint8_t  rx_buffer_index_t;
#endif
```

but the problem with uint16_t is, at least with ISR shared variables, that they might be changed inside the ISR, so such reads and writes have to be atomic (hopes that Microchip will add a CCP byte to disable ISRs for 2/4 instructions, without doing anything else (Spence, do you know if they take any ideas?)), and it's usually more trouble then it's worth. So, it was basically intentional. 

But now that I think about it, any normal access to the slave buffers happen in an ISR and any accesses to the master buffers happen without an ISR, so technically it should be possible to add support for bigger buffers witthout worrying about atomic accesses.

MX682X, 2022-10-02:
@specternecter please try replacing the Wire library files with this modified library - it should have the needed fixes for buffer lengths >255 bytes.

[src.zip](https://github.com/SpenceKonde/megaTinyCore/files/9692079/src.zip)

specternecter, 2022-10-02:
> @specternecter please try replacing the Wire library files with this modified library - it should have the needed fixes for buffer lengths >255 bytes.
> 
> [src.zip](https://github.com/SpenceKonde/megaTinyCore/files/9692079/src.zip)

It works! I had to change the BUFFER_LENGTH in my custom boards.txt to 257 instead of 256 but I'm now getting the last byte and you have no idea how much I appreciate that. If you leave the boards file at 256, Wire.onReceive() won't freeze anymore but it will return 0xFF when my actual final byte is 0x00. So it doesn't get the final byte if the BUFFER_LENGTH is set to 256 but setting the length to 257 is not a problem that I'm concerned with. Thanks!

MX682X, 2022-10-02:
found the bug:
```
#if (BUFFER_LENGTH > 256)  // should be 255 
  typedef uint16_t twi_buffer_index_t;
#else
  typedef uint8_t  twi_buffer_index_t;
#endif
```
Here is the fixed version, could you test it please?
[src.zip](https://github.com/SpenceKonde/megaTinyCore/files/9692830/src.zip)

specternecter, 2022-10-02:
> found the bug:
> 
> ```
> #if (BUFFER_LENGTH > 256)  // should be 255 
>   typedef uint16_t twi_buffer_index_t;
> #else
>   typedef uint8_t  twi_buffer_index_t;
> #endif
> ```
> 
> Here is the fixed version, could you test it please? (there were actually a couple more things that I missed) [src.zip](https://github.com/SpenceKonde/megaTinyCore/files/9692830/src.zip)

Not a problem. I don't mind checking until it's working correctly if you want. This version still requires the buffer length to be set to 257 though unfortunately.

MX682X, 2022-10-02:
0x(FF)FF usually means that the buffer does not contain any more data. Can you check the amount of data that was received by calling Wire.available() and printing the return value in Wire.onReceive()?

specternecter, 2022-10-02:
Wire.available() = 255

MX682X, 2022-10-02:
with 257 Buffer size or 256?

specternecter, 2022-10-02:
> with 257 Buffer size or 256?

Yeah I'm sorry, I realized a few minutes ago that I should have ran it under both

Wire.available() for 256 byte buffer: 255
Wire.available() for 257 byte buffer: 256

Data being sent from the test device: E9 06 02 2F 62 24 38 40 EE 00 00 00 05 00 00 00 00 00 00 00 00 00 00 01 20 00 04 3C 58 55 01 00 32 54 CD AB 00 00 00 00 00 00 00 00 00 00 00 00 68 65 61 64 73 2F 69 64 66 2D 72 65 6C 65 61 73 65 2F 76 34 2E 34 2D 30 2D 67 64 34 35 37 32 00 61 72 64 75 69 6E 6F 2D 6C 69 62 2D 62 75 69 6C 64 65 72 00 00 00 00 00 00 00 00 00 00 00 00 00 31 38 3A 33 32 3A 35 30 00 00 00 00 00 00 00 00 4A 75 6C 20 20 34 20 32 30 32 32 00 00 00 00 00 76 34 2E 34 2E 31 2D 34 37 32 2D 67 63 39 31 34 30 63 61 66 38 63 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Data I get from 256 byte buffer: E9 06 02 2F 62 24 38 40 EE 00 00 00 05 00 00 00 00 00 00 00 00 00 00 01 20 00 04 3C 58 55 01 00 32 54 CD AB 00 00 00 00 00 00 00 00 00 00 00 00 68 65 61 64 73 2F 69 64 66 2D 72 65 6C 65 61 73 65 2F 76 34 2E 34 2D 30 2D 67 64 34 35 37 32 00 61 72 64 75 69 6E 6F 2D 6C 69 62 2D 62 75 69 6C 64 65 72 00 00 00 00 00 00 00 00 00 00 00 00 00 31 38 3A 33 32 3A 35 30 00 00 00 00 00 00 00 00 4A 75 6C 20 20 34 20 32 30 32 32 00 00 00 00 00 76 34 2E 34 2E 31 2D 34 37 32 2D 67 63 39 31 34 30 63 61 66 38 63 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF

Data I get from 257 byte buffer: E9 06 02 2F 62 24 38 40 EE 00 00 00 05 00 00 00 00 00 00 00 00 00 00 01 20 00 04 3C 58 55 01 00 32 54 CD AB 00 00 00 00 00 00 00 00 00 00 00 00 68 65 61 64 73 2F 69 64 66 2D 72 65 6C 65 61 73 65 2F 76 34 2E 34 2D 30 2D 67 64 34 35 37 32 00 61 72 64 75 69 6E 6F 2D 6C 69 62 2D 62 75 69 6C 64 65 72 00 00 00 00 00 00 00 00 00 00 00 00 00 31 38 3A 33 32 3A 35 30 00 00 00 00 00 00 00 00 4A 75 6C 20 20 34 20 32 30 32 32 00 00 00 00 00 76 34 2E 34 2E 31 2D 34 37 32 2D 67 63 39 31 34 30 63 61 66 38 63 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

So everything is correct, it just somehow drops that last byte when defined as 256

specternecter, 2022-10-02:
I also decided to define the buffer length as 257 and then send it 257 0s because I thought it might help with debugging the issue.  I only got back 256 bytes

specternecter, 2022-10-02:
I set the buffer to 255 and sent it 255 bytes.

Wire.available(): 254

It's dropping the last byte on any defined buffer length if you actually send that length.

MX682X, 2022-10-02:
I think I figured out where it comes from. In twi.c, line 622. If you remove the -1 from BUFFER_LENGTH-1, it should work. I hope....

specternecter, 2022-10-02:
It worked!

SpenceKonde, 2022-10-20:
Fixed now.

closed 2022-10-20 by SpenceKonde