SpenceKonde/megaTinyCore #841: Error: cannot parse fuse, 'invalid literal for int() with base 0: '{bootloader.CODESIZE} on t1604 with Serial2UPDI

pcfreak1201, 2022-10-03:
Arduino: 1.8.19 (Linux), Board: "ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204, ATtiny1604, 10 MHz internal, 2.6V (10 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), 64ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), On all pins, with new implementation."
```
/home/stefan/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1/python3 -u /home/stefan/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.1/tools/prog.py -t uart -u /dev/ttyUSB0 -b 230400 -d attiny1604 --fuses 0:0x00 1:0x00 2:0x00 5:0b11000101 6:0x07 7:{bootloader.CODESIZE} 8:{bootloader.BOOTSIZE} 9:{bootloader.CODESIZE} {bootloader.pymcuprogstring} -v 
Fehler beim Brennen des Bootloaders. (Error burning bootloader.)
Error: cannot parse fuse, 'invalid literal for int() with base 0: '{bootloader.CODESIZE}''
```

pcfreak1201, 2022-10-03:
jtag2updi works fine:
```
/home/stefan/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18/bin/avrdude -C/home/stefan/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.1/avrdude.conf -v -pattiny1604 -cjtag2updi -P/dev/ttyUSB1 -b115200 -e -Ufuse0:w:0x00:m -Ufuse1:w:0x00:m -Ufuse5:w:0b11000101:m -Ufuse6:w:0x07:m -Ufuse8:w:0x00:m -Ufuse7:w:0x00:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/stefan/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.1/avrdude.conf"
         User configuration file is "/home/stefan/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB1
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATtiny1604
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
         S_MCU hardware version: 1
         S_MCU firmware version: 6.00
[... 78 lines cut ...]
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse8 data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of fuse8 verified
avrdude: reading input file "0x00"
avrdude: writing fuse7 (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of fuse7 written
avrdude: verifying fuse7 memory against 0x00:
avrdude: load data fuse7 data from input file 0x00:
avrdude: input file 0x00 contains 1 bytes
avrdude: reading on-chip fuse7 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse7 verified

avrdude done.  Thank you.
```

pcfreak1201, 2022-10-03:
Looks same as https://github.com/SpenceKonde/megaTinyCore/discussions/831

SpenceKonde, 2022-10-04:
Already fixed with #831

closed 2022-10-04 by SpenceKonde