SpenceKonde/megaTinyCore #844: Error: cannot parse fuse - Flashing in Visual Studio Code

sshwarts, 2022-10-11:
I get the "Error: cannot parse fuse, 'invalid literal for int() with base 0: '{bootloader.CODESIZE}'' while trying to upload in Visual Studio Code. I can upload fine in the Arduino IDE but not in Visual Code (the config should be the same).
 
  "configuration": "chip=3216,clock=1internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,attach=allenabled",
    "board": "megaTinyCore:megaavr:atxy6",
    "sketch": "Coaster.ino",
    "output": "./build",
    "port": "/dev/tty.usbserial-DN02WXRC",
    "programmer": "megaTinyCore:serialupdi921k_wd1"
    
    ?? Is this related to whjat you fixed a few days ago??

sshwarts, 2022-10-12:
More details. On the Arduino side the command looks like:
/Users/scotts/Library/Arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1/python3 -u /Users/scotts/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.1/tools/prog.py -t uart -u /dev/cu.usbserial-DN02WXRC -b 921600 -wd 1 -d attiny3216 --fuses 2:0x01 6:0x04 7:0x00 8:0x00 -f/var/folders/44/lkx9rq0j08747wbzjsfh26yw0000gn/T/arduino_build_119818/SMTTINY3216.ino.hex -a write -v 

On the Visual Code is looks like:
/Users/scotts/Library/Arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1/python3 -u /Users/scotts/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.6.1/tools/prog.py -t uart -u /dev/tty.usbserial-DN02WXRC -b 345600 -d attiny3216 --fuses 5:0b11000101 6:0x04 7:{bootloader.CODESIZE} 8:{bootloader.BOOTSIZE} -f/Users/scotts/Arduino/Coaster/build/SMTTINY3216.ino.hex -a write -v 

Clearly, those fuse settings are not being set for some reason.

closed 2022-10-12 by sshwarts

SpenceKonde, 2022-10-20:
> Leaving this for someone else...
> There was some sort of conflict between my Arduino IDE and megaTinyCore
> and ATTinyCore. Once I removed ATTinyCore, it worked fine.

Can you please provide more details? I think it's is an arduino bug. When
the bug is manifesting, clench tools -> programmer. You will notice it us
blank. Re-select the desired programmer and itll wik fine again . Happens
when you switch between cores - it uploads using the previously selected
programmer and platform.txt settings. Guaranteed not to work if you've
changed cores

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***