SpenceKonde/megaTinyCore #852: Sketch upload failed for ATtiny1627 with optiboot_x

mcuee, 2022-11-01:
Error log:

```
"C:\\Users\\xfchen\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-size" -A "C:\\Users\\xfchen\\AppData\\Local\\Temp\\arduino_build_550277/Blink.ino.elf"
Sketch uses 678 bytes (4%) of program storage space. Maximum is 15972 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 2044 bytes for local variables. Maximum is 2048 bytes.
C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/avrdude.conf -v -pattiny1627 -carduino -D -PCOM21 -PCOM21 -b115200 -Uflash:w:C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277/Blink.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/avrdude.conf"

         Using Port                    : COM21
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATtiny1627
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 9.1
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9428 (probably t1627)
avrdude: reading input file "C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277/Blink.ino.hex"
avrdude: writing flash (1190 bytes):

Writing | ################################################## | 100% 0.07s

avrdude: 1190 bytes of flash written
avrdude: verifying flash memory against C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277/Blink.ino.hex:
avrdude: load data flash data from input file C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277/Blink.ino.hex:
avrdude: input file C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277/Blink.ino.hex contains 1190 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.07s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0200
         0xff != 0x3b
avrdude: verification error; content mismatch

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected
```

mcuee, 2022-11-01:
Bootloader initially failed to burn due to the known issues.
```
C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/avrdude.conf -v -pattiny1627 -ccuriosity_updi -Pusb -e -Ufuse0:w:0x00:m -Ufuse1:w:0x00:m -Ufuse5:w:0b1111{bootloader.altreset}10:m -Ufuse6:w:0x04:m -Ufuse8:w:0x00:m -Ufuse7:w:0x00:m -Uflash:w:C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/bootloaders/hex/optiboot_Curiosity1627.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : curiosity_updi
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny1627
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Microchip Curiosity in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.21 (rel. 37)
         Serial number   : MCHP3258021800000382
         Vtarget         : 3.31 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 100 kHz

avrdude: Partial Family_ID returned: "tiny"
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9428 (probably t1627)
avrdude: erasing chip
avrdude: reading input file "0x00"
avrdude: writing fuse0 (1 bytes):
[... 122 lines cut ...]
avrdude: verifying fuse8/bootend memory against 0x00
avrdude: 1 byte of fuse8/bootend verified
avrdude: reading input file 0x00 for fuse7/append
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse7/append ...
avrdude: 1 byte of fuse7/append written
avrdude: verifying fuse7/append memory against 0x00
avrdude: 1 byte of fuse7/append verified
avrdude: reading input file C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/bootloaders/hex/optiboot_Curiosity1627.hex for flash
         with 466 bytes in 2 sections within [0, 0x1ff]
         using 8 pages and 46 pad bytes
avrdude: writing 466 bytes flash ...

Writing | ################################################## | 100% 0.28 s

avrdude: 466 bytes of flash written
avrdude: verifying flash memory against C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/bootloaders/hex/optiboot_Curiosity1627.hex

Reading | ################################################## | 100% 0.14 s

avrdude: 466 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-11-01:
Using "Export compiled Binary" and the exported binary will have the 0x200 offset.

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xfchen\AppData\Local\Arduino15\packages -hardware C:\Users\xfchen\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xfchen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xfchen\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:microchipo:board_name=cn1627,clocko=20internal,bodvoltage=1v8,bodmode=disabled,millis=enabled,resetpin=UPDIaltrst,startuptime=8,wiremode=mors,printf=default,attach=allenabled -vid-pid=03EB_2175 -ide-version=10819 -build-path C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277 -warnings=all -build-cache C:\Users\xfchen\AppData\Local\Temp\arduino_cache_297627 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avr-gcc.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.arduinoOTA.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.python3.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -verbose C:\Users\xfchen\Documents\Arduino\Tiny1627\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xfchen\AppData\Local\Arduino15\packages -hardware C:\Users\xfchen\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xfchen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xfchen\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:microchipo:board_name=cn1627,clocko=20internal,bodvoltage=1v8,bodmode=disabled,millis=enabled,resetpin=UPDIaltrst,startuptime=8,wiremode=mors,printf=default,attach=allenabled -vid-pid=03EB_2175 -ide-version=10819 -build-path C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277 -warnings=all -build-cache C:\Users\xfchen\AppData\Local\Temp\arduino_cache_297627 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avr-gcc.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.arduinoOTA.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.python3.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -verbose C:\Users\xfchen\Documents\Arduino\Tiny1627\Blink\Blink.ino
Using board 'microchipo' from platform in folder: C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1
Using core 'megatinycore' from platform in folder: C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1
Warning: Board DxCore:megaavr:avrddopti doesn't define a 'build.board' preference. Auto-set to: MEGAAVR_AVRDDOPTI
Detecting libraries used...
"C:\\Users\\xfchen\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1627 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERB1 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_CuriosityNano1627 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=1UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -DLED_BUILTIN=PIN_PB4 -DPIN_BUTTON_BUILTIN=PIN_PB7 "-IC:\\Users\\xfchen\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.1\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\xfchen\\AppData\\Local\\Arduino15\\pack [... cut ...]
Global variables use 4 bytes (0%) of dynamic memory, leaving 2044 bytes for local variables. Maximum is 2048 bytes.
cmd /C copy C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277\Blink.ino.lst C:\Users\xfchen\Documents\Arduino\Tiny1627\Blink\Blink.ino.t1627opti.20c0.mB1.v261.lst 
        1 file(s) copied.
cmd /C copy C:\Users\xfchen\AppData\Local\Temp\arduino_build_550277\Blink.ino.map C:\Users\xfchen\Documents\Arduino\Tiny1627\Blink\Blink.ino.t1627opti.20c0.mB1.v261.map 
        1 file(s) copied.

```

mcuee, 2022-11-01:
However, even if I use avrdude to upload the exported hex file, it will still fail.

```
C:\work\avr\avrdude_test\avrdude_bin>
                                      .\avrdude -v -pattiny1627 -carduino -PCOM21 -b115200 -D -U .\Blink.ino.t1627opti.20c0.mB1.v261.hex

avrdude: Version 7.0-20221029 (b569966)
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

avrdude: input file .\Blink.ino.t1627opti.20c0.mB1.v261.hex auto detected as Intel Hex
         Using Port                    : COM21
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATtiny1627
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   32      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      16384   64      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino for bootloader using STK500 v1 protocol
         Hardware Version: 3
         Firmware Version: 9.1

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9428 (probably t1627)
avrdude: reading input file .\Blink.ino.t1627opti.20c0.mB1.v261.hex for flash
         with 678 bytes in 1 section within [0x200, 0x4a5]
         using 11 pages and 26 pad bytes
avrdude: writing 678 bytes flash ...

Writing | ################################################## | 100% 0.10 s

avrdude: 678 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.t1627opti.20c0.mB1.v261.hex

Reading | ################################################## | 100% 0.09 s

avrdude avr_verify() error: verification mismatch, first encountered at addr 0x0200
        device 0xff != input 0x3b
avrdude do_op() error: verification mismatch

avrdude done.  Thank you.

```

mcuee, 2022-11-01:
Using avrdude 6.3 from megaTinyCore, it works.

```
C:\work\avr\avrdude_test\avrdude_bin> C:\Users\xfchen\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/avrdude.conf -v -pattiny1627 -ccuriosity_updi -Pusb -D -U .\Blink.ino.t1627opti.20c0.mB1.v261.hex

avrdude.exe: Version 6.3-20201216
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\xfchen\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : curiosity_updi
avrdude.exe: Found CMSIS-DAP compliant device, using EDBG protocol
             AVR Part                      : ATtiny1627
             Chip Erase delay              : 0 us
             PAGEL                         : P00
             BS2                           : P00
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 0
             StabDelay                     : 0
             CmdexeDelay                   : 0
             SyncLoops                     : 0
             ByteDelay                     : 0
             PollIndex                     : 0
             PollValue                     : 0x00
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
               prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
               fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
               fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
               lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
               data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
               usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
               flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
               eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

             Programmer Type : JTAGICE3_UPDI
             Description     : Microchip Curiosity in UPDI mode
             ICE hardware version: 0
             ICE firmware version: 1.21 (rel. 37)
             Serial number   : MCHP3258021800000382
             Vtarget         : 3.31 V
             JTAG clock megaAVR/program: 0 kHz
             JTAG clock megaAVR/debug:   0 kHz
             JTAG clock Xmega: 0 kHz
             PDI clock Xmega : 100 kHz

avrdude.exe: Partial Family_ID returned: "tiny"
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude.exe: Device signature = 0x1e9428 (probably t1627)
avrdude.exe: reading input file ".\Blink.ino.t1627opti.20c0.mB1.v261.hex"
avrdude.exe: input file .\Blink.ino.t1627opti.20c0.mB1.v261.hex auto detected as Intel Hex
avrdude.exe: writing flash (1190 bytes):

Writing | ################################################## | 100% 0.13s

avrdude.exe: 1190 bytes of flash written
avrdude.exe: verifying flash memory against .\Blink.ino.t1627opti.20c0.mB1.v261.hex:
avrdude.exe: load data flash data from input file .\Blink.ino.t1627opti.20c0.mB1.v261.hex:
avrdude.exe: input file .\Blink.ino.t1627opti.20c0.mB1.v261.hex auto detected as Intel Hex
avrdude.exe: input file .\Blink.ino.t1627opti.20c0.mB1.v261.hex contains 1190 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.08s

avrdude.exe: verifying ...
avrdude.exe: 1190 bytes of flash verified

avrdude.exe: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude.exe done.  Thank you.
```

mcuee, 2022-11-01:
@SpenceKonde 
It seems to me there is a bug, not so sure why it does not build the hex file with the 0x200 offset. I choose the right board option, "Official Microchip board w/Optiboot".

The bootloader burning issue is a known one which you have already fixed in git.

The avrdude git issue is not your issue and I will raise an issue in avrdude.

mcuee, 2022-11-01:
I think this issue has been fixed in the git master. I just copied the three txt files in the directory and it works fine. Take note I am using the latest avrdude git master (including avrdude.conf). 

```
PS C:\work\arduino> git clone https://github.com/SpenceKonde/megaTinyCore.git

PS C:\work\arduino\megaTinyCore\megaavr> ls

    Directory: C:\work\arduino\megaTinyCore\megaavr

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----           1/11/2022  7:20 pm                bootloaders
d----           1/11/2022  7:20 pm                cores
d----           1/11/2022  7:20 pm                extras
d----           1/11/2022  7:20 pm                libraries
d----           1/11/2022  7:20 pm                tools
d----           1/11/2022  7:20 pm                variants
-a---           1/11/2022  7:20 pm         567532 avrdude.conf
-a---           1/11/2022  7:20 pm         247507 boards.txt
-a---           1/11/2022  7:20 pm          26776 LICENSE.md
-a---           1/11/2022  7:20 pm          17455 platform.txt
-a---           1/11/2022  7:20 pm           3851 programmers.txt

PS C:\work\arduino\megaTinyCore\megaavr> cp *.txt C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1\

```

mcuee, 2022-11-01:
The only strange thing is the warning.
```
Bootloader file specified but missing: C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1\bootloaders\optiboot_Curiosity1627.hex
```

It is missing a `hex` in betwern. The file is here:
`C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1\bootloaders\hex\optiboot_Curiosity1607.hex`.

mcuee, 2022-11-01:
@SpenceKonde 

I will close this issue as the main issue has been fixed in git master. The warning is a small thing you can take a look to see if it has been fixed or not.

mcuee, 2022-11-01:
Hmm, the above is with Arduino IDE 2.01. With 1.8.19 it failed.

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xiaof\AppData\Local\Arduino15\packages -hardware C:\Users\xiaof\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xiaof\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xiaof\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:microchipo:board_name=cn1627,clocko=20internaltuned,bodvoltage=1v8,bodmode=disabled,millis=enabled,resetpin=UPDIaltrst,startuptime=8,wiremode=mors,printf=default,attach=allenabled -vid-pid=03EB_2175 -ide-version=10819 -build-path C:\Users\xiaof\AppData\Local\Temp\arduino_build_13713 -warnings=none -build-cache C:\Users\xiaof\AppData\Local\Temp\arduino_cache_691517 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avrdude.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.arduinoOTA.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.python3.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -verbose C:\Users\xiaof\Documents\Arduino\Uno\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xiaof\AppData\Local\Arduino15\packages -hardware C:\Users\xiaof\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xiaof\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xiaof\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:microchipo:board_name=cn1627,clocko=20internaltuned,bodvoltage=1v8,bodmode=disabled,millis=enabled,resetpin=UPDIaltrst,startuptime=8,wiremode=mors,printf=default,attach=allenabled -vid-pid=03EB_2175 -ide-version=10819 -build-path C:\Users\xiaof\AppData\Local\Temp\arduino_build_13713 -warnings=none -build-cache C:\Users\xiaof\AppData\Local\Temp\arduino_cache_691517 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avrdude.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.arduinoOTA.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.python3.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -verbose C:\Users\xiaof\Documents\Arduino\Uno\Blink\Blink.ino
Using board 'microchipo' from platform in folder: C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1
Using core 'megatinycore' from platform in folder: C:\Users\xiaof\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.1
Detecting libraries used...
"C:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1627 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERB1 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_CuriosityNano1627 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=1UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -DLED_BUILTIN=PIN_PB4 -DPIN_BUTTON_BUILTIN=PIN_PB7 -DCLOCK_TUNE_INTERNAL "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.1\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.1\\cores\\megatinycore" "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardwar [... cut ...]
Writing | ################################################## | 100% 0.09s

avrdude: 722 bytes of flash written
avrdude: verifying flash memory against C:\Users\xiaof\AppData\Local\Temp\arduino_build_13713/Blink.ino.hex

Reading | ################################################## | 100% 0.09s

avrdude avr_verify() error: verification mismatch, first encountered at addr 0x0200
        device 0xff != input 0x3b
avrdude do_op() error: verification mismatch

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected
```

mcuee, 2022-11-01:
I will carry out more testing to have more consistent test results.

mcuee, 2022-11-01:
Looks I will have to wait for the next release to carry out more consisten testing. The current results may not be valid.

mcuee, 2022-11-04:
Issue has been fixed by using another hex file ``optiboot_tx2z_rst.hex` (default UART0 pins, UPDI on RA0, Reset on the alternative PB4 pin) instead of  `optiboot_curiosity1627.hex`.

* https://github.com/avrdudes/avrdude/issues/1166#issuecomment-1302174034

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -qq -pattiny1627 -cpkobn_updi -Pusb -e 
-Ufuse0:w:0x00:m -Ufuse1:w:0x00:m -Ufuse5:w:0b11111110:m -Ufuse6:w:0x04:m 
-Ufuse8:w:0x02:m -Ufuse7:w:0x00:m -Uflash:w:optiboot_tx2z_rst.hex:i && echo OK
OK

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM20 -b 115200 -p t1627 -D 
-U .\hello_world_256B_offset_0x200.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9428 (probably t1627)
avrdude: reading input file .\hello_world_256B_offset_0x200.hex for flash
         with 256 bytes in 1 section within [0x200, 0x2ff]
         using 4 pages and 0 pad bytes
avrdude: writing 256 bytes flash ...

Writing | ################################################## | 100% 0.01 s

avrdude: 256 bytes of flash written
avrdude: verifying flash memory against .\hello_world_256B_offset_0x200.hex

Reading | ################################################## | 100% 0.01 s

avrdude: 256 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c pkobn_updi -p t1627 
-U flash:v:.\hello_world_256B_offset_0x200.hex:i -qq && echo OK
OK

```

mcuee, 2022-12-04:
Basically `optiboot_curiosity1627.hex` is broken.

SpenceKonde, 2022-12-05:
Huh. Well I've got a curiosity 1627. I'll subject this specimen to a thorough investigation. 

If it does not divulge to me the nature of the problem and fix in an expedient manner, I'll make that chip wish it were still silicate rock!

SpenceKonde, 2022-12-05:
Okay, I have found multiple problems in every Microchip board def. Except the 1627 which is the one tyou claim is broken,.

mcuee, 2022-12-05:
> Okay, I have found multiple problems in every Microchip board def. Except the 1627 which is the one you claim is broken,.

I actually did not claim that the board def for ATtiny1627 Curiosity Nano was broken.

I only say that `optiboot_curiosity1627.hex` is broken. I will check again.

mcuee, 2022-12-06:
Same issue here. The default optiboot hex file `optiboot_xplained817.hex` is broken but the alternative hex file `optiboot_txyz.hex` is okay for the Microchip ATtiny817 Xplained Mini and Pro board.
* https://github.com/SpenceKonde/megaTinyCore/issues/842#issuecomment-1338614824

SpenceKonde, 2022-12-24:
These will be fixed in 2.6.5

closed 2022-12-24 by SpenceKonde

mcuee, 2022-12-25:
@SpenceKonde 

Still missing one optiboot_x hex file for ATtiny817 Xplained Pro.

mcuee, 2022-12-25:
The hex file is now working for ATtiny1627 Curiosity Nano -- but I am using avrdude as I could not get DxCore to work.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -v -pattiny1627 -cpkobn_updi -Pusb -e -Ufuse0:w:0x00:m -Ufuse1:w:0x00:m -Ufuse5:w:0b11111110:m -Ufuse6:w:0x04:m -Ufuse8:w:0x02:m -Ufuse7:w:0x00:m -Uflash:w:optiboot_Curiosity1627.hex:i

avrdude: Version 7.0-20221223 (8573442)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

         Using Port                    : usb
         Using Programmer              : pkobn_updi
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATtiny1627
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7       append      0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8       bootend     0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   10      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          1    1      0     0     0 0x00 0x00
           tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no         61   61      0     0     0 0x00 0x00
           sernum                  0     0     0    0 no         10    1      0     0     0 0x00 0x00
           osccal16                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osccal20                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc16err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           osc20err                0     0     0    0 no          2    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   32      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      16384   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Curiosity nano (nEDBG) in UPDI mode
         ICE HW version  : 0
         ICE FW version  : 1.21 (rel. 37)
         Serial number   : MCHP3258021800000382
         Vtarget         : 3.31 V
         PDI/UPDI clk    : 100 kHz

avrdude: partial Family_ID returned: "tiny"
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9428 (probably t1627)
avrdude: erasing chip
avrdude: reading input file 0x00 for fuse0/wdtcfg
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse0/wdtcfg ...
avrdude: 1 byte of fuse0/wdtcfg written
avrdude: verifying fuse0/wdtcfg memory against 0x00
avrdude: 1 byte of fuse0/wdtcfg verified
avrdude: reading input file 0x00 for fuse1/bodcfg
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse1/bodcfg ...
avrdude: 1 byte of fuse1/bodcfg written
avrdude: verifying fuse1/bodcfg memory against 0x00
[... 163 lines cut ...]
04d0  ff cf ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
04e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
04f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> dump flash 0x3f00
3f00  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3f10  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3f20  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3f30  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3f40  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3f50  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3f60  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3f70  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
[... cut ...]
3f90  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3fa0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3fb0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3fc0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3fd0  ff ff ff ff ff ff ff 62  6c 69 6e 6b 5f 61 74 74  |.......blink_att|
3fe0  69 6e 79 31 36 32 37 5f  6f 70 74 69 62 6f 6f 74  |iny1627_optiboot|
3ff0  2e 68 65 78 00 e6 07 0b  01 15 04 d2 04 05 3b 1e  |.hex... ......;.|

avrdude> quit

```