SpenceKonde/megaTinyCore #853: Problem with add/remove with board manager

jtannenba, 2022-11-01:
I was goin to tinker with an attiny817 and was reading the readme on programming it.  I had IDE 1.8.19 and downloaded 1.8.13.  Loading that, uninstall the previous one which should be ok. Found I had 2.4.2 installed.  When trying to update I get this error: 

java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:213)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:276)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onRemovePressed$3(ContributionManagerUI.java:210)
	... 1 more

I also get the same error if I try to remove it.  I'm running Window 10 pro and have the latest java installed.

Is this from trying to downgrade the IDE?

Thanks,
Joe

hmeijdam, 2022-11-02:
Deleting the core manually seems to be[ the workaround](https://github.com/SpenceKonde/megaTinyCore/issues/296) for this issue

SpenceKonde, 2022-11-02:
I think that was the cursed version, where I screwed up the board manager json update, and it resulting in this issue for the unfortunate souls who installed it during that brief window.

SpenceKonde, 2022-11-02:
Closing. I had backed myself into a corner where I had to choose which group of users to trigger this bug on (causing it for nobody wasn't an option). It was decided that the users that had 2.4.2 were likely the smallest group at that point in time, as it had been out for a very short time, and another version was already out - so the people who updated quickly would be fine, and the people who rarely updated were unlikely to be impacted. In any event, you want to wait for 2.6.2, because "burn bootloader" is broken in 2.6.1, and I just fixed the last bug an am waiting for CI testing to finish.

closed 2022-11-02 by SpenceKonde