SpenceKonde/megaTinyCore #854: error: redefinition of 'AnalogComparator Comparator0

lerouxb, 2022-11-05:
https://github.com/SpenceKonde/megaTinyCore/blob/34e46152b0ca44638cfb6ea42e04c4294a7a6a8a/megaavr/libraries/Comparator/src/Comparator.cpp#L86-L87

These should probably be Comparator1 and Comparator2. Currently it declares Comparator0 three times.

SpenceKonde, 2022-11-05:
This in 2.6.2? Sounds like there's a problem with our testing regime, cause
all checks pass

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

lerouxb, 2022-11-06:
2.6.1, but I can see it in main as well. 2.6.2 404s if I try and install it in Arduino 1.8.13. On mac in case that matters.

SpenceKonde, 2022-11-06:
Fixed in 2.6.3

closed 2022-11-06 by SpenceKonde