SpenceKonde/megaTinyCore #855: Comparator.cpp:243:36: error: expected ',' or '...' before 'register8_t'

lerouxb, 2022-11-05:
https://github.com/SpenceKonde/megaTinyCore/blob/34e46152b0ca44638cfb6ea42e04c4294a7a6a8a/megaavr/libraries/Comparator/src/Comparator.cpp#L242

Missing comma

SpenceKonde, 2022-11-05:
Thanks what version is this, 2.6.2?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

lerouxb, 2022-11-06:
2.6.1. Unable to install 2.6.2 - I get a 404 error. I can see the same code in the master branch.

hmeijdam, 2022-11-06:
This error?
[image]

lerouxb, 2022-11-06:
Yes sorry exactly that error - I think my brain autocompleted it as a 404 but in reality I don't know more than that.

SpenceKonde, 2022-11-06:
fixed in 2.6.3

closed 2022-11-06 by SpenceKonde