SpenceKonde/megaTinyCore #859: Add WDT Fuse Configuration

matthew-mower, 2022-11-08:
I'm using a ATTINY1624 without a bootloader and would like to use the WDTCFG fuses rather than relying on successfully configuring it at startup. It should be possible to configure these bits in the bootloader menu.

I'm planning on implementing this, but am very new to the arduino boards setup. I imagine this would something like below, correct?

```
menu.wdtwindow=Default WDT Window Timeout Period
menu.wdtperiod=Default WDT Timeout Period
atxy4.menu.wdtwindow.off=Off
atxy4.menu.wdtwindow.off.bootloader.windowbits=0000
atxy4.menu.wdtwindow.1s=1 seconds
atxy4.menu.wdtwindow.1s.bootloader.windowbits=1000
atxy4.menu.wdtperiod.off=Off
atxy4.menu.wdtperiod.off.bootloader.periodbits=0000
atxy4.menu.wdtperiod.1s=1 seconds
atxy4.menu.wdtperiod.1s.bootloader.periodbits=1000

atxy4.bootloader.WDTCFG=0b{bootloader.windowbits}{bootloader.periodbits}
```

SpenceKonde, 2022-11-08:
Yeah something like that. You can also manually write the fuse after doing burn bootloader (WDT configuration is only changed upon  - this is an uncommonly used enough feature, and some people find the sheer number of to be a nuisance and source of confusion, and resist my adding of additional ones. 

Do note that Optiboot should not be used if the watchdog timer is configured in this way. Windowed mode will prevent the bootloader from working at all (it would very rapidly slam into the window, as it issues WDRs much faster than that, the modification need to support that would be a nightmare to implement). And non-windowed mode will override the bootloader's timeout.

SpenceKonde, 2022-11-08:
This isn't a particularly challenging thing to add though- now the tricky question - does it get set on upload, or on burn bootloader? 

I think it qualifies for being set during all uploads, as the criteria is that only things which could potentially softbrick a chip should be on bootload only. I cut off both options before the minimum so that we don't run the risk of it resetting while UPDI is trying to put the chip into a state where the WDT won't reset it, like how spamming the SWRR immediately upon startup can complicate programming

SpenceKonde, 2022-11-08:
This will be in 2.6.4 which I have to release anyway because 2.6.3 was no good.

closed 2022-11-08 by SpenceKonde