SpenceKonde/megaTinyCore #866: Cannot compile FastLed library on Attiny1614

argargreg, 2022-11-25:
I'm trying to compile a program that includes the infamous Fastled for the Attiny1614 with optiboot. Arduino throws me several similar errors on the fact that "PINB", "DDRA" "DDRB" etc. are not declared, and it comes from a macro in FastLed. 

```
In file included from C:\Users\<REPLACED>\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastled_avr.h:4:0,
                 from C:\Users\<REPLACED>\Documents\Arduino\libraries\FastLED\src/platforms.h:41,
                 from C:\Users\<REPLACED>\Documents\Arduino\libraries\FastLED\src/FastLED.h:54,
                 from C:\Users\<REPLACED>\RDXSync\perso\LEDMI\LEDMI_archive\code\MIDI_lightup_piano\Arduino\MIDI_lightup_piano_fw\MIDI_lightup_piano_fw.ino:2:
C:\Users\<REPLACED>\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h: In static member function 'static volatile uint8_t& __gen_struct_DDRA::r()':
C:\Users\<REPLACED>\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:70:26: error: 'DDRA' was not declared in this scope
 #define _FL_IO(L,C) _RD8(DDR ## L); _RD8(PORT ## L); _RD8(PIN ## L); _FL_DEFINE_PORT3(L, C, _R(PORT ## L));

C:\Users\<REPLACED>\Documents\Arduino\libraries\FastLED\src/platforms/avr/fastpin_avr.h:70:42: error: invalid initialization of reference of type 'reg8_t {aka volatile unsigned char&}' from expression of type 'PORT_t {aka PORT_struct}'
 #define _FL_IO(L,C) _RD8(DDR ## L); _RD8(PORT ## L); _RD8(PIN ## L); _FL_DEFINE_PORT3(L, C, _R(PORT ## L));
```

There are a bunch of warnings too on "ISO C++1z does not allow 'register' storage class specifier [-Wregister]", I'm guessing it's not important.

This happens AFTER I add the following case in Documents\Arduino\libraries\FastLED\src\platforms\avr:
```
#elif defined(__AVR_ATtiny841__) || defined(__AVR_ATtiny441__)
#define MAX_PIN 11

_FL_DEFPIN(0, 0, B); _FL_DEFPIN(1, 1, B); _FL_DEFPIN(2, 2, B);
_FL_DEFPIN(3, 7, A); _FL_DEFPIN(4, 6, A); _FL_DEFPIN(5, 5, A);
_FL_DEFPIN(6, 4, A); _FL_DEFPIN(7, 3, A); _FL_DEFPIN(8, 2, A);
_FL_DEFPIN(9, 1, A); _FL_DEFPIN(10, 0, A); _FL_DEFPIN(11, 3, B);

#define HAS_HARDWARE_PIN_SUPPORT 1

#elif defined(__AVR_ATtiny1614__)
#define MAX_PIN 13

_FL_DEFPIN(8, 1, B); _FL_DEFPIN(11, 2, A); _FL_DEFPIN(3, 7, A);
_FL_DEFPIN(9, 0, A); _FL_DEFPIN(12, 3, A); _FL_DEFPIN(2, 6, A);
_FL_DEFPIN(13, 0, A); _FL_DEFPIN(7, 2, B); _FL_DEFPIN(1, 5, A);
_FL_DEFPIN(10, 1, A); _FL_DEFPIN(6, 3, B); _FL_DEFPIN(0, 4, A);

#define HAS_HARDWARE_PIN_SUPPORT 1

#elif defined(ARDUINO_AVR_DIGISPARK) // digispark pin layout
```

Of course, I added #define __AVR_ATtiny1614__ before including Fastled (I'm sure there is a better define which is already defined?).

I don't know what to do to be able to compile this library for Attiny1614, never modified the platforms folder before other than to change a buffer size. Could you advise or fix? 

And thanks again for your involvement on AttinyCore and megaTinyCore - I have been using the former for years in pro and perso settings and they've been a life saver. Just like megaTinyCore, I'm sure.

argargreg, 2022-11-25:
When I add the attiny1614 to the cases that select how to construct the nomenclature of the pin, port etc, I get a lot of these errors and warnings still. I'm surprised I'm the first one to use Fastled with these cores.

warning: asm operand 18 probably doesn't match constraints
error: impossible constraint in 'asm'

SpenceKonde, 2022-11-28:
I fixed FASTLED to work on all modern AVRs, submitted a pull request, and the author merged it. Are you using latest fastled? It

SpenceKonde, 2022-11-28:
OH! FAST_LED, not FABLED. 

Yeah we know that one doeswn't work, The guy who understood the code died a few years ago in a boating accident and the whole project has been in adownward spiral. I looked at it,m and the code is just incomprehensible - it's a classic example iof kitchensinkism. 

There is a shortage of proper 2812/11 libraries for these parts  - that's why THE CORE INCLUDES tinyNeoPixel (identical API to adafruit neopixel) and tinyNeoPixel_Static with has small changes in the constructor syntax but saves like 1k of flash and the memory reported when you compile reflects the memory used by the sketch (adafruit's ignores the buffer, can easily get larger than the total memory on some tinyAVRs). 

I also improved the assembly and corrected the errors in their usage of constraints which avoided catastrophic failure only by virtue of the compile's optimization being toothless when classes were involved. Had it been able to inline, the whole thing would have collapsed. These improvements were most marked on the classic tinyAVRs, where I was able to get rid of the tools submenu and permit any pin to be used for LED output.

SpenceKonde, 2022-11-28:
Closing. That library is both out our our hands and unlikely to ever achieve it's former glory. Suggest migrating to different library anyway.

closed 2022-11-28 by SpenceKonde