SpenceKonde/megaTinyCore #867: Power Down Sleep & I2C

ayushsharma82, 2022-12-04:
Hi @SpenceKonde ,

I was using the following code to enter power down sleep for my Attiny402 which is acting as an I2C slave but the following code is still causing times where I2C line is stuck upon sleep:

___Existing Code :___
```cpp
set_sleep_mode(SLEEP_MODE_PWR_DOWN);
sleep_enable();
sleep_cpu();
```

For switching to v2.6.4, , do we have to replace the above logic with only `sleep_cpu()` line?

___New Code:___
```cpp
sleep_cpu();
```

Regards,
Ayush Sharma

SpenceKonde, 2022-12-05:
As I understand @MX682X please correct me if wrong, the problem happens when you set_sleep_mode during an I2C operation,. We (by which I mean him, cause I would need to study the code for a very long time to understand it before trying to change it) take very vigorous countermeaasures; nothing can foil them except setting the sleep mode while a transaction is ongoinmg, at which point everything will hit the fan.

MX682X, 2022-12-05:
@ayushsharma82 assuming you have something like 
```
loop() {
[...]
set_sleep_mode(SLEEP_MODE_PWR_DOWN);
sleep_enable();
sleep_cpu();
[...]
}
```
then there is a chance that, depending on the length of your loop, that the power down sleep will be reenabled between the address match irq and data irq. As, in this case, only the address match irq provides a wake-up from power down, the device will stay in eternal sleep. To avoid that, the library buffers the sleep mode on address match and restores in at the stop condition. During the transaction the sleep is set to idle, thus providing a way of minimizing power consumption as far as possible.

If you don't want idle sleep, and you only have the TWI as wakeup source, it should be possible to do this:

```
loop() {
[...]
set_sleep_mode(SLEEP_MODE_PWR_DOWN);
sleep_enable();
sleep_cpu();
//woken up by TWI, do something and/or wait until the transaction is finished
while (i2cBytesReceived == 0) { //global variable set in onReceive() Wire function
} 
[...]
}
```

But I recommend using one of the following:
```
setup() {
[...]
set_sleep_mode(SLEEP_MODE_PWR_DOWN);
sleep_enable(); //here or in loop
[...]
}

loop() {
[...]
sleep_enable(); //here or in setup
sleep_cpu();
[...]
}
```

```
loop() {
[...]
set_sleep_mode(SLEEP_MODE_PWR_DOWN);
sleep_enable();
while (i2cBytesReceived == 0) {  //global variable set in onReceive() Wire function
   sleep_cpu();
}
[...]
}
```

ayushsharma82, 2022-12-05:
@SpenceKonde & @MX682X thanks for following back on this issue. Yes my sleep code is in the loop and I'm using TWI address match and GPIO interrupt ( for some other task ) to wakeup.

Today I'll be building a dedicated test bench which will initiate I2C transactions in a loop at short intervals. I'll test out all the suggestions by @MX682X and reply with the results. My desired sleep state is Power Down mode as that provides the best efficient for my battery powered device.

ayushsharma82, 2022-12-06:
Got it. Thanks for giving time to my issue and clearing my doubts, I highly respect it!

@SpenceKonde, I first tried using megaTinyCore using PlatformIO but quickly figured out that their registry contains a very old version of this repo. You may want to talk with the platformIO guys to get it updated, as these latest releases contain a lot of bug fixes.

Regards,
Ayush Sharma

SpenceKonde, 2022-12-24:
Yeah I dunno what's up with the platform IO people. I think they don't like the hackjobbed unofficial toolshcain

closed 2022-12-27 by SpenceKonde