SpenceKonde/megaTinyCore #872: SerialUPDI cli IndexError: bytearray index out of range

wohali, 2022-12-17:
When trying to flash an ATTiny412 directly from the CLI using a pre-generated .hex file, I receive the following error:

```
$ python3 ./prog.py -d attiny412 -wd 1 -u /dev/ttyUSB0 -a write -f ./ATTiny-OSC-TCD0.X.production.hex
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.3 - Jan 2022
Using serial port /dev/ttyUSB0 at 115200 baud.
Delaying next op after each page write command by 1.0ms (-wd).
Target: attiny412
Action: write
File: <...path/to/>ATTiny-OSC-TCD0.X.production.hex
Pinging device...
Ping response: 1E9223
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.06s
Writing from hex file...
Writing flash...
[==================================================] 4/4
Writing fuses...
[==================================================] 9/9
Writing lockbits...
Traceback (most recent call last):
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/./prog.py", line 285, in <module>
    main()
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/./prog.py", line 128, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/./prog.py", line 250, in pymcuprog_basic
    run_pymcu_action(pymcu._action_write, backend,
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/./prog.py", line 138, in run_pymcu_action
    status = func(backend, *args, args_pymcu)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/pymcuprog_main.py", line 340, in _action_write
    _write_memory_segments(backend, result, args.verify, blocksize=args.blocksize, pagewrite_delay=args.pagewrite_delay)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/pymcuprog_main.py", line 388, in _write_memory_segments
    backend.write_memory(segment.data, memory_name, segment.offset, blocksize=blocksize, pagewrite_delay=pagewrite_delay)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/backend.py", line 526, in write_memory
    self.programmer.write_memory(data=data, memory_name=memory_name, offset=offset_byte, blocksize=blocksize, pagewrite_delay=pagewrite_delay)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/programmer.py", line 190, in write_memory
    self.device_model.write(memory, offset, data, blocksize=blocksize, pagewrite_delay=pagewrite_delay)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/nvmserialupdi.py", line 178, in write
    self.avr.nvm.write_flash(offset_aligned, chunk, blocksize=blocksize, bulkwrite = bulk, pagewrite_delay=pagewrite_delay)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/serialupdi/nvm.py", line 118, in write_flash
    return self.write_nvm(address, data, use_word_access=True, blocksize=blocksize,  bulkwrite=bulkwrite, pagewrite_delay=pagewrite_delay)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/serialupdi/nvm.py", line 203, in write_nvm
    self.execute_nvm_command(nvmcommand)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/serialupdi/nvm.py", line 77, in execute_nvm_command
    return self.readwrite.write_byte(self.device.nvmctrl_address + constants.UPDI_NVMCTRL_CTRLA, command)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/serialupdi/readwrite.py", line 64, in write_byte
    return self.datalink.st(address, value)
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/serialupdi/link.py", line 361, in st
    return self._st_data_phase([value & 0xFF])
  File "/home/joant/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/tools/libs/pymcuprog/serialupdi/link.py", line 300, in _st_data_phase
    self.logger.error("expecting ACK after ST, but got: %02x", response[0])
IndexError: bytearray index out of range
```

SpenceKonde, 2022-12-28:
Please post a hex file that reproduces this - it looks like you may be using a hex file that embeds fuses and lockbits. That facility is not tested (and in fact I don't know how it works) and it may be that something isn't being passed correctly between the far too many layers of pymcuprog (why why why did we ever base it on that monstrosity. Oh, I remember.... )

wohali, 2022-12-28:
Here you go.

pymcuprog seems to work fine on its own - that's what I'm having to use to program this project, incidentally, like this:

```
pymcuprog write -d attiny412 -t uart -u /dev/ttyUSB0 -f ATTiny-OSC-TCD0.X.production.hex --erase --verify
```
[ATTiny-OSC-TCD0.X.production.hex.zip](https://github.com/SpenceKonde/megaTinyCore/files/10316227/ATTiny-OSC-TCD0.X.production.hex.zip)

SpenceKonde, 2022-12-29:
Thanks. As I suspected, the hex file came from some other utility that embeds fuses and lockbits into the hex file, unlike what comes out of avr-gcc as invoked by platform.txt 

Pymcuprog - isn't that a bit slow? When we picked that up we were seeing like 1k/s upload speeds and worse. The compatibility problems likely owe to the agressive performance improvements we needed to do to make the upload speed fast enough to be practical. As I said, we have not done testing with the fuses and lockbits embedded in the hex file. Hell, I didn't even know how that was done until now. I think if you didn't have the fuses and lockbits embedded into the hex file, and instead specified them at the command line (as is the intended usage of prog.py) it would work.  But I will put this on the list of issues we need to work out.

wohali, 2022-12-30:
> Thanks. As I suspected, the hex file came from some other utility that embeds fuses and lockbits into the hex file, unlike what comes out of avr-gcc as invoked by platform.txt

Yup, I couldn't quickly work out how to do the TCD0 tricks I needed with your core, though I did start with it - esp. because I need to mess with fuse settings and I don't need anything other than TCD0 in this target application.
 
> Pymcuprog - isn't that a bit slow? [snip]

Yup, but it works. Very eager to switch to your utility, when it can work with this file.

> As I said, we have not done testing with the fuses and lockbits embedded in the hex file. Hell, I didn't even know how that was done until now. I think if you didn't have the fuses and lockbits [...]

I prefer to have my fuse settings in the same source file as the rest of my application when possible. That's chiefly because I'm forgetful, but also because build systems come and go, and I'd rather have a generic Makefile (or equiv.) than one that includes application-specific settings.

SpenceKonde, 2023-01-02:
Yeah - we don't support setting the TCD0 fuse from the IDE. Only time we set it is on burn bootloader, and only back to it's default ("burn bootloader should return the part to a fully defined state"). It seemed sort of silly, since you still need to do initialization on TCD, so why bother with multiple tools submenus to set a fuse ( 2, since it has 256 different values broadly grouped into two classes) when you could just set it while configuring the TCD to begin with? You'd need to set TCD to not be the millis timer and call takeOverTCD0() first anyway to prevent analog/digitalWrite() on PC0, PC1 from making a mess - and then configure TCD0

TCD0 is a mighty unfriendly timer IMO.... between the enable-locking of everything (remember that the core does kick off TCD0 for PWM, so you either need to override init_TCD0 with an empty function, or stop it, wait until ENRDY is set, and THEN you can begin configuring things. And FAULTCTRL is under CCP for some whacko reason. It's very powerful.... but... unfriendly. 

The issue with SerialUPDI is a bug, however. I think something is going wrong where we have to pass all these arguments between the 7 onion-like layers we inherited from the abomination that is pymcuprog (basing SerialUPDI on that was not our best decision)

mcuee, 2023-02-19:
@wohali 

If you can not get the python SerialUPDI CLI utility to work, you may want to give avrdude 7.1 a try now that you have the binary file for the avr chip generated by megaTinyCore.
https://github.com/avrdudes/avrdude/releases/tag/v7.1 (Windows binary)

You can also use Arduino version here (Windows, Linux and macOS binary)
https://github.com/arduino/avrdude-packing/releases

SpenceKonde, 2023-03-01:
@wohali Can you let me know what the settings were being set in that hex file? I'm trying to understand the format, but I can't really figure that out without knowing what fuses and apparently other settings (there's more data that doesn't make sense as flash contents than just the fuses) I'm not able to do that.

SpenceKonde, 2023-03-01:
Is it these? 
The ~strikethrough~ bytes are just checksum bytes. 
The **bold** bytes are the ones that 
:02 0000 04 00 82 ~78~ <-- record of type 4, not normal data
:09 0000 **00 00 00 7D 00 30 F6 FA 00 00** ~5A~ <- fuses
:02 0000 04 00 83 ~77~ <-- record of type 4, not normal data
:01 0000 00 **C5** ~3A~ <---Lockbyte? This would indicate chip not being locked.

SpenceKonde, 2023-07-01:
Hey, I don't suppose you can see if calling pymcuprog with -v generates verbose output? I know there was some way to get absurdly verbose output from it. 

What I need a log of is the *successful* writing of a lockbyte - it is not clear from either the code or the documentation what the correct process to write a lockbyte is.... (I remember now looking into it a while back and realizing that there was literally no documentation on how to lock a chip - I know what the start and end states should look like, but I don't know whether I write it like flash or like fuses. 

The issue is that the bulk flash write function is being called **with only 1 byte of data**. So then it rightshifts it 1 bit and subtracts 1 to get the number of times the command needs to be repeated - because it thinks it's writing by word not by byte (hence the rightshift) and the subtracting 1 is because the unrepeated command will handle one of the words. So it tells the chip to disable it's response signature, and to expect 512 bytes. We then send it 1 byte. It is waiting for 511more. The next three bytes it sends are 0x55, 0xc2, 0x06,  then  0x55, 0x44, 0x00, 0x10. The device is expecting another 504 bytes (even though the part has only 64-byte pages, mind you...), But at this point prog.py is expecting to get an ACK, because 0x55, 0xc2, 0x06 was sent after it wrote all the data it had - that would normally turn RSD back on. since we lied to the device about how much data to expect, though, that doesn't happen, and it gets interpreted as data. (I haven't a clue what it's doing with additional data - I have an image in my head of [someone loading up the vehicles in these pictures](https://www.google.com/search?q=ridiculously+overloaded+vehicles&tbm=isch) (google image search that brings up a bunch of those viral photos of people with insane amounts of stuff piled/stuffed/tied-to a car, truck, or motorbike - if you know the kind of pictures I'm describing, you don't need to waste time looking at them)

So yeah, I can trap that error. But I need a verbose log of a SUCCESSFUL WRITE TO THE LOCKBYTE in order to successfully write a file like that, because **microchip has apparently chosen not to document how to use their code security feature**. 

That's a real endorsement of the security of the mechanism, that they don't even tell you how to use it in the documentation..... That's totally how a company behaves when they're confident in their security measures, right? We used to do that with security stuff where I worked, the result was just that internally nobody did a good job of testing it because the documentation was intentionally incomplete. 

It doesn't stop any of the bad guys, who know that already (in the case of chips, specialist firmware extraction companies that specialize in dumping the memory of locked chips for a fee - AFAICT, the methods for breaking classic AVR lockbits are well known in that community - but I have not seen any modern AVRs listed on any of the places I saw advertising it (I was looking for one of those annotated die photos you sometimes see, but of any modern AVR. Like where they mark what parts of the chip die are the ram and what's the flash and where's the ALU, and where are the peripherals. I'm incredibly curious about what the ratio, roughly of space taken up by peripherals vs ALU vs RAM vs Flash. I do not think there are any publically accessible die shots of modern AVRs). Companies and governments have been decapping microchips and examining them, as long as people have been encapsulating chips.)

wohali, 2023-07-01:
HI @SpenceKonde , I'll give it a go the next time I'm in the lab, which will be after the long weekend.

SpenceKonde, 2023-07-14:
Any word on whether I've fixed this? I'd really like to declare this one fixed or know that it's still broken while I remember where the code involved was (and ofc it needs to be ported to DxCore's version of SerialUPDI wish due to the other bug is not identical, because I don't have a way of passing the "is this a Dx or a tiny" piece of information from where I know it to where I need to know it in order to accurately read out the serial number)

wohali, 2023-07-15:
Sorry for the delay - will get you an answer anon.

wohali, 2023-07-21:
You asked about the configuration bits, here they are from MPLAB X:

```
FUSES = {
	.WDTCFG = 0x00, // WDTCFG {PERIOD=OFF, WINDOW=OFF}
	.BODCFG = 0x00, // BODCFG {SLEEP=DIS, ACTIVE=DIS, SAMPFREQ=1KHz, LVL=BODLEVEL0}
	.OSCCFG = 0x7D, // OSCCFG {FREQSEL=16MHZ, OSCLOCK=CLEAR}
	.TCD0CFG = 0x30, // TCD0CFG {CMPA=CLEAR, CMPB=CLEAR, CMPC=CLEAR, CMPD=CLEAR, CMPAEN=SET, CMPBEN=SET, CMPCEN=CLEAR, CMPDEN=CLEAR}
	.SYSCFG0 = 0xF6, // SYSCFG0 {EESAVE=CLEAR, RSTPINCFG=UPDI, CRCSRC=NOCRC}
	.SYSCFG1 = 0xFA, // SYSCFG1 {SUT=2MS}
	.APPEND = 0x00, // APPEND {APPEND=User range:  0x0 - 0xFF}
	.BOOTEND = 0x00, // BOOTEND {BOOTEND=User range:  0x0 - 0xFF}
};

LOCKBITS = 0xC5; // {LB=NOLOCK}
```

Next up, running `pymcuprog` in a way so as to generate lots of verbose debugging info. Give me a little bit here...

wohali, 2023-07-21:
Results (timed to show how long programming takes - not too bad, honestly):

```
$ time pymcuprog write  -v debug -d attiny414 -t uart -u /dev/ttyUSB0 -f /home/joant/Wakko/Projects/MFMEmulator/SA1000/ATTiny-OSC-TCD0.X/dist/default/production/ATTiny-OSC-TCD0.X.production.hex --erase --verify
Connecting to SerialUPDI
pymcuprog.programmer - INFO - Setting up programming session for 'attiny414'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device attiny414
pymcuprog.serialupdi.physical - INFO - Opening port '/dev/ttyUSB0' at 115200 baud (timeout 1.0s)
pymcuprog.serialupdi.physical - DEBUG - send : [0x00]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x03
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC3, 0x08]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x02
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC2, 0x80]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x20]
pymcuprog.serialupdi.link - INFO - UPDI init OK
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xE6]
pymcuprog.serialupdi.application - INFO - SIB: 'tinyAVR P:0D:0-3M2 (01.59B14.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'tinyAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:0'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(01.59B14.0)'
pymcuprog.serialupdi.application - INFO - NVM type 0: 16-bit, page oriented write
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x20]
pymcuprog.serialupdi.application - INFO - PDI revision = 0x02
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8B]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x82]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8B]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x82]
pymcuprog.serialupdi.application - INFO - Entering NVM programming mode
pymcuprog.serialupdi.link - DEBUG - Writing key
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xE0]
pymcuprog.serialupdi.physical - DEBUG - send : [0x20, 0x67, 0x6F, 0x72, 0x50, 0x4D, 0x56, 0x4E]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x07
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x87]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x10]
pymcuprog.serialupdi.application - DEBUG - Key status = 0x10
pymcuprog.serialupdi.application - INFO - Apply reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC8, 0x59]
pymcuprog.serialupdi.application - INFO - Release reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC8, 0x00]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8B]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x28]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8B]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x08]
pymcuprog.serialupdi.application - DEBUG - Now in NVM programming mode
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xE6]
pymcuprog.serialupdi.application - INFO - SIB: 'tinyAVR P:0D:0-3M2 (01.59B14.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'tinyAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:0'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(01.59B14.0)'
pymcuprog.serialupdi.application - INFO - NVM type 0: 16-bit, page oriented write
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x20]
pymcuprog.serialupdi.application - INFO - PDI revision = 0x02
[... 23 lines cut ...]
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x69, 0x00, 0x11]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.link - DEBUG - Repeat 3
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xA0, 0x02]
pymcuprog.serialupdi.link - DEBUG - LD8 from ptr++
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x24]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x1E, 0x92, 0x23]
pymcuprog.nvm - INFO - Device ID: '1E9223'
pymcuprog.pymcuprog_main - ERROR - Device ID mismatch: read out '1E9223'; expected '1E9222'.
pymcuprog.serialupdi.application - INFO - Leaving NVM programming mode
pymcuprog.serialupdi.application - INFO - Apply reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC8, 0x59]
pymcuprog.serialupdi.application - INFO - Release reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC8, 0x00]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x03
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC3, 0x0C]
Done.
pymcuprog.serialupdi.physical - INFO - Closing port '/dev/ttyUSB0'

real	0m2.236s
user	0m0.107s
sys	0m0.019s
```

wohali, 2023-07-21:
Results (including time to run, which was 2.737s):

```
$ time pymcuprog write  -v debug -d attiny412 -t uart -u /dev/ttyUSB0 -f /home/joant/Wakko/Projects/MFMEmulator/SA1000/ATTiny-OSC-TCD0.X/dist/default/production/ATTiny-OSC-TCD0.X.production.hex --erase --verify
Connecting to SerialUPDI
pymcuprog.programmer - INFO - Setting up programming session for 'attiny412'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device attiny412
pymcuprog.serialupdi.physical - INFO - Opening port '/dev/ttyUSB0' at 115200 baud (timeout 1.0s)
pymcuprog.serialupdi.physical - DEBUG - send : [0x00]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x03
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC3, 0x08]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x02
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC2, 0x80]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x20]
pymcuprog.serialupdi.link - INFO - UPDI init OK
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xE6]
pymcuprog.serialupdi.application - INFO - SIB: 'tinyAVR P:0D:0-3M2 (01.59B14.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'tinyAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:0'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(01.59B14.0)'
pymcuprog.serialupdi.application - INFO - NVM type 0: 16-bit, page oriented write
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x20]
pymcuprog.serialupdi.application - INFO - PDI revision = 0x02
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8B]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x82]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8B]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x82]
pymcuprog.serialupdi.application - INFO - Entering NVM programming mode
pymcuprog.serialupdi.link - DEBUG - Writing key
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xE0]
pymcuprog.serialupdi.physical - DEBUG - send : [0x20, 0x67, 0x6F, 0x72, 0x50, 0x4D, 0x56, 0x4E]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x07
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x87]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x10]
pymcuprog.serialupdi.application - DEBUG - Key status = 0x10
pymcuprog.serialupdi.application - INFO - Apply reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC8, 0x59]
pymcuprog.serialupdi.application - INFO - Release reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC8, 0x00]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8B]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x28]
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x0B
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x8B]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x08]
pymcuprog.serialupdi.application - DEBUG - Now in NVM programming mode
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xE6]
pymcuprog.serialupdi.application - INFO - SIB: 'tinyAVR P:0D:0-3M2 (01.59B14.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'tinyAVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:0'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:0'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(01.59B14.0)'
pymcuprog.serialupdi.application - INFO - NVM type 0: 16-bit, page oriented write
pymcuprog.serialupdi.link - DEBUG - LDCS from 0x00
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x80]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x20]
pymcuprog.serialupdi.application - INFO - PDI revision = 0x02
[... 636 lines cut ...]
pymcuprog.serialupdi.link - INFO - ST to ptr
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x69, 0x8A, 0x12]
pymcuprog.serialupdi.physical - DEBUG - receive : [0x40]
pymcuprog.serialupdi.link - DEBUG - LD8 from ptr++
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0x24]
pymcuprog.serialupdi.physical - DEBUG - receive : [0xC5]
pymcuprog.programmer - INFO - Verifying...
OK
pymcuprog.serialupdi.application - INFO - Leaving NVM programming mode
pymcuprog.serialupdi.application - INFO - Apply reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC8, 0x59]
pymcuprog.serialupdi.application - INFO - Release reset
pymcuprog.serialupdi.link - DEBUG - STCS to 0x08
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC8, 0x00]
pymcuprog.serialupdi.link - DEBUG - STCS to 0x03
pymcuprog.serialupdi.physical - DEBUG - send : [0x55, 0xC3, 0x0C]
Done.
pymcuprog.serialupdi.physical - INFO - Closing port '/dev/ttyUSB0'

real	0m2.737s
user	0m0.205s
sys	0m0.043s

```

wohali, 2023-07-21:
Does this give you what you need?

SpenceKonde, 2023-07-21:
Okay, as I suspected Write Fuse to write to the lockbyte. Okay. I don't understand exactly what is confusing it that leads to it treating the lockbit write as a full page write to flash, but we can catch that case very easily, because in NVMv0, if the flash-write function is handed a single byte write to 0x128A, it can easily recognize that and handle it properly (I think I have implemented that in latest versions). Needs to be checked on Dx too. probably an analogous bug for NVMv2 because I think the bug is before the codepath splits up between V0 and V2- I know how the lockbits are written there at least.

closed 2023-07-21 by SpenceKonde