SpenceKonde/megaTinyCore #876: [BUG]: can't burn bootloader on Ubuntu, invalid literal '0b010{bootloader.bodmodebits}'

okawo80085, 2022-12-21:
So i've been trying to flash new FUSE values from Arduino IDE, but every time i try i get an invalid literal error.

Error message in Arduino IDE:
```
Error: cannot parse fuse, 'invalid literal for int() with base 0: '0b010{bootloader.bodmodebits}''
Failed to burn bootloader: uploading error: exit status 1
```

Tried both core version 2.6.1 and 2.6.4, both have the same issue.

Tried re installing the Arduino IDE and megaTinyCore, didn't help.

What am i missing?

okawo80085, 2022-12-21:
Correction, only `Sampled/Sampled` BOD FUSE values result in this error, `Enable/Enable` and others work fine... ?

SpenceKonde, 2022-12-21:
This appears to be a bug in the core, have urgent work for a client to do today though

okawo80085, 2022-12-21:
Take your time, this is not urgent for me

commit 5bfe1994 (2022-12-22): Fix #876  #871 and the bootloaders for microchip boards

Fix #876  #871 and the bootloaders for microchip boards,. And a case of unsafe failure to use the volatile keyword for millis related values
closed 2022-12-22 by SpenceKonde