SpenceKonde/megaTinyCore #881: Device signature = 0x1e942a (probably t1624). Expected signature for ATtiny1614 is 1E 94 22

uterator, 2022-12-30:
Hi,
Thank you for the all work done in megaTinyCore.
I am using it for a long time, and now I faced an issue after switching from 1604 to 1624. 
At uploading a program on ATtiny1624 it fails with 
```
avrdude: Device signature = 0x1e942a (probably t1624)
avrdude: Expected signature for ATtiny1614 is 1E 94 22
         Double check chip, or use -F to override this check.
Failed to burn bootloader: uploading error: exit status 1
```

Both mentioned signatures match the specs
[ATtiny1614](https://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny1614-16-17-DataSheet-DS40002204A.pdf) 0x1E 0x94 0x22
[ATtiny1624](https://ww1.microchip.com/downloads/aemDocuments/documents/MCU08/ProductDocuments/DataSheets/ATtiny1624-26-27-DataSheet-DS40002234B.pdf) 0x1E 0x94 0x2A

It was working successfully for a long time with ATtiny1604 before moving to ATtiny1624. I just updated the megaTinyCore version to the latest one to have a support for 1624 (and replaced the board if that makes sense).

**Environment**
Arduino IDE: `Version: 2.0.0-rc9.2, Date: 2022-08-10T13:01:55.702Z, CLI Version: 0.26.0-rc.1 [fc2ea723]`
OS: `Linux`
megaTinyCore: `Version 2.6.4`
board: `Arduino UNO`

**Config**
[image]

**command**
```
avrdude "-C/home/gegham/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/avrdude.conf" -v -pattiny1614 -cjtag2updi -P/dev/ttyUSB0 -b115200 -e "-Ufuse0:w:0b00000000:m" "-Ufuse1:w:0x00:m" "-Ufuse2:w:0x02:m" "-Ufuse5:w:0b11000101:m" "-Ufuse6:w:0x04:m" "-Ufuse8:w:0x00:m" "-Ufuse7:w:0x00:m" ""
```

As we see the `-pattiny1614` parameter is passed to the `avrdude` command which is not correct as it had to be 1624. I manually changed it to `-pattiny1624` and run the `avrdude` command in terminal, it didn't fail but I am not sure if it was a correct thing to do as I kept all parameters for fuses the same from the old command with `-pattiny1614` (I don't see even the program worked after that).

Am I missing something in configuration or it can be corrected in the lib?
Thanks

**Full logs**
```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/gegham/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/avrdude.conf"
"/home/gegham/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18/bin/avrdude" "-C/home/gegham/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.4/avrdude.conf" -v -pattiny1614 -cjtag2updi -P/dev/ttyUSB0 -b115200 -e "-Ufuse0:w:0b00000000:m" "-Ufuse1:w:0x00:m" "-Ufuse2:w:0x02:m" "-Ufuse5:w:0b11000101:m" "-Ufuse6:w:0x04:m" "-Ufuse8:w:0x00:m" "-Ufuse7:w:0x00:m" ""
         User configuration file is "/home/gegham/.avrduderc"

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : ATtiny1614
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
         S_MCU hardware version: 1
         S_MCU firmware version: 6.00
         Serial number:          00:00:00:00:00:00
         Vtarget         : 5.0 V

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.16s

avrdude: Device signature = 0x1e942a (probably t1624)
avrdude: Expected signature for ATtiny1614 is 1E 94 22
         Double check chip, or use -F to override this check.
Failed to burn bootloader: uploading error: exit status 1
```

SpenceKonde, 2023-01-01:
Please re-test. I see several problems.
The Chip submenu does not appear to have anything selected; I believe in this case it uses the first option, which is ATtiny1614 in the most recent release - though this may be an interface difference (a major downgrade if so). Be sure that all submenus have an option selected, particularly "chip". 
You list board as Arduino Uno; this is in direct conflict with everything else you say, because the board you are referencing otherwise is ATtiny3224/1624/(snip)/204. Please clarify what is meant here. 
You are using a pre-release 2.0.0 version of the IDE. We do not test with prereleases, and the RC versions have had a fairly large number of regressions. The latest 1.8.x release should work. I use 1.8.13 because of a bug in 1.8.14 and later (but fixed in 2.0.0) that makes development more troublesome, but the most recent releases have worked around that one.

uterator, 2023-01-03:
@SpenceKonde Thank you for the answer, and yes you are right, chip setting was incorrect. 
And thank you for the IDE version notice, however the 2.0.0-rc9.2 worked as well and I like the editor there (built on [theia](https://theia-ide.org/)) but I will continue with 1.8.13 as you recommend.

Regarding to chip selection, the 2.0.0-rc9.2's menu doesn't show the submenu selected values, should be a UI bug.
[image]
I used to not see values there so I didn't pay attention on chip's value before reporting, not a something which I do every day. If not that bug, the reason of the reported issue would be very obvious. 

The selected chip and other selected options are visible on 1.8.13's menu.
[image]

As for the board, I mentioned the device which is used only as a programmer for my tiny MCUs.

Now my setup is 1.8.13 with mTC 2.6.4 and it works.

Thank you.

closed 2023-01-03 by uterator