SpenceKonde/megaTinyCore #894: Issue with Wire library - sending 0 causes an error

icarusbop, 2023-01-09:
### Discussed in https://github.com/SpenceKonde/megaTinyCore/discussions/891

<div type='discussions-op-text'>

<sup>Originally posted by **icarusbop** January  6, 2023</sup>
Hello

I have a Raspberry Pi connected to a test circuit with an ATTINY 1624, using the I2c interface.
The Rpi is using the SMBUS2 python library, the 1624 is using the Wire  Library

I can successfully send data from the ATTINY 1624 to the Rpi over I2c, with one small exception - if I try to send a value of zero(0), I get an error on the Rpi receiving side. I have tested with a loop and can send every value from 255 down to 1, it only fails at zero.

I can't be sure if the issue is on the Rpi side or the 1624 side, but I have tried testing different functions on the Rpi library side to achieve the same effect (receiving a string of bytes, but only keeping the first byte - this uses a different function and therefore presumably different code), and I get the same error. This makes me wonder if the issue might be on the wire library for the 1624?

Here is the minimum code for the 1624 to replicate my issue (I can post the Rpi test code if required)

```
#include <Wire.h> //I2c

byte ToSend=255; //I set this to 255 to test every value, issue only occurs when we get to zero
unsigned char CharBuff;	//I copy the value to send into this char buffer	
	char SendBuffer[1] ; //make a buffer to put the send data in

void RequestEvent() {
	/*on interrupt send data to I2c master */	
	CharBuff=ToSend--; //copy the value sent then decrease for next time.			
	memcpy(&(SendBuffer), &CharBuff, 1);  //copy data into the send buffer
	Wire.write(SendBuffer); //send the data to the I2c 	
}	

void setup() {	
	Wire.begin(58); //open the I2C bus
	Wire.onRequest(RequestEvent); // register event for master reading
}

void loop() {
}
```

The error I receive on the Rpi side is:

>   File "/usr/local/lib/python3.7/dist-packages/smbus2/smbus2.py", line 396, in read_byte
>     ioctl(self.fd, I2C_SMBUS, msg)
> OSError: [Errno 121] Remote I/O error

</div>

MX682X, 2023-01-09:
I'm not sure how the compiler handles an array with the length of 1, but if it does it like with any other array, you are likely calling `Wire.write((char*)SendBuffer);`, this is handled by the print class like this:
```
size_t write(const char *str) {
      if (str == NULL) {
        return 0;
      }
      return write((const uint8_t *)str, strlen(str));
    }
```
strlen(SendBuffer) equals 0, thus no data is being written. And sending no data is not allowed or something. I have to admit, that I don't know why it works with the other values though.

Anyway, the best way to confirm it, is to try this:

```
void RequestEvent() {
	uint8_t data = ToSend--;
	Wire.write(data); //send the data to the I2c 	
}	
```

P.S.: The Arduino library has it's own buffer, it does not require a user supplied buffer. The write functions just copy the supplied data.

SpenceKonde, 2023-01-10:
MX682X has nearly hit the nail on the head. 

What appears to be happening here, determined from examining the assembly listing is that it's going towards this:
``` 
   size_t write(const char *str) {
      if (str == NULL) {
        return 0;
      }
      return write((const uint8_t *)str, strlen(str));
    }
```
As you can see, it does just what he described. 

And of course it does. You are passing in a pointer to an array of characters with no length specified. That's a C string, and a 0x00, the nul, marks the end of a C string. So you've given it an empty string. and told it to send that. Okay, there's nothing to send. What I sort of wonder is - what does wire do in this case? You're backing it into a bad corner... as far as it understands, there is no data to send. The master is trying to clock out a byte, but onRequest didn't write any data to the Wire buffer. What does it do when that happens? It's clearly a situation in which there is no *correct* response that will do what the user wanted, but I'm somewhat curious about what it does :-P 

Really what surprises me is that that works for other numbers. I guess the buffer gets filled with up to the next 32 bytes in memory unless one of them is a 0, and you just don't see that because the master only asks for one byte. I think the only reason that it works at all is that it's initializing the subsequent byte to 0 because it likes to word align things. 

One issue this code raises in my mind is that you may not understand how the wire slave interrupts work. 

There are two interrupts that the user actually sees, onRequest and onReceive. 
onReceive is fired when, after an address match with the read/write bit set 0 (write), followed by the master writing whatever data, and then generating a stop condition. Then, when you have that data in the library's buffer. it calls the supplied onReceive. when you get the interrupt with all that data. This one is simple. 

The onRequest is fired after an address match with the read/write bit set 1 (read). Immediately. Because that is a declaration from the master that it intends to read one or more bytes - you need to tell wire what to send, even though you have no idea how many bytes the master is going to try to get (per the I2C standard, the master indicates the end of a read by responding to the final byte with a NACK instead of an ACK), so you need to prepare for the longest read that you support. 

The I2C buffer is 32 bytes, so that's the most you can write to it (further character are ignored, and write returns 0's). That size limit applies to all parts supported by DxCore or mTC with two exceptions: Dx/Ex-series part with 4096b ram or more, in which case it's bigger (which does not cause compatibility problems), or it's a 2k flash 0/1-series tiny, in which case it has just 128b of ram and we *had* to reduce the buffer size (to 16b in this case), since there are multiple buffers that need to exist). That means that some libraries won't work there, but nobody will know because the binary for a library like that is gonna come out too large to fit in the 2k flash that those parts have too)

MX682X, 2023-01-10:
>  I'm somewhat curious about what it does 

Fair enough. When data is requested, the ISR calls onRequest(). Next it checks if the head is not zero. This is useful, in case of a Register like implementation - master writes a register number and then tries to read that register. If onRequest hasn't written any data, like in the OP's case, a NACK is issued.

SpenceKonde, 2023-01-10:
Oh! Of course! during onRequest it's stretching the clock so if it has nothing to send at the end of it, obviously it just nacks. Cool. Duh

SpenceKonde, 2023-01-10:
Close because conclusively demonstrated to to incorrect user code rather than a defect in the core.

closed 2023-01-10 by SpenceKonde

icarusbop, 2023-01-10:
> I'm not sure how the compiler handles an array with the length of 1, but if it does it like with any other array, you are likely calling `Wire.write((char*)SendBuffer);`, this is handled by the print class like this:
> 
> ```
> size_t write(const char *str) {
>       if (str == NULL) {
>         return 0;
>       }
>       return write((const uint8_t *)str, strlen(str));
>     }
> ```
> 
> strlen(SendBuffer) equals 0 (as strlen increments until it finds 0x00), thus no data is being written. And the RPi is requesting 1 byte that is never send, thus the remote I/O error, I guess. And it works with other values because the initialization sets all unused ram to 0x00, thus the next byte after SendBuffer is 0x00. So, technically, your code probably had an out-of-bounds bug.
> 
> Anyway, the best way to confirm it, is to try this:
> 
> ```
> void RequestEvent() {
> 	uint8_t data = ToSend--;
> 	Wire.write(data); //send the data to the I2c 	
> }	
> ```
> 
> P.S.: The Arduino library has it's own buffer, it does not require a user supplied buffer. The write functions just copy the supplied data.

Hello and thanks for your response - I have had to read it a few times and take some time before I (think) I've figured out what you are saying (hence my delayed response)

I tried the little sample replacement you provided and it works and can transmit all values including zero - so I guess that means the issue is with my code somehow and not the library. Thanks very much for your help.

Regarding the buffer - in my main project the transmit buffer is set up as a union as I need to be able to send different types of data down it. I'll look into figuring out an alternative way of doing this.

MX682X, 2023-01-10:
> Hello and thanks for your response - I have had to read it a few times and take some time before I (think) I've figured out what you are saying (hence my delayed response)

If there are uncertainties, feel free to ask about them. I know that my thoughts can be kinda chaotic and all over the place and so can be my explainations.

> Regarding the buffer - in my main project the transmit buffer is set up as a union as I need to be able to send different types of data down it. I'll look into figuring out an alternative way of doing this.

then I would suggest to try following. It will treat the union as array and copy all bytes to the internal buffer.

`Wire.write((uint8_t*)&(*name of union*), sizeof(*name of union*));`

icarusbop, 2023-01-11:
> > Hello and thanks for your response - I have had to read it a few times and take some time before I (think) I've figured out what you are saying (hence my delayed response)
> 
> If there are uncertainties, feel free to ask about them. I know that my thoughts can be kinda chaotic and all over the place and so can be my explainations.
> 
> > Regarding the buffer - in my main project the transmit buffer is set up as a union as I need to be able to send different types of data down it. I'll look into figuring out an alternative way of doing this.
> 
> then I would suggest to try following. It will treat the union as array and copy all bytes to the internal buffer (up to the size of the buffer, see post above).
> 
> `Wire.write((uint8_t*)&(*name of union*), sizeof(*name of union*));`

Hello - this works great, thanks very much.
I did just try sending the union, without the '(uint8_t*)' part, but that didn't work.
If you have a moment please could you explain it to me?

Thank you.

SpenceKonde, 2023-01-11:
Wouldn't sending the union directly just give an error that there was no matching function, or would it get implicitly converted to something dumb? Doesn't wire, for some reason, **truncate** types larger than a byte when passed to write? And hence the there are two equivalent ways out - either you cast the pointer to the union to a uint8_t *, and then you can write() the bytes as long as you tell it how many, or you have a byte array in your union, of length equal to the longest element, and use that. (Why would arduino have it truncate things to bytes? Likely because the master and slave write are consistent on this point, and if they tried it with automatically splitting up larger datatypes, they found that as master it was a disaster because people use int16_t's when they should be using uint8_t and int8_t, at least in arduinoland, so things would get two bytes instead of one, hence writing to the intended register but also the one located after it, and everyhing would spiral down from there.

MX682X, 2023-01-11:
> I did just try sending the union, without the '(uint8_t*)' part, but that didn't work.

As I don't know what didn't work (I can come up with three things: Compiler gave an error, Wrong data sent, or no data sent) I can not tell what happend with the code. 

Based on your initial post, I guess you are not familiar with pointers. To sum up: Pointers are integers, in the case of an AVR, that are 16 bit wide. Every byte in the RAM has it's own address. They are used to pass around big variables, without having to copy the value itself. To get the address of the first byte of a variable, you can use the & operator.

Why to use '(uint8_t*)' - this is called casting, and tells the compiler that the value or variable after the cast is, in this case, a pointer to an array with 1 byte long elements. It does not chnage the value however. If the compiler complains, that means it has noticed that a pointer to a union is not the same as a pointer to a char/uint8_t array. It wants to make sure that the data a function gets can be processed by said function. But as we want to send an array of single bytes we can safely tell the compiler to treat the union simply as an array in this case. (Sizeof then calculates how many bytes there are to send)

I do not belive the code is using Wire.write(uint16_t) and truncating it, as it has a size argument that is also passed

icarusbop, 2023-01-12:
Thanks to both of you for your explanations.

I am new to pointers, this has really helped me understand what is going on with the casting to a pointer.
It all makes more sense now.

Thanks for all the help.