SpenceKonde/megaTinyCore #898: Please document the WDT timeout and WDT window Tools menu options

technoblogy, 2023-01-19:
In recent versions of megaTinyCore two new **Tools** menu options have appeared:

**WDT timeout**
**WDT window**

Please could someone document them in the section:

https://github.com/SpenceKonde/megaTinyCore#list-of-tools-sub-menus

In particular, what does "note that it cannot be disabled if set here!" mean? Can it be disabled by any other method?

Also, can someone explain why we need these options in the **Tools** menu, because the page **Reference: All about Resets** seems to explain how to achieve the same effect from a program:

https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Reset.md

SpenceKonde, 2023-01-21:
> n recent versions of megaTinyCore two new **Tools** menu options have appeared:
> 
> **WDT timeout** **WDT window**
> 
> Please could someone document them in the section:
> 
> https://github.com/SpenceKonde/megaTinyCore#list-of-tools-sub-menus

Yeah some time. Prioriry: Low. 
> 
> In particular, what does "note that it cannot be disabled if set here!" mean? Can it be disabled by any other method?
If WDT is enabled in the fuses, it cannot be turned off by the sketch. 
> Also, can someone explain why we need these options in the **Tools** menu, because the page **Reference: All about Resets** seems to explain how to achieve the same effect from a program:

Because people don't fucking read my docs very carefully, I have to put key pieces of information in multiople places to try to reduce the number of people who dot see them. 
Ref. Reset only concerns a WDT reset creamed at runtime I by using a short timeout to get a faster restat. 

> https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Reset.md

technoblogy, 2023-01-22:
Thanks for the answers.

> In particular, what does "note that it cannot be disabled if set here!" mean? Can it be disabled by any other method?

I understood it as: "Once you've set this fuse you can never unset it with another Burn Bootloader". I suggest rewording:

"note that it cannot be disabled if set here!"

to:

"if set here it cannot be turned off by the sketch"

commit f1665290 (2023-01-27): Fix #898

Document WDT menu options.
closed 2023-01-27 by SpenceKonde