SpenceKonde/megaTinyCore #903: megatinycore on a Chromebook - device-specs folder issue

SpenceKonde, 2023-01-26:
### Discussed in https://github.com/SpenceKonde/megaTinyCore/discussions/902

<div type='discussions-op-text'>

<sup>Originally posted by **PaulEast** January 26, 2023</sup>
Good morning everyone, firstly I'm in awe of this development and have it working perfectly on a Linux box but I'm now trying to get it working on my Chromebook.

I have the board package and additional board manager link in place on Arduino version1.8.13 (installed on the Chromebook linux installation).

When compiling I get 2 error messages:

`Warning: platform.txt from core 'megaTinyCore' contains deprecated compiler.path={runtime.tools.avr-gcc.path}/bin/, automatically converted to compiler.path=/usr/bin/. Consider upgrading this core.
Detecting libraries used...`

and then...

`avr-g++: error: device-specs/specs-attiny404: No such file or directory
exit status 1
Error compiling for board ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204.`

I have checked and there is a device-specs/specs-attiny404 folder.

I am at the edge of my knowledge on Linux and the backend of Arduino, if this is just an edge case that won't work that's fine but I thought I'd see if the collective wisdom here could help.

Thanks in advance.</div>

SpenceKonde, 2023-01-27:
I am convinced this is caused by the same thing that causes the 

```

Warning: platform.txt from core 'megaTinyCore' contains deprecated compiler.path={runtime.tools.avr-gcc.path}/bin/, automatically converted to compiler.path=/usr/bin/. Consider upgrading this core.
```
But I don't know what the deprecated path needs to be changed to in order to make it use the compiler toolchain I supply.

SpenceKonde, 2023-01-27:
And there is no reason this wouldn't impact DxCore in the same way.

SpenceKonde, 2023-01-29:
One of the luminaries of Arduino and experts on the IDE has explained to me how that error message works and where to find the replacements. However, the thing that it is rewriting is not one of the ones that the Arduino IDE rewrites. It appears that you are using a copy of the Arduino IDE distributed through a linux package manager. Those are *notorious* for being modified by package manager administrators who don't let their lack of relevant background knowledge about particular pieces of software stand in the way of their making unwise changes to the IDE. 

In this case it appears that they have added a rewrite rule that changes the path specified to the version of the toolchain I provide and which was installed by board manager based on the json file, which supports all released parts (even ones released just months ago), and rewrites it to point to some other version of the toolchain, which is having trouble with a part released years ago. I will also note that the rewrite in question will break all of my hardware packages, at least one official hardware packages, several other third party hardware packages, and possibly nearly all hardware packages. God knows what else they've done to it. Uninstall the package manager version, download the *real* Arduino IDE version 1.8.13 from arduino.cc and install that, and reinstall the core, and that should sort things out. 

(1.8.13 seems to be the most functional version - there was a huge UX fix related to the programmers that are supported by third party hardware packages that was added in 1.8.13, so earlier versions are no good, because you are shown incompatible programmers in the tools -> programmer list, often multiple ones with the same name, of which only one will work with each core. And 1.8.14 introduced a nasty bug, and the 2.0.x versions, though they fix that bug introduced a large number of severe bugs and have been reported to not work with my cores. I'm waiting until 2.0.x works before I devote limited time to that)

closed 2023-01-29 by SpenceKonde

PaulEast, 2023-02-08:
I'm sorry I didn't get a notification that you'd relied.

Can confirm removing 1.8.13 installed from the package manager and installing 1.8.19 downloaded direct from arduino.cc works perfectly. 

@SpenceKonde thank you and the expert who helped so much, this is an amazing resource and a great team of people! 

(For anyone in future getting similar on a Chromebook I used [these instructions](https://kb.learnrobotics.org/software/installation/arduino-ide-on-chromebook/))