SpenceKonde/megaTinyCore #905: Unable to install MegaTinyCore on Ubuntu Linux, Arduino IDE 1.8.19

makeriot2020, 2023-01-29:
Hi Guys, 
Having issues installing this core on ArduinoIDE 1.8.19 via board Manager.
Get the following errors: ( see below )
I have manually navigated to the github pages mentioned, and the files seem to be gone?
Any idea what happened, or where to get them? or alternative options to install ? 

Error downloading https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:145)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:79)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:56)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:108)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 503
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:222)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:176)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:125)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:143)
	... 5 more
 Error downloading https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:145)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:79)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:56)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:108)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 503
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:222)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:176)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:125)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:143)
	... 5 more
 Error downloading https://SpenceKonde.github.io/DxCore/DxCore-1.5.3.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading https://SpenceKonde.github.io/DxCore/DxCore-1.5.3.tar.bz2

SpenceKonde, 2023-01-29:
I cannot reproduce this issue. That error indicates that you are unable to download the file specified. This could be due to restricted or absent internet access, or because of a temporary server outage (as it's hosted on github this is beyond my control). 

The file downloads fine for me now, so I've got to kick this one back to you. Are you able to download that file, (the URL in the last two lines) on the computer that has this issue? I can downloaded it on my system, so if you still can't, there is a connectivity issue on the system in question. If you can but the IDE can't, it's suggestive of some sort of overzealous security configuration on your system. Either way, I think it's a problem that only you can solve, because it's a system level problem, unrelated to the core itself. 

If it works when you try installing it now, it was a transient server issue at Github which they have resolved.

makeriot2020, 2023-01-29:
Hi,
Thank you for the prompt response.
I have tried again just now. and no luck...
The system does not have any funny security issues... Just tried other
cores as a test, and that works...

From what I can tell, the files in the DXCore/Raw folder on your github, is
not visible to me, for some reason....

I got similar:

*Error downloading
https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
<https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2>java.lang.RuntimeException:
java.lang.Exception: Error downloading
https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
<https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2>
at
cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:179)
at java.lang.Thread.run(Thread.java:748)Caused by: java.lang.Exception:
Error downloading
https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
<https://github.com/SpenceKonde/DxCore/raw/gh-pages/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2>
at
cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:145)
at
cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:79)
at
cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:56)
at
cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:108)
at
cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
... 1 moreCaused by: java.io.IOException: Received invalid http status code
from server: 503 at
cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:222)
at
cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:176)
at
cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:125)
at
cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:143)
... 5 more*

Trying with wget, i got this:
[image: image.png]
where it just hangs...

I understand that it is NOT your issue, was only hoping for a few
pointers...
It could be that it is because I am in Thailand, and they like to mess
around with internet. ( like the big firewall in China, is a big role model
to them...)

Let me try with a VPN and see what happens

Thank you :)

SpenceKonde, 2023-01-29:
Of all the weird things that could be getting blocked o_O Like, I can see why, oh, github issues/discussions, or github in general might be blocked. But large binaries hosted on github? bizzare!

makeriot2020, 2023-01-29:
Yeah, never been blocked before ... but then Thais are Thais :) someone
might believe there is porn on Github of all places, and block it just in
case :(
anyway, at least I got your Core installed now, so now can start testing
these new chips :)

Thanks again

Regards

SpenceKonde, 2023-01-30:
Great, glad you got it installed, in spite of the poorly executed obstruction from on high. Since this issue is solved, closing issue.

closed 2023-01-30 by SpenceKonde