SpenceKonde/megaTinyCore #907: Using delay() or millis() halts the execution

dabu667, 2023-02-02:
On ATtiny412 delay() and millis() do not work. Only delayMicroseconds() does. I have searched but couldn't find an answer - is this supposed to happen?
My configuration: 
[image]

hmeijdam, 2023-02-02:
I fear you have another problem. Just tested the "blink" example on an Attiny412 and it is blinking happy on PA3

dabu667, 2023-02-02:
You are absolutely right!. I was using main() function. Thank you!

hmeijdam, 2023-02-02:
That's why I recognized your issue. Besides the same delay() issue I had my [share of confusion](https://github.com/SpenceKonde/megaTinyCore/discussions/700) myself, when I used my own main() to save some space.