SpenceKonde/megaTinyCore #909: Error on compiling - "No USARTs? No supported parts exist without one, something is wrong"

ssarna1, 2023-02-06:
When trying to verify an empty sketch for tiny416 i'm getting this:
```
arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/XXX/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/XXX/.arduino15/packages -libraries /home/XXX/Arduino/libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=416,clock=20internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=1A86_7523 -ide-version=10819 -build-path /tmp/arduino_build_725294 -warnings=none -build-cache /tmp/arduino_cache_423435 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=/home/XXX/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/XXX/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.avrdude.path=/home/XXX/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=/home/XXX/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18 -prefs=runtime.tools.arduinoOTA.path=/home/XXX/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/XXX/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/home/XXX/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=/home/XXX/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6 -verbose /tmp/untitled758557843.tmp/sketch_feb05a/sketch_feb05a.ino
arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/XXX/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/XXX/.arduino15/packages -libraries /home/XXX/Arduino/libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=416,clock=20internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=1A86_7523 -ide-version=10819 -build-path /tmp/arduino_build_725294 -warnings=none -build-cache /tmp/arduino_cache_423435 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=/home/XXX/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/XXX/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.avrdude.path=/home/XXX/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=/home/XXX/.arduino15/packages/DxCore/tools/avrdude/6.3.0-arduino17or18 -prefs=runtime.tools.arduinoOTA.path=/home/XXX/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/XXX/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/home/XXX/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=/home/XXX/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6 -verbose /tmp/untitled758557843.tmp/sketch_feb05a/sketch_feb05a.ino
Using board 'atxy6' from platform in folder: /home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5
Using core 'megatinycore' from platform in folder: /home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5
Opcje projektu zmienione, przeładuj całość
Uwaga: platform.txt z rdzenia 'megaTinyCore' zawiera przestarzałe compiler.path={runtime.tools.avr-gcc.path}/bin/, przekonwertowane automatycznie na compiler.path=/usr/bin/. Rozważ uaktualnienie tego rdzenia.
Detecting libraries used...
"/usr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto   -w -x c++ -E -CC -mmcu=attiny416 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny416 -DARDUINO_ARCH_MEGAAVR -DMEGATINYCORE="2.6.5-dev" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=5UL -DMEGATINYCORE_RELEASED=1  -DARDUINO_attinyxy6  "-I/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/api/deprecated" "-I/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore" "-I/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/variants/txy6" "/tmp/arduino_build_725294/sketch/sketch_feb05a.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/usr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing  -flto   -w -x c++ -E -CC -mmcu=attiny416 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny416 -DARDUINO_ARCH_MEGAAVR -DMEGATINYCORE="2.6.5-dev" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=5UL -DMEGATINYCORE_RELEASED=1  -DARDUINO_attinyxy6  "-I/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/api/deprecated" "-I/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore" "-I/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/variants/txy6" "/tmp/arduino_build_725294/sketch/sketch_feb05a.ino.cpp" -o "/tmp/arduino_build_725294/preproc/ctags_target_for_gcc_minus_e.cpp"
In file included from /home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/Arduino.h:28:0,
                 from /tmp/arduino_build_725294/sketch/sketch_feb05a.ino.cpp:1:
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/core_devices.h:291:4: error: #error "No AC? No supported parts exist without one, something is wrong"
   #error "No AC? No supported parts exist without one, something is wrong"
    ^
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/core_devices.h:300:4: error: #error "No ADC? No supported parts exist without one, something is wrong"
   #error "No ADC? No supported parts exist without one, something is wrong"
    ^
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/core_devices.h:331:4: error: #error "No TCA? No supported parts exist without one, something is wrong"
   #error "No TCA? No supported parts exist without one, something is wrong"
    ^
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/core_devices.h:347:4: error: #error "No TCBs? No supported parts exist without one, something is wrong"
   #error "No TCBs? No supported parts exist without one, something is wrong"
    ^
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/core_devices.h:362:4: error: #error "No TWI? No supported parts exist without one, something is wrong"
   #error "No TWI? No supported parts exist without one, something is wrong"
    ^
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/core_devices.h:370:4: error: #error "No SPI? No supported parts exist without one, something is wrong"
   #error "No SPI? No supported parts exist without one, something is wrong"
    ^
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/core_devices.h:387:4: error: #error "No USARTs? No supported parts exist without one, something is wrong"
   #error "No USARTs? No supported parts exist without one, something is wrong"
    ^
In file included from /home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/variants/txy6/pins_arduino.h:27:0,
                 from /home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/Arduino.h:670,
                 from /tmp/arduino_build_725294/sketch/sketch_feb05a.ino.cpp:1:
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/timers.h:97:6: error: #error "TCD0, selected for millis, does not exist on this part"
     #error "TCD0, selected for millis, does not exist on this part"
      ^
In file included from /home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/HardwareSerial.h:36:0,
[... cut ...]
/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/cores/megatinycore/UART_check_pins.h:35:4: error: #error "No USARTs present. All supported parts have at least 1 - something is wrong!"
   #error "No USARTs present. All supported parts have at least 1 - something is wrong!"
    ^
exit status 1
Błąd kompilacji dla płytki ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406.
```

Arduino IDE version is 1.8.19 and OS is Ubuntu 22.04

Is there anything obvious i'm missing? I can compile for normal arduino and for ESP32 so this seems to be isolated to megatinycore.

SpenceKonde, 2023-02-07:
Please open preferences, imder show verbose output, check compiling. Attempt to verify the sketch again, it will fail again,. Copy the entire contents of the console log, paste it into a text file. attach it here if it fits, otherwise zip it and email to spencekonde@gmail.com.

There are a few hints I might be able to find, it's of course difficult to debug since I have only windows systems and my skill with linux is very rudimentary.

SpenceKonde, 2023-02-07:
Say, you're not using Arduino from a package manager are you?

Those version have all been hacked up by people without any relevant knowledge of Arduino and dogmatic beliefs about the One True Way to organize supporting files and executables, entirely unaware that the assumptions underlying that dogma are ivalid in typical embedded system development

The only version of Arduinoi IDE that should be expected to work is one downloaded from Arduino.cc. Ran into this with someone with another package manager Arduino IDE, and it had the rewrite keys so that the correct version of the compiler, which board manager downloads and installs for you was not being used, and instead an incompatible version mission hundreds of critical files was instead.

SpenceKonde, 2023-02-07:
Also it pases tje CI whicj runs on linux

pcfreak1201, 2023-02-08:
I'm on Mint and had no errors yet. Can you compile the UART-example?
Because Windows and Linux have/had a different character base and I see some Czech looking chars - may be a path-problem? Linux is 64bit or 32bit?
Package-manager or ZIP shouldn't be the problem on linux, I think.

ssarna1, 2023-02-08:
[output.txt](https://github.com/SpenceKonde/megaTinyCore/files/10688884/output.txt)
here is the log.

It's been over a year since i've installed that, but i've checked with "apt list" and arduino is there, so i'd assume it was installed with apt.

> Czech looking chars

Almost, a bit more to the North-East. But anyway, there is no weird letters in my username. And i'm trying to compile an empty sketch, there should be no reason why UART example should pass i think.

So Should i uninstall and install again without package manager? Or should we try to debug this?

ssarna1, 2023-02-08:
Okay so i've downloaded the Arduino.cc version and then:
`
tar xf arduino-1.8.19-linux64.tar.xz
cd arduino-1.8.19/
./install.sh
`

Now i have two of them, the one just installed have already all the libraries and boards i had in the first one, and in this one its compiling just fine. I guess i'll keep them both for the time.

pcfreak1201, 2023-02-10:
As long as the ~/.arduino15/ ... files are with the last installed version compatible, there shouldn't be a problem by using the tar-ball. If you have a backup of your system, you can try to remove the package-manager version.
The version of the package-manager arduino ist mostly quite old. So if you want to be up to date, the way you went yesterday sould be OK. But as Spence always says: the 1.8.13 is the "golden" version. (But on linux I'd never problems with the 1.8.19).

SpenceKonde, 2023-02-10:
It's not a matter of old or new. It's that the package manager versions have been modified by third parties to conform to their pregromening  who don't understand the consequences, regardless of what version they're from. That's the darkside of  application package managers. Embedded software still relies on they consider heretical. 

Anway. To summaraize: Isssue was cause by  the apt version right? Never every use package manager of Arduino.

closed 2023-02-10 by SpenceKonde

SpenceKonde, 2023-02-18:
The versions of the IDE that are distributed with package managers are known to be buggy in wildly differing ways. 
Each software repo has their ideology, and Those Who Add Apps modify the apps they add to conform to their requirements. Knowing little about embedded software development or arduino (they're app repository wranglers by trade) they break something (frequently the overall theme of their changes is "The latest version of the compiler toolchain is correct and therefor that is all we supply. Atpack? never heard of that"). But avr-gcc versions past 7 don't always get better because of poor tuning for (and other than LTO, new versions haven't improved) 

Uninstall the version installed through apt. Right now, it is a trap: it looks like an Arduino IDE in many ways, and acts a lot like one. But it doesn't always behave correctly, (#903, #909, etc.).It has one and likely multiple other subtle bugs that will cost you days debugging some time down the road. Get rid of it. 
Reinstall the version from Arduino.cc, install megatinycore via board manager and you'll be set