SpenceKonde/megaTinyCore #913: Bug (Windows): 'Export Compiled Binary' step using incorrect paths with cmd

jschneider1207, 2023-02-14:
## Versions

### Arduino 1.8.19 
### megaTinyCore 2.6.5

## Tl;dr

Windows sucks and command prompt doesn't know what a `/` is. Used `git-bash` instead.

## Details

I was having trouble compiling anything for my attiny1626 on a fresh install of megaTinyCore, error I was getting (with verbose output) was:
```batch
Arduino: 1.8.19 (Windows 10), Board: "ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406, ATtiny1626, 1 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (no reset pin), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), Disabled (recommended), No delay before window "opens""

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jess\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jess\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\jess\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=1626,clock=1internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=0403_6001 -ide-version=10819 -build-path C:\Users\jess\AppData\Local\Temp\arduino_build_799150 -warnings=none -build-cache C:\Users\jess\AppData\Local\Temp\arduino_cache_693975 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\jess\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\jess\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.python3.path=C:\Users\jess\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\jess\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jess\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\jess\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\jess\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\jess\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -verbose C:\Users\jess\AppData\Local\Temp\untitled830910629.tmp\sketch_feb13a\sketch_feb13a.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jess\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jess\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\jess\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=1626,clock=1internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=0403_6001 -ide-version=10819 -build-path C:\Users\jess\AppData\Local\Temp\arduino_build_799150 -warnings=none -build-cache C:\Users\jess\AppData\Local\Temp\arduino_cache_693975 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\jess\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\jess\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.python3.path=C:\Users\jess\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\jess\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jess\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\jess\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\jess\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\jess\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -verbose C:\Users\jess\AppData\Local\Temp\untitled830910629.tmp\sketch_feb13a\sketch_feb13a.ino

Using board 'atxy6' from platform in folder: C:\Users\jess\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.5

Using core 'megatinycore' from platform in folder: C:\Users\jess\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.5

Detecting libraries used...

"C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1626 -DF_CPU=1000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERB1 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny1626 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.5-dev\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=5UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy6 "-IC:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.5\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.5\\cores\\megatinycore" "-IC:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.5\\variants\\txy6" "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_799150\\sketch\\sketch_feb13a.ino.cpp" -o nul

Generating function prototypes...

"C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1626 -DF_CPU=1000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERB1 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny1626 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.5-dev\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=5UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy6 "-IC:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.5\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.5\\cores\\megatinycore" "-IC:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.5\\variants\\txy6" "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_799150\\sketch\\sketch_feb13a.ino.cpp" -o "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_799150\\preproc\\ctags_target_for_gcc_minus_e.cpp"

"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_799150\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny1626 -DF_CPU=1000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERB1 -DCORE_ATTACH_ALL -DARDUINO=10819 -DARDUINO_AVR_ATtiny1626 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.5-dev\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=5UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy6 "-IC:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.5\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.5\\cores\\megatinyc [... cut ...]
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.01s
Writing from hex file...
Writing flash...
[==================================================] 13/13
Action took 0.11s
Verifying...
[==================================================] 2/2
Verify successful. Data in flash matches data in specified hex-file
Action took 0.05s

```

SpenceKonde, 2023-02-25:
I do all my development on windows. I do not see that problem. Switching to git-bash would introduce a dependance on an external tool, and every external tool needs to be compiled for all 7 supported platforms, and might trigger other bugs

The odd structure of those commands is itself a workaround for another bug in the Arduino IDE that mangling the commands in leading edge versions of the IDE.

jschneider1207, 2023-02-25:
>  Switching to git-bash would introduce a dependance on an external tool

Thats fair. Was first the first thing I tried, it also looks to work fine with powershell, which isn't external.

```ps
Compiling libraries...
Compiling core...
Using precompiled core: C:\Users\jess\AppData\Local\Temp\arduino_cache_925929\core\core_3ad210d13afee29758db1a9ae947a078.a
Linking everything together...
"C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-gcc" -Wall -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x0 -mrelax -fno-gcse -mmcu=attiny1626 -o "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.elf" "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849\\sketch\\attiny_test.ino.cpp.o" "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/..\\arduino_cache_925929\\core\\core_3ad210d13afee29758db1a9ae947a078.a" "-LC:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849" -lm
"C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-objcopy" -O binary -R .eeprom "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.elf" "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.bin"
"C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.elf" "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.eep"
"C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.elf" "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.hex"
ps echo. && "C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.elf" > "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.lst"
      PID    PPID    PGID     WINPID  TTY  UID    STIME COMMAND
    39812       1   39812      39812  con  500 11:22:24 /usr/bin/ps
ps echo. && "C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-nm" --numeric-sort --line-numbers --demangle --print-size --format=s "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.elf" > "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.map"
      PID    PPID    PGID     WINPID  TTY  UID    STIME COMMAND
    32920       1   32920      32920  con  500 11:22:24 /usr/bin/ps
"C:\\Users\\jess\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-size" -A "C:\\Users\\jess\\AppData\\Local\\Temp\\arduino_build_541849/attiny_test.ino.elf"
Sketch uses 3435 bytes (20%) of program storage space. Maximum is 16384 bytes.
Global variables use 159 bytes (7%) of dynamic memory, leaving 1889 bytes for local variables. Maximum is 2048 bytes
```

Do you have any suggestions or is there any other info you want to see? This is off a fresh install of both Arduino 1.8.19 and the library.

SpenceKonde, 2023-02-26:
I'm rather mystified as to why it happens to you but not others. I would also be surprised if the objdump part worked on some cores but not on mTC/DxC

The path is built up the same way for all cores (and the "cmd /C echo. &&" was required to work around a case where windows would strip an extra pair of quotes off which would manifest on newer versions of the IDE.

SpenceKonde, 2023-03-01:
This change was made at the direct instruction of the known expert on the Arduino IDE and member of the Arduino team, becaus the version before that would not work n post 1.8.x versions of the IDE and he indicated that that change to the IDE would/could not be undone, and I was told to use the current construction. That seems to be separate from what you're seeing though

I do not know why it doesn't work for you and do not feel qualified to change it without input from more knowledgeable individuals. 

@per1234 - I'm sorry to say that means you. Do you have any comment?

mcuee, 2023-04-06:
Indeed this is very strange. No issues here under Windows 11 and Arduino 1.8.19.

```
...

cmd /C echo. && "C:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino_build_786444/Blink.ino.elf" > "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino_build_786444/Blink.ino.lst"
 
cmd /C echo. && "C:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-nm" --numeric-sort --line-numbers --demangle --print-size --format=s "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino_build_786444/Blink.ino.elf" > "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino_build_786444/Blink.ino.map"
 
"C:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-size" -A "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino_build_786444/Blink.ino.elf"
Sketch uses 612 bytes (3%) of program storage space. Maximum is 16384 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 2044 bytes for local variables. Maximum is 2048 bytes.
cmd /C echo. && copy C:\Users\xiaof\AppData\Local\Temp\arduino_build_786444\Blink.ino.lst C:\Users\xiaof\Documents\Arduino\Tiny1626\Blink\Blink.ino.t1626.1c0.mB1.Os.v265.lst 
 
        1 file(s) copied.
cmd /C echo. && copy C:\Users\xiaof\AppData\Local\Temp\arduino_build_786444\Blink.ino.map C:\Users\xiaof\Documents\Arduino\Tiny1626\Blink\Blink.ino.t1626.1c0.mB1.Os.v265.map 
 
        1 file(s) copied.
```

SpenceKonde, 2023-04-07:
I've gotta say, I think there was something messed up with the terminal configuration initially. That's the only explanation that seems to make sense - we have ample evidence that he is fiddling with advanced OS features since he so easily switched command interpreters. Looks like he just had the misfortune to have chosen some combination of option (what exactly triggers it I'd still like to know, since we can't change the weirdo syntax because without it new versions of the IDE inappropriately strip quotes or something. I think the full discussion was somewhere on MegaCoreX issues

SpenceKonde, 2023-04-20:
No comment from op in month and others not experiencing issue. Conclude was system problem unrelated to Arduino or the core.

closed 2023-04-20 by SpenceKonde

jschneider1207, 2023-06-03:
> No comment from op in month and others not experiencing issue. Conclude was system problem unrelated to Arduino or the core.

Forgot about this, sorry! I just updated from 2.6.5 to 2.6.8, overwrote my tweak and now its busted again 😿

SpenceKonde, 2023-06-04:
I don't know what "Elixer" is or how it's related to this problem. Could you enlighten me?

Also, I have no idea what you mean by "swapped it back to `ps`" - I don't even know how you got ps on windows, I thought that was posix only..... 

But  - let's face it, your configuration is bizzarro, probably one of the most extreme weird configs i've heard of. I mean, FFS you used 3 command interpreters in the first two posts.... The problem is likely something related to your efforts to make the windows computer more linuxy, which I suspect are extensive and pervasive on that machine. In any event: I don't know how to fix it nor how to further debug. I'm out of my league here - not only do I not know the answer (and know that I don't), I don't even follow the question :-/ 

I don't understand these programs for newfangled desktop computers with 32, even 64 bit address spaces (who needs that much ram?! 65536 bytes ought to be enough memory for anyone!