SpenceKonde/megaTinyCore #924: Arduino Crypto library doesn't work with ATTINY3226

specternecter, 2023-03-17:
The Arduino cryptography library (https://github.com/rweather/arduinolibs/tree/master/libraries/Crypto) says that it should work with all Arduino devices. I've used it with many different devices and never had an issue, until I tried to use it with an ATTINY3226. It appears to be referencing timers that don't exist, but I'm not completely sure. It looks to me like the Crypto library needs to be updated to support the ATTINY3226,  but I also know you say to let you know if libraries don't work with this core.

specternecter, 2023-03-17:
Best that I can tell, it's trying to use watchdog as the noise source. I've tried disabling this in RNG.cpp since watchdog isn't recommended and also enabling watchdog in the menu. Both attempts return the same errors.

Arduino: 1.8.19 (Windows 10), Board: "ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406, ATtiny3226, 10 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI w/reset on PB4 (2-series only), 64ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), Disabled (recommended), No delay before window "opens""

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Brad\AppData\Local\Arduino15\packages -hardware C:\Users\Brad\Desktop\Dropbox\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Brad\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Brad\Desktop\Dropbox\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=3226,clock=10internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDIaltrst,startuptime=64,wiremode=mors,printf=default,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\Brad\AppData\Local\Temp\arduino_build_773967 -warnings=none -build-cache C:\Users\Brad\AppData\Local\Temp\arduino_cache_315438 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Brad\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\Brad\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.python3.path=C:\Users\Brad\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Brad\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Brad\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Brad\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\Brad\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\Brad\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -verbose C:\Users\Brad\Desktop\Dropbox\Arduino\rIOT-EXTENSION_FACTORY_EEPROM\rIOT-EXTENSION_FACTORY_EEPROM.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Brad\AppData\Local\Arduino15\packages -hardware C:\Users\Brad\Desktop\Dropbox\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Brad\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Brad\Desktop\Dropbox\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=3226,clock=10internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDIaltrst,startuptime=64,wiremode=mors,printf=default,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\Brad\AppData\Local\Temp\arduino_build_773967 -warnings=none -build-cache C:\Users\Brad\AppData\Local\Temp\arduino_cache_315438 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Brad\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\Brad\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.python3.path=C:\Users\Brad\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Brad\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Brad\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Brad\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\Brad\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\Brad\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -verbose C:\Users\Brad\Desktop\Dropbox\Arduino\rIOT-EXTENSION_FACTORY_EEPROM\rIOT-EXTENSION_FACTORY_EEPROM.ino

Using board 'atxy6' from platform in folder: C:\Users\Brad\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.5

Using core 'megatinycore' from platform in folder: C:\Users\Brad\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.5

Detecting libraries used...

"C:\\Users\\Brad\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny3226 -DF_CPU=10000000L -DCLOCK_SOURCE=0 -DT [... cut ...]
C:\Users\Brad\Desktop\Dropbox\Arduino\libraries\Crypto\RNG.cpp:490:29: note: suggested alternative: 'WDT'

     _WD_CONTROL_REG = (1 << WDIE);

                             ^~~~

                             WDT

Using library EEPROM at version 2.1.3 in folder: C:\Users\Brad\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.5\libraries\EEPROM 

Using library Crypto in folder: C:\Users\Brad\Desktop\Dropbox\Arduino\libraries\Crypto (legacy)

exit status 1

Error compiling for board ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406.

SpenceKonde, 2023-03-17:
I think you would get it to compile (with a warning saying there were no entropy sources) by just commenting out 

#define RNG_WATCHDOG 1      // Harvest entropy from watchdog jitter.

Though then you'd need to figure out how to scare up some entropy on these parts. I don';t know where the devil the code that shows how they get entropy from the WDT is. Might be possible to use some other async timer, like the RTC in place of it,

specternecter, 2023-03-17:
I've been trying to comment that out for hours and it somehow just keeps
finding a way to be defined

specternecter, 2023-03-18:
I got in touch with the author of the Crypto library. He did me a solid and traced the problem and ended up disabling the WDT-based noise source for the ATtiny3226. I am now able to get noise from one of the other devices on my board and it works. You know these devices well so I wanted to ask if you have a noise source you would recommend that would allow me to make it work for all the new AVRs?

specternecter, 2023-03-18:
Also a way to determine if its a megaTinyCore chip for selecting the new noise source? They use if(__AVR__) and if(__ESP32__) but is there a way to do if(__MEGATINYCORE__) or easily implement it?

hmeijdam, 2023-03-18:
I think [this](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Defines.md#identifying-core-version) is what you are looking for

specternecter, 2023-03-18:
> I think [this](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Defines.md#identifying-core-version) is what you are looking for

That is actually everything I'm looking for. Thanks!

SpenceKonde, 2023-03-18:
Yup. I'd probably test if __AVR_ARCH__ == 103 for modern tinyAVR or mega0-series. Otherwise it's apparently 102 or 104 depending on flash size (I have made inaccurate statements about this in the past) for Dx/Ex-series parts.

specternecter, 2023-03-18:
Even better. I'll have to look into that. Also, I looked into True Random Number Generation on AVR platforms and really the only good way (using software alone) is by sampling differences in timers. So the Crypto library using watchdog as a noise source actually makes a lot of sense when you want a truly random number in an acceptably short time period.

SpenceKonde, 2023-03-18:
It's really not possible to clean up the ADC readings on a floating pin well enough to squeeze some entropy out of it? Even if you put a big long wire dangling off some otherwise unused pin with analog capability?

specternecter, 2023-03-18:
Well that's the thing. You actually can but it's hackable. If somebody has physical access to the pin they can pull it to ground and your number is no longer "random". Based on my research, watchdog is all-around the most secure way to do it without external components

specternecter, 2023-03-18:
Now that I think of it, ADC on these devices isn't that slow and you could in theory check for any tampering of the pin. If readings were detected as not being sufficiently random, it could just run the random function again. So if grounded, it would never complete. In order to hack it you would have to supply a "random" signal that you actually knew, but that would not be that easy to accomplish. 

I think what I'm going to do is implement watchdog for all devices that use the same timers as the ATtiny3226 but also implement a fallback to use ADC for devices that don't. It wouldn't be the fastest or most secure method but with tamper-detection written into the function it should be a sufficient catch-all fallback.

SpenceKonde, 2023-03-18:
*whistles a  tune while waiting for you to realize that there isn't a WDT interrupt on these*. 

There are a couple of ways you could do the same kind of thing though.... You can set the RTC to generate an interrupt and use that like the WDT (running it from internal low frequency oscillator

You can also get a *completely untrimmed/uncontrolled* "clock" with the CCL. Start CCL block 0 or 2, set to invert it's own output with no sync or filter. (inputs feedback, masked masked, truth = 0x02. That will oscillate with a frequency of around like 80-100 MHz. Route that to an event channel. Then the other even lut,you do the same thing, except inputs are feedback, masked, event, clock source set as in[2],. and the filter enabled. That will divide the clock by 8., Hmm that would still be too fast for a synchronous timer to count on wouldn't it?

Don't worry! Switch the first LUT's in[0] from feedback to event, and route that event channel (which you havethe output of that logic block routed to) to this CCL's input event, and set that as the input instead of feedback. That slows the propagation by about a factor of two. so you have a 4-5 MHz clock that is hence slow enough to tally with a synchronous timer, but which is generated not only fully asynchronously, but also through pathways which are not controlled by the process engineers - at least not carefully. 

Whereas the internal 32k osc used by the RTC and such strives to be reasonably accurate with temperature compensation and such. There ain't no trimming or calibration here, and the oscillation frequency is highly dependant on voltage and temperature! BUT WAIT! There's more! 

Since we used the first timer with an event channel in the loop to slow it down, we don't need to use an even LUT! We can use LUT1 outputting to event channel, and using that as it's input and inverting it, instead of the even lut. Then LUT0 can take the link or event channel as it's input 2, selected as clock, and the feedback input as it's input 0. 

BAM! You now have a version of this which will work ON ALL MODERN AVRS - EVERYTHING WITH A CCL and EVSYS!!!

Now how you take this random asynchronous drifting clock at 4-5 MHz or thereabouts clock and turn that into usable entropy, that's not my department.

specternecter, 2023-03-19:
OMG, this is perfect! I came back here to ask if you thought using the RTC interrupt would be a good idea because yes, I noticed none of these have a watchdog interrupt lol. But I did find that it appears all of them have a TCA and TCB timer so I think "#if defined(TCB0_CNTL) || defined(TCA0_SPLIT_LCNT)" should be fine for detecting if it's a modern AVR that would need to use this method.

specternecter, 2023-03-19:
If I can get a random asynchronous drifting clock I can turn it into usable entropy.  But after looking at it, CCL isn't really my department either lol. Is there any CCL example I could look at that would help me configure this clock? I don't want to mess it up considering all the security depends on this clock working correctly.

SpenceKonde, 2023-03-19:
That's fair, lemme see if I can't throw a 2-LUT /8 clock together, shouldn't take long (I couldn't security my way out of a wet paper bag, but CCL is no problem (because I find it interesting, while I don't find security interesting.

specternecter, 2023-03-19:
See I'm the opposite. I implement security into almost everything I do but
never really found a use for CCL. I do however look forward to seeing how
it works and what it can do

SpenceKonde, 2023-03-19:
Okay. I appear to have a dead 1626 PCB.... which resists all attempts to figure out what's wrong with it. What the hell. 

That's why this took me a ridiculous length of time. 

Two approaches: 
Civilized: 
```
void setup() {
  // Pins enabled for debugging. 
  // Sketch should work without OUTEN being set, even on parts impacted by the LINK input errata.
  VPORTA.DIR |= 0x10; // LUT1, the slowed one comes out here. 
  VPORTA.DIR |= 0x80; // LUT0 comes screaming out of PA7 
  VPORTA.DIR |= 0x04;
  VPORTA.DIR |= 0x02;
  CCL.LUT1CTRLB=0x03; //INSEL0 = 3 (Event A)
  // Nothing in ctrlc.
  CCL.TRUTH0 = 0x01; // if all inputs are masked except IN0, this will invert the input. 
  CCL.LUT0CTRLB = 0x01; // Feedback input0
  CCL.LUT0CTRLC = 0x03; // use output of LUT1 for input 2, the clock.  
  CCL.TRUTH1 = 0x01; // same as before. 
  CCL.LUT1CTRLA = 0x41;  // enable, output to pin (for debugging - 0x01 without output on a pin. 
  CCL.LUT0CTRLA = 0x63; //Enable, with CLK = IN2 and the filter option enabled. Output to pin for debugging.
  // Instead set to 0x23 for no pin output. 
  EVSYS.CHANNEL0 = EVSYS_CHANNEL0_CCL_LUT1_gc; //Set EVSYS to use output of LUT1 as Channel0.
  EVSYS.USERCCLLUT1A = EVSYS_USER_CHANNEL0_gc;
  EVSYS.USERCCLLUT0A = EVSYS_USER_CHANNEL0_gc;
  EVSYS.CHANNEL1 = EVSYS_CHANNEL1_CCL_LUT0_gc;
  EVSYS.USERTCB0COUNT = EVSYS_USER_CHANNEL1_gc;
  TCB0.CTRLB = 0;
  TCB0.CCMP = 1000; /* placeholder - I don't know what you want this to be; thisis the number the timer counts up to, maximum. */
  TCB0.INTCTRL = 1;
  TCB0.INTFLAGS = 3;
  TCB0.CTRLA = 0x0F; //enable with event clock
  // okay, everything should be set! 
  CCL.CTRLA = 0x01; // enable the whole CCL 
  /* You should now have: 
   * VERY high frequency signal coming out of PA4. This is what we are dividing down. 
   * A frequency low enough to use as a clock for TCB0 from PA4, and PA7 should be counting up to
   * Then 
   */
} 
ISR(TCB0_INT_vect) {
  uint8_t temp = TCB0.INTFLAGS;
  if (temp & 1) {
    VPORTA.IN |= 2;
  }
  TCB0.INTFLAGS=temp;
}
void loop() {
}
```
On my chip, I see about 58.5 MHz out of CCL1, generating the high speed output on PA7, then I knock it down to 7.3 by sending using it as a clock for CCL0 (I originally used the LINK input , but then I remembered that that is broken on some parts and since we're already using an event channel to get the signal back to LUT1's input, we can just use that one without using up any more part resources). all we're doing with the filter is effectively prescaling the clock (which we have the LUT1 output pointed to) by 8 (both CCL blocks invert their own input; LUT1 has no filter enabled, so it's operating asynchronously with no clock, while LUT0 is clocked, but with a wacky 58.5 MHz signal that's highly sensitive to the environment. So it's also not synchronized with the system clock). This 7.3 MHz signal at 20 MHz F_CPU, is slow enough to meet the nyquist criteria.. 

But this may not be what you want. You say you want jitter right? That has longer time-scale anomalies (ie, the chip gets warm and it drops from 3.7 kHz to 3.1)  

If what you want is jitter in the timing of the thing clocking TCB, see below. I made a mistake here 
```c
void setup() {
  // Pins enabled for debugging. 
  // Sketch should work without OUTEN being set, even on parts impacted by the LINK input errata.
  //VPORTA.DIR |= 0x10; // LUT0, the slowed one comes out here. 
  VPORTA.DIR |= 0x80; // LUT1 comes screaming out of PA7 
  VPORTA.DIR |= 0x02;
  CCL.LUT1CTRLB=0x03; //INSEL0 = 3 (Event A)
  // Nothing in ctrlc.
  //CCL.TRUTH0 = 0x01; // if all inputs are masked except IN0, this will invert the input. 
  //CCL.LUT0CTRLB = 0x01; // Feedback input0
  //CCL.LUT0CTRLC = 0x03; // use output of LUT1 for input 2, the clock.  
  CCL.TRUTH1 = 0x11; // same as before. 
  CCL.LUT1CTRLA = 0x41;  // enable, output to pin (for debugging - 0x01 without output on a pin. 
[... 11 lines cut ...]
  // okay, everything should be set! 
  CCL.CTRLA = 0x01; // enable the whole CCL 
  /* You should now have: 
   * VERY high frequency signal coming out of PA4. This is what we are dividing down. 
   * A frequency low enough to use as a clock for TCB0 from PA4, and PA7 should be counting up to
   * Then 
   */
} 
ISR(TCB0_INT_vect) {
  uint8_t temp = TCB0.INTFLAGS;
  if (temp & 1) {
    VPORTA.IN |= 2;
  }
  TCB0.INTFLAGS=temp;
}
void loop() {
}
```
This makes WAY MORE jitter. It does not come witthin a mile of meeting the nyquist sampling criteria - so you are getting tons of aliasing, causing not only dramatic jitter, but also an extreme dependance on F_CPU and Temperature, with little apparent rhyme or reason to it. The variation between parts and between models is enouigh that you will likely need some manner of post-processing, but I remember reading that that's something that security folk always do when getting entropy from physical sources. 

Design notes: Used CCL LUT1 and 0 because originally I was planning to clock LUT0 from the LINK input, which goes to the output of the lut 1 position higher than it. But then I realized that wouldn't be as portable because link didn't work withoiut setting OUTEN on some 0/1-series tinyAVRs. Odd-numbered CCLs can only use an event channel to get their own input. Otherwise, feedback refers to the output of either the even numbered CCL connected to the same sequencer, or the output of that sequencer (if it's enabled). 

[... cut ...] ER0 for TCB0, ASYNCUSER2 and 3 to the first event input on LUT0 and LUT9 respectively) - (resuming if i didn't care about 0/1series), and I was using the output of the CCL tat's generating the prescaled output, I could use LUT 2 for the initial clock generation, it's FEEDBACK input would pull from it's output, there would be no need to use an event channel for it, and then CCL1 could be used for the used (with an event channel and to bring it's output around, as well as to the timer. 

But yeah,

specternecter, 2023-03-19:
This is incredible work! I think the second approach would be the best from both a security and portability standpoint. I really appreciate not only the work that went into this, but also the detailed explanation and comments. I can actually visualize what's happening and why. 

Looking at the datasheet, I understand that if OUTEN isn't set, the signal isn't actually sent to the pins. If that's correct and CCL.LUT1CTRLA = 0x01, are PA4 and PA7 still usable while running this entropy clock? Not for anything complex but just being set high or low without effecting the clock?

SpenceKonde, 2023-03-19:
Yeah, they can still generate events if OUTEN isn't set - I just did it like that in the demo so I could easily visually see what was going on, and you could too. The extent to which the pulse lengths jump around is rather impressive. And varies based on every conceivable variable o_O

specternecter, 2023-03-19:
Awesome! I thought that was the case but I had concerns over whether somebody could accidentally mess up the clock and if I needed to somehow lock them out of using those pins. This is all really impressive and it actually appears to allow much more security than the watchdog method of classic AVRs. Word's can't express how much I appreciate this because I probably couldn't have ever implemented something this impressive.

I always wondered what CCL was useful for and this is an incredibly interesting use case. Thanks!

SpenceKonde, 2023-03-19:
This is absolutely not what the CCL is intended for :-P I've used the CCL for more mundane things too: it;s key to a method I have for redirecting WS2812 control signals down 1 of 4 paths with an ATtiny424 (3 non-chainable strings of fairly lights plus daisy chain to the next splitter) I've got a scheme that I think will let me drive 2812's in the background with me only having to keep the 2-byte buffered SPI fifo from running dry, which would dramatically reduce the number of controllers I need to achieve my design goals, you can use it to get a third PWM channel out of TCD0...

just read some messages further up; 
testing for the registers associated with peripherals that are currently universal is not a robust way tyo determine if a part is modern or not. So far everything has TCB's, yes. TCA? in 18-24 months, there will be a part family which has ditched the TCA and has something called a TCE with WEX. Whether that's a good thing or not, I couldn't tell you. Will that be the norm on new parts after that or will the TCA return? No clue. 

Testing __AVR_ARCH__ is the right way unless you also need to not catch the ancestral xMega parts in the net. If that's a concern, I would say test peripherals that are much more fundamental (register names within PORT? I suspect CCL, RTC, SLPCTRL, CPUINT, NVMCTRL and EVSYS are going to be on all modern AVRs. And of course PORTMUIX....

specternecter, 2023-03-19:
It's obvious to me that you're pushing the limits of these devices and it's
impressive.

Yeah I guess I never considered they might replace TCA. I knew it wasn't
the cleanest way but it worked as a simple method i could target with. I
don't think catching xMega would be a necessity. Aside from me using one at
one time, I don't think I've ever seen one used or anticipate any
complaints about Crypto not working on one. But I guess testing for CCL
would probably be the most logical way. Where classic AVRs define
RNG_WATCHDOG to get noise using watchdog, modern AVRs would define RNG_CCL
to get noise from the wild CCL clock. It makes sense to test for CCL. If it
did break something in a modern AVR, I would just have to search for
something specific to that AVR and handle it appropriately. Any differences
that would break it would likely be spread across a relatively large range
of devices. if the device doesn't have CCL, it should break because there's
no way to make the clock anyway. I think I'm going to check for CCL and
handle any issues that may arise from that as they come up, if they come
up. This CCL clock appears to greatly increase security over watchdog (more
testing needed to confirm), and in that case it would be preferable for any
device that has CCL to use it over watchdog.

SpenceKonde, 2023-03-21:
I don't think catching xMega is a concern either. I don't think it was for anyone, xMega is the only product line Microchip hasn't so much as prodded with a ten foot pole since taking over. In fact the only thing they've done involving xMega seems to have been sending the engineers scavenging for designs that they could clean, polish up, sand down some rough spots and remove the tangle of obscure features of dubious utility but undeniable complexity that clung to everything xMega.

SpenceKonde, 2023-03-24:
Great to hear! And I'm glad my self-inverting CCL trick is actually good for *something* after all! 

It's funny how you get better results when you fail to meet the nyquist criteria, but I guess it greatly magnifies the variations.

specternecter, 2023-03-24:
Yeah it appears to be a great method. It took a while to determine what the
sampling frequency of the RTC needs to be set to for a given CPU frequency.
The RTC ticks 32768 times a second and every CPU speed above 4MHz runs the
fastest when the RTC is set to sample CCL every 2 ticks. But at 4MHz and
below, RTC sampling rate needs to be slowed down and to what number varies
based on CPU speed. If you run sampling rate too fast it will severely
impede the AVRs ability to run anything. So I had to run a test at all the
CPU speeds and all the RTC sampling rates to determine the minimum and
fastest sampling rates for each CPU speed. Now while I know that at 10MHz
Crypto can now generate 256 random bits in as little as 250ms, that's
probably not gonna be how fast it ends up running when everything is pushed.

Just because it can run that fast doesn't mean it should. The final specs
will end up being a compromise between speed and amount of randomness.
That's what I'm in the process of determining now. While it is impossible
to actually predict randomness, there are two relatively simple tests that
give us a pretty good idea. The first test is looking for occurrence of
each number in a large set of random numbers. If given a large sample there
shouldn't be numbers that occur too often vs the others. In a perfect world
they'd all occur the same amount of times. Even if they did though, that
doesn't mean it's actually random and that's where the next test comes in.
We have to look at patterns in the number. The human mind is really good at
picking out patters in images. So we have the device generate a bitmap
image using random numbers. If there's any sort of pattern in the numbers
we'll likely be able to see it in the image. These tests need to be run at
different CPU speeds and different sampling rates to determine how best to
set everything up. So while the fastest 10MHz can randomize 256 bits is
250ms, it might turn put that it's best to run it at say 650ms. I hope to
have all this done and pushed by next week. I don't want to just get it
running, but to get it running at the best of its ability, the right way.

specternecter, 2023-04-02:
Modern AVRs are now officially supported in the Crypto library. It takes about 320ms to randomize 256 bits of data at 10MHz and about 3500ms at 1MHz. This is MUCH faster than classic AVRs. I have ran it through many tests with different modern chips and I see no issues in either stability or its ability to generate truly random numbers. I see no patterns in the data or any issues with number diversity. My contact information is listed in the source code so I can fix any issues using the CCL method, should they come up. Thanks for all your help @SpenceKonde, we're now up and running.

SpenceKonde, 2023-04-04:
Has this been merged into the library repo so that I can update the library compatibility page?

specternecter, 2023-04-05:
Yeah it's good to go

SpenceKonde, 2023-04-10:
Great, added to libraries page, closing this, Thanks!

closed 2023-04-10 by SpenceKonde

SpenceKonde, 2023-06-19:
@specternecter - Say, you wouldn't happen to have a function just kicking around that would generate a 16-bit random integer to seed a PRNG with would you?

specternecter, 2023-06-19:
I suppose an easy way to do this woud be to take advantage of the small differences in execution time. Ideally you would want to include some sort of unpredictable value such as reading internal chip temperature or pin noise but a barebones way to do it that would work on all Arduinos would be:

uint16_t simpleSeed(){
  uint16_t tempSeed;
  // flip each bit at a somewhat unpredictable rate using time mod 2 and then XOR the result with time again for some added entropy
  for(uint8_t i = 0; i < 16; i++){
    if((micros()%2)^micros() == 1){
      bitSet(tempSeed, i);
    } else {
      bitClear(tempSeed, i);
    }
  }
  // now do time multiplied by tempSeed and XOR with tempSeed 
  return (micros()*tempSeed)^tempSeed;
}

I tested this with an ESP32C3 running this on startup. You get a very unique number every time its reset

specternecter, 2023-06-19:
I suppose an easy way to do this woud be to take advantage of the small differences in execution time. Ideally you would want to include some sort of unpredictable value such as reading internal chip temperature or pin noise but a barebones way to do it that would work on all Arduinos would be:

uint16_t simpleSeed(){
  uint16_t tempSeed;
  for(uint8_t i = 0; i < 16; i++){
    // flip each bit at a somewhat unpredictabe rate using time mod 2
    if((micros()%2) == 1){
      bitSet(tempSeed, i);
    } else {
      bitClear(tempSeed, i);
    }
  }
  // take the random(ish) number and use it to create random(ish) number of steps
  uint32_t xFactor = (micros()*tempSeed)%24;
  for(uint32_t i = 0; i < xFactor; i++){
    // flip each bit at a somewhat unpredictabe rate using time mod 2
    if(micros()%2 == 1){
      bitSet(tempSeed, i%17); // bit is mod 17 so it goes 0-16 over and over
    } else {
      bitClear(tempSeed, i%17); // bit is mod 17 so it goes 0-16 over and over
    }
    delayMicroseconds(1); // add a little more entropy
  }
  // now take our random(ish) number and XOR with time once more for good measure
  return tempSeed^micros();
}

I ran this on startup a couple hundred times with an ESP32C3 and didn't find a duplicate number

specternecter, 2023-06-19:
As a side note, bitSet(tempSeed, i%17) should really be bitSet(tempSeed, i%16);. I don't see any difference in results but I chose mod 17 to give it a strange cycle.