SpenceKonde/megaTinyCore #927: Error flashing bootloader to 1604

ssarna1, 2023-03-26:
I've made a UPDI programmer like this one: [ATtiny1604 UPDI HV Programmer](https://oshwlab.com/wagiminator/attiny1604-updi-hv-programmer)

And now i'm on programming it, everything is set up as described there [DIY ATtiny HV UPDI Programmer](https://github.com/Dlloydev/jtag2updi/wiki/DIY-ATtiny-HV-UPDI-Programmer)

and this is what i got:(fuses beforehand are verified correctly)
`avrdude: reading input file "/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/bootloaders/hex/optiboot_txyz{bootloader.entrycond}.hex"
avrdude: can't open input file /home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/bootloaders/hex/optiboot_txyz{bootloader.entrycond}.hex: No such file or directory
avrdude: read from file '/home/XXX/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.5/bootloaders/hex/optiboot_txyz{bootloader.entrycond}.hex' failed`

I assume something is not set correctly within Arduino and instead of some part of filename, a variable name "{bootloader.entrycond}"is displayed? It is installed from tarball, i'm on Ubuntu. IDE version is 1.8.19.

SpenceKonde, 2023-03-29:
What does the Arduino tools menu look like?

ssarna1, 2023-03-30:
[image]

SpenceKonde, 2023-03-31:
You're not using the most recent release of the core, or are running some mixed up combination of two releases. Latest released version should have a bootloader entry condition menu and a reset/updi pin config menu for optiboot boards

IIRC while adding that to 2.6.6, I found a few places where the bootloader filename didn't resolve correcty. 

So this should be fixed in 2,6,6. 

Don't use github/2.6.7 version, I busted pwm completely the other night and am still trying to get that sorted out (part of the issue is that not only were users demanding it, I had suddenly come into a use case where *I* needed buffered pwm, so i figured time to implement the pwm pin selection menu. Which is in, except no matter what you select, no pins have PWM.

ssarna1, 2023-04-02:
How do i get this 2.6.6? It's neither in board manager nor in releases on github.

SpenceKonde, 2023-04-02:
Oh fuck!

Okay, well, apparently I failed to actually release it before I rendered the core non-useable in the development version, thinking I'd released a working version.

SpenceKonde, 2023-04-07:
Well yeah it would. 

As long as you didn't care about the new features, I mean. 

There are only 22 bootloader binaries supplied with 2.6.4. 2.6.6 would have shipped with 80, adding the same functionality that DxCore has.

SpenceKonde, 2023-04-07:
I mean 2.6.5 shipped with all 80 too. We just substituted in "entrycond" but the menu sent "entrycondition" or something silly like that.  Though I think as I was looking at it i fixed a bunch of bunk bootloaders. Or maybe that was DxCore. I know one one occasion I was in there for one simple bug, and wound up finding a huge number of bad binaries.

SpenceKonde, 2023-04-20:
Closed - fixed by 2.6.7

closed 2023-04-20 by SpenceKonde