SpenceKonde/megaTinyCore #931: Loading Hex on to ATtiny3217 using Softlog ICP2 Programmer UPDI

braydonfriar, 2023-04-03:
Hey SpenceKonde, is this possible to do? I am able to load the hex, but the device seems to not be functional. When using an ATMEL ICE in Visual Studio, it works fine. I know you specify some programmers to use, but it would be preferable if I would be able to use an industrial grade programmer. Thanks!

Programmer - https://www.microchip.com/en-us/development-tool/tpg100001

SpenceKonde, 2023-04-03:
modern tinyAVR devices use the UPDI upload protocol, the product you linked as far as I can tel supports ISP, JTAG, and PDI, but not UPDI, and looks like it was released prior to the release of modern AVRs, and is marked for phaseout (read: We're gonna stop makin em soon) (probably for this reason).

There aren't many official programmers that do UPDI and fewer still do HV UPDI and I'm unsure if ANY do HV for the new DD-series. 

There;s the MPLAB snap. I own two but couldn't them microchip studio to detect them and I don't know why. 
And i've got 2 ATMEL-Ice (along with 3 curiosity nanos which I use mostly for debugging problems with curiosity nanos, as I dislike almost every design decision made on that board). But I've probably got 3-5 jtag2usb's 2 micro UPD, over half a dozen rigged serial adapters, and I'm going to be getting a SerialUPDI board made with my next order to the board house - gives a serial port and a UPDI port) (also a luxury single port version with a switch to choose UPDI or USART and UPDI, JST XH6 and "microjst 6" (aka molex picoblade, but it looks like jst) for use with space constrained applcation (trust me, you don't want to have to crimp a picoblade terminal, even if you're a master of larger crimps) while pre-wires 6p "microjst" cables are all over)

 which should work too (only for tinyAVR).

braydonfriar, 2023-04-04:
Shoot… I posted the wrong link… I meant to send the link to the G3 version of the ISP2. I’m not as much as an idiot as my original post would indicate. Haha. Which this one can support UPDI. If you have any information on why this one, that supports UPDI, doesn’t seem to be working (but could also be largely based on user error) that would be sweet.  I was able to load the program, but got promoted with a Unverisal Hex Loader that I could have f’d up. The link is below, but I don’t think it is a hyperlink, as I am on my phone.  Sorry!

https://www.microchip.com/en-us/development-tool/TPG100002-G3

I appreciate the information on the other options.  I’ll be looking at your SerialUPDI software to see if that can possibly be a solution.  Is my belief true I can’t use a nano for SerialUPDI to do HV UPDI as the nano can not source 12V.   

Anyway, I appreciate that you have this open source core, as it makes my life a hell of a lot easier.  Thanks!

braydonfriar, 2023-04-04:
[image]

Hey, I think I may have figured it out myself. Turns out I needed to enable the 12V UPDI Programming Entry. Just a question based on the picture above. The picture above shows the ICP software interpret my Config bits for 20 MHZ. Will the core simply override these settings (or interpret them differently)? I'm just a bit worried something funky is going on since there is a power draw difference of around .8 mA depending on if I program it with the Atmel ICE (using Visual Micro) vs the ICP2 G3 programmer (using ICP software).  Thanks for the help.

SpenceKonde, 2023-04-04:
I don't know how visual micro could set the fuses correctly for a programmer which I have provided no instructions on how to communicate with. Thus, if you have 20 MHz selected, burn with that, and have 4 MHz selected in that third party IDE, I would expect it to burn the fuses like you set in the GUI, and if they did not precisely match the ones that the core expected that it would have set, things would be many flavors of fucked up.

For example, if the 20 MHz oscillator is selected, but you compile for 4 MHz, you get 5 MHz, and all timing will run 5 4ths of the normal speed. Because it thinks it set it for 16 MHz so it could divide by 4 to get 4 MHz (there's no /5 prescaler option)

SpenceKonde, 2023-04-04:
I would strongly advise doing test burns to see how it behaves. 

As I've said countless times, I do not have time to maintain three cores run a small business AND LEARN EVERY STUPID THIRD PARTY IDE THAT COMES ALONG with their own idea of "supporting" arduino cores, and figure out how to make them do what they're supposed to do. I have already written out the recipes for compiles and uploads for Arduino IDE. 

Then these other IDEs come along and are all yeah we support arduino cores. NO YOU FUCKING DON'T! DO YOU PARSE THE JSON FILE TO FIND THE REQUIRED VERSIONS OF THE TOOLCHAIN AND UPLOAD TOOLS? **NO!** DO YOU PROVIDE AN ANALOGY OF THE TOOLS SUBMENU SYSTEM? **NO** DO YOU HONOR PLATFORM.TXT? **NO!** It seems that all it takes to say you support arduino cores is to be able to compile C and C++ using AVR-gcc.

braydonfriar, 2023-04-05:
Thanks. I'll be moving this project over to the arduino IDE to ensure that everything goes smoothly. Appreciate the help!

closed 2023-04-05 by braydonfriar