SpenceKonde/megaTinyCore #935: sketch_path seems to be not longer defined in arduino-cli

ArminJo, 2023-04-17:
As you can see [here](https://github.com/Arduino-IRremote/Arduino-IRremote/actions/runs/4723699010/jobs/8379998876#step:3:195), the environment valiable sketch_path sems to be undefined at least in arduino-cli  **Version: 0.32.2** Commit: 2661f5d9 Date: 2023-04-12T09:41:54Z 

[Here ](https://github.com/Arduino-IRremote/Arduino-IRremote/actions/runs/4553455603/jobs/8030040602#step:3:63)it worked
arduino-cli  **Version: 0.31.0** Commit: 940c9457 Date: 2023-02-22T16:15:52Z

SpenceKonde, 2023-04-17:
What do they say we are to use ro replace it?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

ArminJo, 2023-04-17:
I cannot decide, if it is a bug or feature of arduino-cli, but on windows, sketch_path still seems to work.

SpenceKonde, 2023-04-17:
I'd say it's a bug in the linux version of the Arduino CLI - How the hell are you supposed to save the hex file if you don't know where to save it to. Please report to the Arduino team, this is a serious bug which would impair usability and maintainability of all three of my cores.

closed 2023-04-17 by SpenceKonde

ArminJo, 2023-04-17:
Sorry but **sketch_path** is used to store the .lst and .map files!!! 
And in ATTinyCore (and all other cores I use for IRremote CI) it gives at least no errorr 😀

per1234, 2023-09-27:
This bug is now tracked by the Arduino CLI developers at https://github.com/arduino/arduino-cli/issues/2340. There is no need to comment there. I'm only adding this comment to notify affected users that there is a formal bug report about it in the appropriate issue tracker.