SpenceKonde/megaTinyCore #936: Serial Programmer unidentified

aklassen117, 2023-04-17:
I followed the Serial programming instructions, however I am trying to use the CP2102 (its what I currently have in hand). I set the baud rate 57600, and wired according to the schematic you provided. Issue is the ide is saying that a programmer is required to upload.

Any ideas?

SpenceKonde, 2023-04-18:
Did you select one from tools -> Programmers?

SpenceKonde, 2023-04-18:
That I'm afraid is the something went wrong error. Could be anything from bad wiring to wrong port selected to resistors preinstalled on the CP2102 board (I know CH340's usually have them. What's the wiring on PA0 look like, it should be direct or through a series resistor not larger in value than 470 ohms, and there should be nothing capacitive connected to PA0, like an autoreset circuit (that's why my boards have a solder bridge jumper to enable that after bootloading

aklassen117, 2023-04-18:
I have the the ground and 3.3V connected to the CP2102 board. PA0 is directry connected to the RX pin, and the TX pin is connected to PA0 through a 220 resistor. I tried removing the resistor and connecting directly and getting the same result. As far as I could see there is no on board resistor in series with the chip, unless the chip itself has a large input resistance I don't know about.
[image]

SpenceKonde, 2023-04-20:
aaaaah!! No no, the resistor between tx and rx is 4.7k, or replace with a small signal diode with band towards Tx of the adapter. 
That sub-470 figure is what's allowable between PA0 of target and RX), between TX and RX must be 4.7k, unless there;s already resistance in series with TX on the board.

SpenceKonde, 2023-04-20:
Closing issue as it does not represent a bug in the core.

closed 2023-04-20 by SpenceKonde

aklassen117, 2023-04-20:
Right, I started with a diode in place of the resistor (in the photo I uploaded above). When that didn't work I switched it to a 4.7k resistor. When that also failed I slowly lowered the resistance to the point were it was essentially shorted.

I also just connected a CH340 and repeated the process, I am getting the same errors. I am still using the same baud rate 57600, as I am having an issue figuring out how to get it to run at higher speeds.

SpenceKonde, 2023-04-27:
Well, was the diode a small signal shottky diode? It needs to be for that to work. 

One thing I've discovered is that UPDI as a protocol, on the tinyAVRs, is balanced on a knife edge in this sort of configuration. 

The UPDI line's transmit function is the gimped 1/10th drive strength you get whenever using the pin that takes the HV reset pulse as I/O (ie, it's a worthless output - that's why they removed that feature from the more recent parts, I suspect.)

So what you need is:
When TX is asserted, current can flow freely enough to be registered as a low on the target. 
When TX is high, since most adapters use totempole outputs, not open drain + pullup, the target is still able to assert a low, despite the shit pin driver. 

If the resistance is too high or the drop across the diode between RX and TX is too high, the programmer can't pull the target's line low enough (it's fighting a standard pullup I think, like 35k ohm - the problem here isn't overpowering the pullup, but getting the voltage low enough to be below Vdd * 0.3 to be registered as a LOW. 

If the resistance is too low (basically anything under 4k or so) or the diode has horrendous reverse leakage (I've seen only a few diodes with such ghastly reverse leakage that it might be a problem), the target won't be able to pull the UPDI line low. 

If you put it on a digital scope you should be able to clearly see what's going on. Do a single trigger on the falling edge of the UPDI line and adjust the horizontal scale as needed to see multiple bytes being sent. 

Under normal operation you will see both directions - but the highs and lows, and the shape of the curves will be different depending on whether target or programmer is the one talking, and if one of them looks pathological, that's a big hint. 

Some adapters already have a resistor in series with TX - these require a lower resistance, or a diode be used. 

A few adapters may have resistors in series with their RX. That's not good, because the impedance of the target's updi pin is so bloody high on tinyAVR it's of the same scale as that, and combined with the internal pullup on the serial adapter, it forms a three way voltage divider:
```
          serial adptr         Adapter                 target updi pin can be approximately modeled
          pullup                 series res.              as an ideal pin with resistor after it. 
Vcc----/\/\/\/\/\--------/\/\/\/\/\-----------/\/\/\/\/\---------
               ??? k                 ?????                      ~2k @ Vdd = 5V, 3.2k @ Vdd = 3v, 3.6k @ Vdd = 1.8 

```

Also, are you sure you have TX and RX the right way around? This is very easy to mess up because the standard doesn't make clear the nomenclature: If i have a pin marked TX - is that the TX pin of the device itself? Or the RX pin, marked TX because that's what should be connected to TX in the standard serial port use case. 

In the Dx and Ex-series reset and updi were made different pins, and they gave up on making reset an output all together - it's now input only when used as I/~O~ and you put the HV pulse on there for HV programming. That lets them make UPDI into a normal pin with a normal drive strength.