SpenceKonde/megaTinyCore #942: compile errors in 2.6.7

Unreal-Dan, 2023-04-21:
```
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c: In function 'analogWrite':
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1205:13: error: 'set_inven' undeclared (first use in this function); did you mean 'strnlen'?
             set_inven = 1;
             ^~~~~~~~~
             strnlen
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1205:13: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1307:5: error: break statement not within loop or switch
     break;
     ^~~~~
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1317:5: error: case label not within a switch statement
     case NOT_ON_TIMER: /*falls through*/
     ^~~~
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1318:5: error: 'default' label not within a switch statement
     default:
     ^~~~~~~
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1325:7: error: break statement not within loop or switch
       break;
       ^~~~~
In file included from C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\api/ArduinoAPI.h:52:0,
                 from C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\Arduino.h:26,
                 from C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_private.h:32,
                 from C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:25:
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c: At top level:
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\api/Common.h:16:26: error: expected ')' before numeric constant
 #define OUTPUT           1    /* used for pinMode() */
                          ^
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1330:16: note: in expansion of macro 'OUTPUT'
   pinMode(pin, OUTPUT);
                ^~~~~~
C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1331:1: error: expected identifier or '(' before '}' token
 } // end of analogWrite
 ^

```

Forgive me if I did something wrong, seems like there's some compile errors in the lib after updating in arduino.

Is it the way we're using it?

Unreal-Dan, 2023-04-21:
I simply removed the contents of analogWrite to deal with this error, fortunately we don't need it. 

However we are still having a rather annoying issue, it seems that the new version has fixed the "burn bootloader" option when choosing "Attiny3217 w/ optiboot" however after burning the bootloader any attempts to upload to the chip fails.

This happens with both attiny1616 and attiny3217, we can burn the optiboot bootloader but subsequent uploads fail with this incorrect signature error:

```
Sketch uses 11056 bytes (69%) of program storage space. Maximum is 15872 bytes.
Global variables use 258 bytes (12%) of dynamic memory, leaving 1790 bytes for local variables. Maximum is 2048 bytes.
C:\Users\danie\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7/avrdude.conf -v -pattiny1616 -carduino -D -PCOM12 -PCOM12 -b115200 -Uflash:w:C:\Users\danie\AppData\Local\Temp\arduino_build_584031/VortexEngine.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7/avrdude.conf"

         Using Port                    : COM12
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATtiny1616
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   10      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         32   32      0     0     0 0x00 0x00
           flash          0     0     0    0 no      16384   64      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

An error occurred while uploading the sketch
Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: Expected signature for ATtiny1616 is 1E 94 21
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.
```

Notice the part at the end saying:

```
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: Expected signature for ATtiny1616 is 1E 94 21
         Double check chip, or use -F to override this check.
```         

It seems to be that it's retrieving the device signature of the Uno that has Jtag2Updi on it.

If I modify boards.txt to include -F to the avrdude command then it actually uploads to my Uno and I have to re-upload jtag2updi to get it working again.

Are you able to shine any light on this issue?

SpenceKonde, 2023-04-21:
Why are you using a programmer if you've bootloaded with optiboot? That's the expected message because it's trying to upload to the programmer as if it were a 1616 with optiboot. But it's a '328p with jtag2updi loaded on it. Optiboot needs direct serial connection to target.

SpenceKonde, 2023-04-21:
did you get those in the released version? I'm certain I fixed this earlier today before releasing.,

Unreal-Dan, 2023-04-21:
Ah we don't have direct serial to target in this case.

Yes I got it in the released version.

I managed to fix it by doing this:

```
atxy7o.upload.workaround=
atxy7o.upload.tool=avrdude
atxy7o.upload.protocol=jtag2updi
atxy7o.upload.prog_interlock="ERROR: Upload using programmer is not supported on optiboot boards"
atxy7o.upload.extra_params= -Ufuse0:w:0x00:m -Ufuse1:w:0x00:m -Ufuse2:w:0x02:m -Ufuse5:w:0b11000100:m -Ufuse6:w:0x04:m -Ufuse8:w:0x02:m -Ufuse7:w:0x00:m
atxy7o.upload.speed=115200
```

I changed the protocol to jtag2updi, and the extra params so that this command would change from:

```
OPTIBOOT 'UPLOAD' AFTER BURN BOOTLOADER WITHOUT OPTIBOOT:

C:\Users\danie\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude
 -CC:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7/avrdude.conf
 -v
 -pattiny3217
 -cjtag2updi
 -D
 -PCOM12
 -PCOM12
 -b115200
 -Uflash:w:C:\Users\danie\AppData\Local\Temp\arduino_build_584031/VortexEngine.ino.hex:i 

PRODUCED THIS:

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0200
         0x40 != 0x4e
avrdude: verification error; content mismatch
```

After changing the above I received:

```
C:\Users\danie\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude
 -CC:\Users\danie\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7/avrdude.conf
 -v
 -pattiny3217
 -cjtag2updi
 -Ufuse0:w:0x00:m
 -Ufuse1:w:0x00:m
 -Ufuse2:w:0x02:m
 -Ufuse5:w:0b11000100:m
 -Ufuse6:w:0x04:m
 -Ufuse8:w:0x02:m
 -Ufuse7:w:0x00:m
 -PCOM12
 -b115200
 -Uflash:w:C:\Users\danie\AppData\Local\Temp\arduino_build_584031/VortexEngine.ino.hex:i 
```

No more -D and it adds all the fuses and uses -c jtag2updi so that it doesn't target the arduino.

Unreal-Dan, 2023-04-21:
I was able to get Flash.h to include and managed to get a flash buffer working with about 350 bytes (my flash storage class needs some work to work with bigger sizes) but I've definitely got it working. 

But is there anything I can do to improve megaTinyCore? Happy to talk on discord or wherever if I can be any help.

SpenceKonde, 2023-04-23:
I would love to talk via whatever channels are convienient for us. I don't have discord, but we have discussions here and hell I can even do zoom meetings since I got a subscription during the pandemic. 

Yes, optiboot requires serial.

SpenceKonde, 2023-04-23:
Personally I am about 10x as concerned with the compile errors right now, as the checkers run without errors and I am not experiencing compile errors. You're using the released version not the pre-release github version? Beause those look like some of the last bugs I fixed....

steve2211, 2023-04-23:
First of all: Thank you so much for this amazing project!! I have never seen such an amazing degree of documentaion in any project before. And its even fun to read as well.

Now back to buisness... I'm having the same issue with tinyCore version 2.6.7.

`#include <SPI.h>
#include <LoRa.h>

#define ss PIN_PA4
#define rst PIN_PB5
#define dio0 PIN_PC2
#define BAT PIN_PA7
#define SENS_OUT PIN_PB2
#define SENS_IN PIN_PC0

unsigned int Period   = 0xFFFF;

int main() {
  init();
  //setup LoRa transceiver module
  LoRa.setPins(ss, rst, dio0);

  if (!LoRa.begin(866E6)) {
    while (1);
  }
  LoRa.setSyncWord(0x4C); //0x4C = L

  analogReference(INTERNAL1V5);

  init_ADC1();
  analogReference1(EXTERNAL);

  while (1) {
    //0.002995 for INTERNAL1V1
    //0.004085 for INTERNAL1V5
    float vbat = analogRead(BAT) * 0.004085;

    float vsense = analogRead1(SENS_IN);

    String payload = "{";
    payload += "\"VBAT\":" + String(vbat) + ",";
    payload += "\"VSENSE\":" + String(vsense);
    payload += "}";

    // send packet
    LoRa.beginPacket();
    LoRa.print(payload);
    LoRa.endPacket();

    delay(10000);
  }
}`

This code compiles and runs fine on version 2.6.5, but I get the same error as Unreal-Dan on Version 2.6.7. But I am not using Optiboot, jsut SerialUPDI - 230400 baud as Programmer. As soon as I downgrade to 2.6.5 it works again.

Unreal-Dan, 2023-04-24:
> Personally I am about 10x as concerned with the compile errors right now, as the checkers run without errors and I am not experiencing compile errors. You're using the released version not the pre-release github version? Beause those look like some of the last bugs I fixed....

can confirm it was on the 2.6.7 release through arduino, I held off on updating manually because I wanted to use the official release.  

For the record you can see in the paths of the compile errors of my first post, it shows 2.6.7

ladyada, 2023-04-24:
quite odd, im also seeing compilation errors with 2.6.7 - our github CI is failing here (i also had the same on arduino IDE but my IDE setup is a pit of snakes at this time) so im more trusting of the CI failures.
https://github.com/adafruit/Adafruit_seesawPeripheral/pull/11

caternuson, 2023-04-24:
Looks like there's a slight difference in the preproc logic around the declaration of `set_inven`:
https://github.com/SpenceKonde/megaTinyCore/blob/fb260956ab07bd20e4557073ef475b3bbaa7825a/megaavr/cores/megatinycore/wiring_analog.c#L1087-L1089
and the usage:
https://github.com/SpenceKonde/megaTinyCore/blob/fb260956ab07bd20e4557073ef475b3bbaa7825a/megaavr/cores/megatinycore/wiring_analog.c#L1183

SpenceKonde, 2023-04-25:
Well I found one problem, I think....

caternuson, 2023-04-25:
The issue now seems to be rate limit related.

It looks like your commit to change the preprocessor logic worked. But then the merging of PR946 failed:
[image]

It failed due to an HTTP error:
[image]
which generally relates to rate limits.

It might work to simply wait and try running that failed job again.

SpenceKonde, 2023-04-25:
Yeah I have to rerun a few jobs that die for reasons like that every run these days

SpenceKonde, 2023-05-03:
Okay, I think I may have fixed it

SpenceKonde, 2023-05-03:
Okay the CI is now a little better, but it's still not satisfying. Yaml is such a bloody confusing syntax. .

closed 2023-05-13 by SpenceKonde