SpenceKonde/megaTinyCore #943: qTouch

MX682X, 2023-04-23:
my biggest pet peeve so far is the missing ptc support....
So I've looked through the assembly, together with ghidra and wrote a basic library to support selfcap and mutual cap buttons.
It is based on the qTouch Modular library, but I tried to do it differently, as much as reasonable, to avoid getting smited by MC (better safe then sorry). Furthermore I'm not gonna do the other stuff like sliders and 2D surfaces.
I've also realized that qTouch does a lot of pointer arithmethic due to the structure of the struct types. By avoiding that I was able to shrink the library size aswell (about 50% I guess).
Also posting it here for now because:
A. It has research about the PTC registers
B. Trying to keep it off the radar for now.
C. Maybe get some feedback for the user interface
[sketch_apr04a-002.zip](https://github.com/SpenceKonde/AVR_Research/files/11186268/sketch_apr04a-002.zip)

P.S.: Fun fact: The DA has actually two ADCs - the second is inside the PTC module - that's why the Tiny PTC "Takes over ADC0" but this restriction does not appy to DA. Not so fun Fact: As the register layout is different, the DA will need it's own library... (the 328PB is V1, the Tiny 1-Series is V2, the DA is V3)

SpenceKonde, 2023-05-12:
DA will be getting some help from Microchip on this it sounds like over the summer. I'm debating what to do with the libraries - do I include them with the core? I feel like I almost already include too many libraries as part of the core. The DA library they'd prefer remain in their repo. 

Will be fun to see how different the APIs are (I predict someone reading the API doc would be hard pressed to realize they were the same peripheral - like the two Flash.h's) and decide what to do about it. Changing assignee to you since it sounds like you plan to take the next move and I use this to plan what to prioritize

SpenceKonde, 2023-05-12:
Oh - and there has *never* been a library I merged into this core from someone else that has had less than like 75% of the docs (re)written by me. In the cases where the writing style of the code author and myself is very different, it's pretty easy to tell who wrote which parts I think......

MX682X, 2023-05-12:
It's nice to read that MC is doing something, because that will cetainly help to figure out some registers.
Except if it ends up as a wrapper for qTouch, because then there will be bloat, for example, it will have to bitwise OR uint64_t variables for the parts with more then 32 ptc pins to figure out the pin function overwrite register (at least that's what I think it is). And this will ultimately lead to me having to figure out the register map of the DA's, which, btw., seem to be similar to the SAMs, based on my notes and FreeTouch. It wouldn't be the only thing even, as they have also seem to have ported the internal VREG for the CPU from the SAMs to the Dx-Series. 
Anyway, in conlclusion, it's better to wait for what MC will deliver, but based on what I have already seen code wise from them, I don't expect much AVR-nice code

MX682X, 2023-05-21:
> Based on qtouch? Were you thinking we had a prayer of it not being based on the qtouch libraries? Whatever you've been smoking, pass it this way.

There is a saying here: The ope dies last...

> BTW - if you are intending to merge your evnetual library to both cores, make sure you're merging the same code and just different #ifdefs active - I like to be able to do merges between the files in the extras directory , not have to make changes to two files that I can't get a useful diff between. You'll notice many files are identical btween mTC and DxC.

Actually, i've been thinking about seperating the chip specific stuff like pinouts to two different files, but include them based on #ifdefs. similar to the io.h files and keep the actual code free of that for better readability.

Anyway, If you're intrested, I've opened up a repository already, but there are probably still some bugs, so I'm gonna need a bit more time. Get the examples ready, test the examples, expand the docs, etc. (in fact I was even working on making it more DA compatible) https://github.com/MX682X/ptc_touch

SpenceKonde, 2023-06-25:
I'm going to move this to a discussion, as I don't think this is an actionable issue for me at the moemnt, so it should not be on my action item list.