SpenceKonde/megaTinyCore #956: Compile issue in ATtiny3216

rajkunwor9, 2023-05-15:
Hi
I saw following error while compiling in Adruino compiler 1.1.8.13,
Can anyone help me please?
.......................

Compiling library "megaTinyCore"
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\libraries\megaTinyCore\ADCErrors.cpp.o
Compiling core...
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\wiring_pulse.S.o
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\WInterrupts.c.o
"C:\\Users\\rkunwor\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-gcc" -c -g -Os -Wall -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mrelax -Werror=implicit-function-declaration -Wundef -mmcu=attiny3216 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10813 -DARDUINO_AVR_ATtiny3216 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.7.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=7UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy6 "-IC:\\Users\\rkunwor\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.7\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\rkunwor\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.7\\cores\\megatinycore" "-IC:\\Users\\rkunwor\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.7\\variants\\txy6" "C:\\Users\\rkunwor\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.7\\cores\\megatinycore\\wiring_analog.c" -o "C:\\Users\\rkunwor\\AppData\\Local\\Temp\\arduino_build_283759\\core\\wiring_analog.c.o"
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\WInterrupts_PB.c.o
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\WInterrupts_PA.c.o
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\WInterrupts_PC.c.o
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\hooks.c.o
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\wiring_pulse.c.o
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\wiring_shift.c.o
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\wiring.c.o
Using previously compiled file: C:\Users\rkunwor\AppData\Local\Temp\arduino_build_283759\core\wiring_digital.c.o
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c: In function 'analogWrite':
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1205:13: error: 'set_inven' undeclared (first use in this function); did you mean 'strnlen'?
             set_inven = 1;
             ^~~~~~~~~
             strnlen
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1205:13: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1307:5: error: break statement not within loop or switch
     break;
     ^~~~~
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1317:5: error: case label not within a switch statement
     case NOT_ON_TIMER: /*falls through*/
     ^~~~
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1318:5: error: 'default' label not within a switch statement
     default:
     ^~~~~~~
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1325:7: error: break statement not within loop or switch
       break;
       ^~~~~
In file included from C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\api/ArduinoAPI.h:52:0,
                 from C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\Arduino.h:26,
                 from C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_private.h:32,
                 from C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:25:
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c: At top level:
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\api/Common.h:16:26: error: expected ')' before numeric constant
 #define OUTPUT           1    /* used for pinMode() */
                          ^
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1330:16: note: in expansion of macro 'OUTPUT'
   pinMode(pin, OUTPUT);
                ^~~~~~
C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\cores\megatinycore\wiring_analog.c:1331:1: error: expected identifier or '(' before '}' token
 } // end of analogWrite
 ^
Using library SoftwareSerial at version 1.0.1 in folder: C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\libraries\SoftwareSerial 
Using library digitalWriteFast at version 1.1.0 in folder: C:\Users\rkunwor\OneDrive - AIR AFFAIRS (AUSTRALIA) PTY. LTD\Documents\Arduino\libraries\digitalWriteFast 
Using library megaTinyCore at version 2.5.12 in folder: C:\Users\rkunwor\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.7\libraries\megaTinyCore 
exit status 1
Error compiling for board ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406.

hmeijdam, 2023-05-15:
If you share your code, including reference to the libraries you include, people may try to run that through their own IDE to see what happens then.

rajkunwor9, 2023-05-16:
Thanks,

But I manually coped code from here and paste to the library file, then it compiled. It seems new version 2.6.5 does not replace the old library files.

If new firmware did not work as expected I will come here to discuss again.

Once again, thanks a lot for your attention.

SpenceKonde, 2023-05-16:
Already fixed in github and reported 7 times since. I will do a release as soon as I have the other compile issues fixed, but I have 3 emails to write and three orders to ship, just like I did at the start of the weekend, and the most important email has still not been completed, and I need to finish all of those tasks before I allow myself to do anything fun like look at code.

closed 2023-05-16 by SpenceKonde

rajkunwor9, 2023-05-16:
Thanks a lot for your contribution on it. I hope your latest fix will work like as previous version was worked well.