SpenceKonde/megaTinyCore #961: [BUG]: wong PWM frequency on internal tuned clocks on attiny402

okawo80085, 2023-05-27:
Noticed an issue where attiny402s flashed with internal tuned clocks would generate PWM with 89hz frequency instead of 50hz with some servos (i have a really big batch of sg90 clones and about 10% of them were unusable cuz of this issue)

Weird thing is that setting the internal clock to the same frequency but untuned generates PWM with correct frequency and all servos work without issue.

Maybe the TCA timer is setup incorrectly for PWM when the chip is flashed with internally tuned clocks?

SpenceKonde, 2023-05-27:
Are you using the included servo library? Because that doesn't depend on a TCA, it uses a TCB... (one of the design goals of the rewrite of that ages ago was specifically removing the dependance on the TCA clock for Servo, same was done for tone); it's now implemented as using the TCB clocked from system clock divided by 2.

I am inclined to suspect that the root of the problem is that the system is not operating at the clock speed you think it is. Possibly something went badly wrong during the tuning process and it now has bad values stored? 

Tuning does not improve accuracy of the clock when used at room temperature - the factory cal has already picked the closest calibration setting at room temperature, and he oscillator has very little voltage dependance. Tuning helps when you want to overclock the chip for some reason or if you are running at temperatures far above or below room temperature, such that you need to correct for that.

okawo80085, 2023-05-27:
Yeah i was using the included servo library, and i think you're right about something going wrong with clock tuning, the PWM jumps from 50hz to 89hz (or 88.39hz sometimes) when flashed with a tuned clock, the thing that im not sure about is why it affects only some servos and not the others.

hmeijdam, 2023-05-28:
I used those cheap SG90's (purple label) also and noticed that while the sticker is similar, The inside can vary. Metal gear, mixed gear and even had some that I could use as a thermometer, as the midpoint shifted if they became warmer.
If I were in your situation, I would open a good and a bad one and compare the PCB's

okawo80085, 2023-05-28:
I'll try that next time i find a some defective ones

SpenceKonde, 2023-06-04:
What does
```
Serial.printHex(CLKCTRL.OSC20MCALIBA);
```
print when the problem is manifesting?

okawo80085, 2023-06-04:
Well... i can't really tell you because all of the times it occurred was on boards that don't have USB or serial outputs and i can't really run them off the Serial2UPDI adapter

SpenceKonde, 2023-06-13:
I am very confused as to how this bizarre situation has come about where you don't have any of the boards that are exhibiting the bug you are reporting. To fix a bug that only impacts some devices, you need to "capture" a malfunctioning device and examine it. 

I suspect the devices are improperly tuned. If I had a board that exhibited the problem in my hand, I'd throw it on the hotplate to take the chip off, mount the chip on a basic breakoout. Then I'd readthe fuses,  If the issue., as I suspect, is tuning, that would make it possible to prove that theory - printing anyhthing to serialwould get gibberish only when a tuned speed was selected. , confirming improper tuning, then you could upload a sketch to dump the contents of the userrow and the SIGROW (not just the first 3 bytes - the wholer 64 bytes of the sigrow, the values I need are near the middle, and which specific ones I need depend on the values you read from the fuses. 

So yeah - that is what is needed to debug this issue - I would be able to tell within like 2 minutes of seeing those three pieces of information (sigrow dump, userrow dump, fuses) whether my theory is correct, and also speculate about how it occurred. s(oh, and what is the target F_CPU? what is the nominal clock speed? 

if you're using tuning to overclock or underclock, particularly to an extreme degree, did you verify that the USERROW contains tuning for the speed you hope to use., not the error code indicating that the the tuning sketch determined that that specimen doesn't work

SpenceKonde, 2023-06-24:
No response, but presently it appears to be some sort of tuning issue, moving this to discussion unless we can show an issue with the core - in any event, without more information from OP there isn't really anything I can do here.