SpenceKonde/megaTinyCore #968: IDE installation url no longer working?

ladyada, 2023-06-07:
hiya 
http://drazzy.com/package_drazzy.com_index.json
is no longer reachable, is there a new URL? you may want to 'host' the package from github - let me know if you need help doing so 
thanks :)

SpenceKonde, 2023-06-08:
We are working on resolving this. 

The filesystem on my webserver became corrupted - the webroot was /var/www/html

Now, it thinks /var/www/html is a file, not a directory. Other data is missing. The system does not run anymore. It's a VM on AWS, so it's just a flush and redeploy, but then I have to reconfigure it. (unfortunately this came at a rather inconvenient time). Rebuilt one will have HTTPS enabled as people have been bitching at me for years about, (What, do they have malicious actors tapping their network connections and using targeted MITM attacks against them to get malware onto their computer via third party hardware packages? At that point, either they have already lost to whomever they had been hoping to evade, or they are [experiencing paranoid delusions. I consider the latter to be more plausible. (link: this song plays in my head when I get complaints about the lack of https)](https://www.youtube.com/watch?v=x9AIbtF9Sv4), either way https isn't going to help). But anyway once a new webserver with a working file system is deployed, and I've figured up how I set up the symlinks and the method I used to get the updated files onto the systems. I'm annoyed, I haven't fucked with apache in ages. 

But I am not going to host the board manager json file on github. I have domain names, and I'm not some sort of scrub who's too cheap or too broke keep a webserver spinning to serve the json file. The only board manager package I host on github is the one that is used by the CI, which also runs from github, and is likely cheaper for them to keep it in house, and it's cheaper for me because the number of toolchain installations that result from one round of CI is unreal. Every other platform is hosted on an AWS S3 bucket. If the cost to be was 2 orders of magnitude higher, then I'd have a problem. 

I'm working on a better tool for autocompiling sketches en masse for CI. What I'd really like is something i can easily make into a chart of pass/fail, and for pass before and pass after, the flash and sram used on the before and the after (and because we'd be pulling the data back to the same place every time, we wouldn't have to worry about yesterdays results until we we at the very endsince I would be  retaining the results. You'd parse and distill the records you got  append to the master csv file and then easily graph execution time as a function of git commit. probably wouldn't even be hard to generate a list of sketches that appear to have experienced a flash size or sram size regression; That way when it found that some closets contained skeletons (warnings, anomalous flash usage, etc) they were marked for further investigation, instead of going unnoticed unless there were errors in the same sketch.  

What if instead of just verify, I generated the .hex, .map. and .lst files, , save them all, ending up with a huge number of files, tar and bzip them, along with the console output) and *somehow* get that out of there and onto my system, where I could extract them But that doesn't change the fact that the CI script is itself sinfully inefficient. It generates an obscene number of installations of the core, far more than need to happen, and takes far longer than it needs to. Definitely not made for being used the way I use it. 

So - there are really 2 problems with the current CI - it's inefficient, and hence results in very heavy traffic on the targets of the site and there's no way to use it to spot regressions that manifest as larger code size. Well, and like everyone else. it depended on a webserver that just put a bullet through it's harddrive. ~_~

SpenceKonde, 2023-06-09:
And I seem to be having an inexplicably hard time connecting to an amazon linux instance in AWS EC2 even with a freshly generated KeyPair, which is really the only roadblock to getting this server up and running again. But it's real hard to execute commands when I can't log in. I'm not sure quite what is being misconfigured.

SpenceKonde, 2023-06-10:
this is fixed now. Still no SSL, but I'm very close.

closed 2023-06-10 by SpenceKonde

ladyada, 2023-06-13:
hiya sorry for the delay in replying - thank you for the fix!!