SpenceKonde/megaTinyCore #978: Not an Issue but a question

mrWheel, 2023-06-24:
@Spence,

Just found out that your great DxCore is available for PlatformIO!! Fantastic!

Is there a change you will make the megaTinyCore also available for PlatformIO??

Highest Regards!

SpenceKonde, 2023-06-24:
DxCore is now available with platformIO? Cool, which version are they giving you guys? 

I thought megaTinyCore had been available for plaformIO for ages.

mrWheel, 2023-06-25:
Hi Spence,

I totally misread this..!

I was triggered by your "`platformio.ini`" template in the DxCore repo and just assumed that it uses the DxCode core but in stead it uses the "`atmelmegaavr`" platform. 

And now I found out that f.i. the "`ATtiny3216`" uses the same "`atmelmegaavr`" core and is also supported by platformIO!

What I don't understand is what part of the "`atmelmegaavr`" core is from your hand and what is not ... (and whether I can use your great and extensive documentation)!

(If it is your's) could you place a template "`platformio.ini`" file in your "`megaTinyCore`" repo?

But it is great I can now program my favourite MPU's with platformIO!

Thanks

SpenceKonde, 2023-06-25:
"megaavr" is the idiotic name that arduino picked to refer to all modern AVRs. It's a stupid stupid name, but it has to be used because they set the precedent. 

Even though the manufacturer calls anything with atmega in the name "megaAVR", So there are many parts that are megavr by Arduino's definition (this is what I call modern AVRs) that aren't megaavr by Microchip's definition, and the vast majority of "megaavr" devices by the manufacturers definition are classic AVRs not modern ones (except the manufacturer hasn't introduced official terms to refer to the types of parts, though classic/modern is gaining traction.), and hence do not use the megaavr architecture. 

What platformio does is w/regards to cores is, the gods of platform IO will periodically update the version of my cores (and other cores) that they include. They cannot be petitioned with prayer or sacrifice - they do this only when and how they see fit. I can sometimes get them to make the safe choice by warning them "Go with the last 2.4.x version until we're a few versions into 2,5,x" or something like that. 

(The final patch version of a given minor version is generally more stable than the initial release of a new minor version, though the core will generally have become more stable than the previous final release before the final release of the current minor version)

It is possible (though as I don;'t use platform IO I don't know how) to update the version of the core to the latest one. 

I don't know what a platform.ini file is, but if you'll tell me what it should be I'm happy to add it.

SpenceKonde, 2023-06-25:
The architectures are simple containers for cores for parts of a given architecture, which can generally use the same libraries because the architecture is very similar. 
atmelmegaavr or whatever isn;'t a core - DxCore, megaTinyCore, and MegaCoreX are cores (that is, they define their own core implementation of the Arduino API functions), atmelatmegaavr is an architecture. Architecture matters for finding platform.txts and board.txts and libraries that are compatible, but that's it - it's just a name, and used only for those purposes.  The platformio atmelmegaavr package appears to be (likely a copy each of) of mTC, DxC, and MCX)

But yeah I', happy to put in more platform.ini 
I just don;'t know jack shit about what should be in it!

mrWheel, 2023-06-25:
Thanks for explaining!

Although I am confused by you stating “you don’t know what a planform.ini file is” … For a AVR128DB32 project I’m working on I use [this](https://github.com/SpenceKonde/DxCore/blob/master/PlatformIO.md) example from your repo:

https://github.com/SpenceKonde/DxCore/blob/master/PlatformIO.md

You should definitely investigate time in Visual Code and platformIO. It is, in my opinion, a better IDE than the Arduino IDE. The latter is easy to use for novice users but the preprocessing in combining all code files in to one and than generate prototypes for every function makes you lazy.. but who am I.

SpenceKonde, 2023-06-26:
And neither do I :-) someone else made that file for me.

ieb, 2023-07-04:
This is what I do.  Hope it helps someone and leaves @SpenceKonde free to do more important things ( ;) )

I've been using for about 2 years with no problems taking the most recent fixes. I don't use VSCode as it uses far too much compute and runs incredibly slowly on my ancient hardware. I do use either SublimeText or vi as my IDE. 

I don't use the distribution from PlatformIO, but I do use pio to build.
I clone this repo, checkout the tag (eg 2.6.8) I want or stay at HEAD.

eg
        cd ~/src
        git clone https://github.com/SpenceKonde/megaTinyCore.git
        git checkout 2.6.8

        mkdir ~/.platformio/packages/framework-arduino-megaavr-megatinycore@2.6.8
        cd ~/.platformio/packages/framework-arduino-megaavr-megatinycore@2.6.8
        ln -s ~/src/megaTinyCore/megaavr/* .
        cat << EOF > package.json
        {
          "name": "framework-arduino-megaavr-megatinycore",
          "version": "2.6.8",
          "description": "Arduino Wiring-based Framework for Microchip tinyAVR 0-series and 1-series chips (megaTinyCore)",
          "keywords": [
            "framework",
            "arduino",
            "microchip",
            "tinyAVR"
          ],
          "repository": {
            "type": "git",
            "url": "https://github.com/SpenceKonde/megaTinyCore"
          }
        }
        EOF
        
        
 In the platformio.ini file I add the following to select the platform packages.
 
         platform_packages = 
             platformio/framework-arduino-megaavr-megatinycore@2.6.8

And any uploader flags needed, eg https://github.com/ieb/CanGNSS/blob/main/platformio.ini

SpenceKonde, 2023-07-24:
Did you hear a whooshing noise? that was the sound of that going over my head. I literally only know about platform IO:
It uses the avr-gcc toolchain when compiling for avr. 
There is a platform.ini file or files somewhere, and that file is important. 
The compiler error output is the same as from the Arduino IDE, so at least I can tell people what flag their configuration isn't passing that it should be....