SpenceKonde/megaTinyCore #984: Neopixel doesn't work at 10MHz

grandaspanna, 2023-07-01:
I've tried it on two different 1614-based custom boards, but the tinyNeoPixel functions don't seem to work if I select 10MHz clock. Using the "strandtest" example on PB3. Same code works if I choose 20MHz. I've also tried 8MHz successfully.

This is using 2.6.7. I had previously been using 2.5.11 which "kinda" worked at 10MHz, but pixels with black (0,0,0) were lighting up faintly.

SpenceKonde, 2023-07-05:
> This is using 2.6.7.

Per changelog, 2.6.8:
>* Bugfix: Fix issues with generating universally valid WS2812 signals with tinyNeoPixel

closed 2023-07-05 by SpenceKonde

grandaspanna, 2023-07-05:
Thanks. I did the update via boards manager just before I posted, but it pulled through 2.6.7. I'll double check things on this end.