SpenceKonde/megaTinyCore #985: 16 bit PWM problem on attiny3226

elektroboard, 2023-07-06:
Hi.
I use Attiny3226 on megatinycore. I want to 16 pwm output from PA4 pin. But I am not success it?
Please take sample sets.
Thanks.

SpenceKonde, 2023-07-06:
Well since the core doesn't support 16-bit pwm, you would have to write
code to implement that - so I'd really need to see thst code

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

elektroboard, 2023-07-07:
Hi.
Here my code. But there is **Compilation error: 'PORTMUX_TCA0_PORTA_gc' was not declared in this scope**

original codes from:https://github.com/microchip-pic-avr-examples/atmega4809-getting-started-with-tca-studio/blob/master/Generating_a_Dual-slope_PWM_Signal/main.c

#include <Arduino.h>

#define PERIOD_EXAMPLE_VALUE        (0x01A0)
#define DUTY_CYCLE_EXAMPLE_VALUE    (0x00D0)

void TCA0_init(void);
void PORT_init(void);

void TCA0_init(void)
{
    // Set waveform output on PORT A
    PORTMUX.TCAROUTEA = PORTMUX_TCA0_PORTA_gc;
    
    TCA0.SINGLE.CTRLB = TCA_SINGLE_CMP0EN_bm | TCA_SINGLE_WGMODE_DSBOTTOM_gc;
    
    // Disable event counting
    TCA0.SINGLE.EVCTRL &= ~(TCA_SINGLE_CNTEI_bm);
    
    // Set PWM frequency and duty cycle (50%)
    TCA0.SINGLE.PERBUF = PERIOD_EXAMPLE_VALUE;
    TCA0.SINGLE.CMP0BUF = DUTY_CYCLE_EXAMPLE_VALUE;
    
    TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV4_gc | TCA_SINGLE_ENABLE_bm;
}

void PORT_init(void)
{
    pinMode(PIN_PA4, OUTPUT); // Set pin 4 of PORT A as output
}

void setup()
{
    PORT_init();
    
    TCA0_init();
}

void loop()
{
    // Main loop
}

hmeijdam, 2023-07-07:
I think you are trying an example written for the Atmega4809 for an attiny3226
Those are quite different MCU's

elektroboard, 2023-07-07:
Have do you a sample code for attiny3226

SpenceKonde, 2023-07-07:
```
Hi.
Here my code. But there is Compilation error: 'PORTMUX_TCA0_PORTA_gc' was not declared in this scope

original codes from:https://github.com/microchip-pic-avr-examples/atmega4809-getting-started-with-tca-studio/blob/master/Generating_a_Dual-slope_PWM_Signal/main.c

`#include <Arduino.h>

#define PERIOD_EXAMPLE_VALUE (0x01A0)
#define DUTY_CYCLE_EXAMPLE_VALUE (0x00D0)

{
pinMode(PIN_PA4, OUTPUT); // Set pin 0 of PORT A as output
}

void setup()
{
PORT_init();

TCA0_init();
}

void loop()
{
// Main loop
}
`
```
Only issue theree is that portmux is being manipulated like it was a Dx, Review the portmux section of th4e tinyAVR datasheet and do it the correct way and you should be good. PORTMUX for TCA on tinyAVR any Dx/Ex/mega0 could hardly be more differerent. 

No there is no option that will move all the TCA pins to a given port. On tinyAVR, vs DxCore/etc, tinyAVR is both more and less flexible. On the downside, at ,most each pin can have only 1 of 2 outputs. On the plus side they get assigned per channel - and not assigned as the whole port. .

Dx/Ea: PORTMUX binds each WO to a pin in the same port. 
TinyAVR: PORTMUX gives you 2 choices per pin.

I also am unsure of whether all of the

elektroboard, 2023-07-07:
Hi. I have a question at
https://github.com/SpenceKonde/megaTinyCore/issues/145.
Please answer it.
Thanks.

Spence Konde (aka Dr. Azzy) ***@***.***>, 7 Tem 2023 Cum,
12:40 tarihinde şunu yazdı:

> Hi.
> Here my code. But there is Compilation error: 'PORTMUX_TCA0_PORTA_gc' was not declared in this scope
>
> original codes [...]

SpenceKonde, 2023-07-07:
Please do not post questions in long-since closed issues, I'm not going to be able to keep track of them..

SpenceKonde, 2023-07-15:
This page has been updated to reflect the trivial differences between the process of getting 16-bit PWM out of a Dx and tinyAVR. https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/TakingOverTCA0.md

Note that no matter what, you do need to write the code that writes 16-bit values to the compare registers and that stuff - that is not part of the Arduino API nor is it one of our API extensions - we stop at letting you tell the core to pretend the timer doesn't exist so you can use it unimpeded

SpenceKonde, 2023-07-15:
OH and there;s no way to get 16 bit pwm out if PA4, that's WO5, WO3,4,5 are split mode only, hence 8 bit only. Available pins tht could be made to output 16bit pwm are PB0/PB3, PB1/PB4, and PB2/PB5. See chapter 21 of the datasheet (TCA) and table 3.1 I/O multiplexing and condideratons

elektroboard, 2023-07-15:
Thanks for the tip!

Spence Konde (aka Dr. Azzy) ***@***.***>, 15 Tem 2023 Cmt,
18:03 tarihinde şunu yazdı:

> OH and there;s no way to get 16 bit pwm out if PA4, that's WO5, WO3,4,5
> are split mode only, hence 8 bit only. Available pins tht could be made to
> output 16bit pwm are PB0/PB3, PB1/PB4, and PB2/PB5. See chapter 21 of the
> datasheet (TCA) and table 3.1 I/O multiplexing and condideratons

closed 2023-07-17 by SpenceKonde