SpenceKonde/megaTinyCore #990: frequancy measurement for Attiny3226

elektroboard, 2023-07-24:
Hello.

I want to measure the frequency of a signal applied to port PA5 (or any other port) with Attiny3226. However, I was not successful. There is an example below but it doesn't work, I think I'm doing something wrong. Can you help me?

#include <Event.h>
int width;

ISR(TCB0_INT_vect){
 // TCB0.EVCTRL ^= TCB_EDGE_bm; // we want to listen for both edges, flip trigger
    width = TCB0.CNT;
    TCB0.INTFLAGS = TCB_CAPT_bm;
}

void setup() {
  Serial.begin(9600);
  Serial.println("RESTART");

 // EVSYS.ASYNCCH0 = EVSYS_ASYNCCH0_PORTA_PIN4_gc;    // asynchronous event from PA7
 // EVSYS.ASYNCUSER0 = EVSYS_ASYNCUSER0_ASYNCCH0_gc;
  // TCB0.CTRLB=0b10111011;
  pinMode(PIN_PA5,INPUT);
 
   
  //TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV256_gc | 1 << TCA_SINGLE_ENABLE_bp;

  TCB0.INTCTRL = TCB_CAPT_bm;//**
  TCB0.EVCTRL = TCB_CAPTEI_bm;//
  TCB0.CTRLB = TCB_CNTMODE_FRQ_gc;//
  TCB0.CTRLA = TCB_CLKSEL_CLKTCA_gc 
              | TCB_ENABLE_bm; //use clock of TCA0
  TCB0.EVCTRL |= (1 << TCB_CAPTEI_bp    /* Event Input Enable: enabled */
				  | 1 << TCB_EDGE_bp    //Negative Edge : Initialize,Positive Edge : Capture = count, interrupt
				  | 1 << TCB_FILTER_bp); /* Input Capture Noise Cancellation Filter: enabled */
	TCB0.CTRLA |= (TCB_CLKSEL_CLKDIV1_gc  /* CLK_PER/2 */
					| 1 << TCB_ENABLE_bp   /* Enable: enabled */
					| 1 << TCB_RUNSTDBY_bp//); /* Run Standby: enabled */
				  | 0 << TCB_SYNCUPD_bp); /* Synchronize Update: disabled */
  sei();
}

void loop() {
  
  Serial.print("frequency = ");
  Serial.print(4*78125/width); // 20 MHZ AND CLOCK/256
  Serial.print(" Hz, ");
  Serial.print(4687500/width);
  Serial.println(" rpm");
  delay(1000);
}

hmeijdam, 2023-07-24:
I would start with changing:

`int width;`

to 

`volatile int width;`

SpenceKonde, 2023-07-24:
The big problems I see are: 
1. You are using frequency measurement mode, but you appear to be trying to get pulse widths. I think your commented out plan with EDGE getting flipped back and forth is probably a better approach. You need to store more than that though - you've gotta track the value that was in CNT cause it doesn't reset in this mode subtract previous cnt from current cnt as uint16_t, and rollover or not, as long as the event is less than 2^16 ticks long you're fine. You can double that maximum on the 2-series by then casting to a uint32_t, then checking and clearing the TCB OVF flag, then adding 2^16 that result if OVF wasn't set. 
2. You didn't set up the EVSYS, every line that might have done that was 1. commented out, and 2. for the 1-series
```c
  EVSYS.CHANNEL0 = EVSYS_CHANNEL0_PORTA_PIN5_gc;
  EVSYS.USERTCB0CAPT = EVSYS_USER_CHANNEL0_gc;
```
Actually, the new style names work on 0/1-series too on mTC.
CHANNEL0 and 1 are sync channels, channel 2-5 are ASYNC, and you still do have to keep that in mind, but porting code is a lot less painful when the names have the same format, and probably the same name (though the selection of generators available to each channel is not the same alas).

3. variables modified in an ISR and read outside of it must be declared volatile. Variables read within an ISR, and written from outside the ISR, if larger than 1 byte, must disable interrupts while writing them outside the ISR, unless the larger-than-1-byte thing is an SFR (special function register), where the hardware ensures this (see "Accessing 16-bit variables" in the datasheet - though note that writing normal C code, you can write a 16-bit value to a 16-bit SFR without worrying about it writing the bytes in the wrong order, it doesn't do that. The hardware also synchronizes them so they appear in the actual register on the same clock cycle. (sidenotes a and b are related, but are not at issue here, just somethings you should be aware of. 
  a.  Note that in the TCA and TCB, depending on mode, conversion of certain registers between a pair of 8 bit registers and a single 16-bit one can occur - or should. It works on the TCA's 5 (?) registers when in split mode, they become 8-bit, but the TCB's CCMP register is supposed to be split into 8-bit registers by 8-bit PWM mode. That doesn't happen - it remains 16-bit in PWM mode, and so you need to write CCMPL followed by CCMPH any time you want to change either value, the change takes effect when CCMPH is written. Or use a union to 
  b. 16-bit SFR writes usually do not need to be guarded by disabling interrupts because of the "writing to 16-bit registers" mechanism, but it does still need to be guarded with disabling interrupts in the case that an interrupt in the program ALSO writes to any 16-bit or larger register on the same instance of the same peripheral.if there is interrupt code that reads or writes to *registers larger than 8 bits* on the *same instance of the same peripheral* 
5. You need to turn off interrupts when configuring the interrupts on timers. You seem to know that, since you put an sei at the end.... but where is it's cli()? Interrupts are enabled when setup starts (I think it's the last thing that happens before we call setup() -  (otherwise until you'd done sei(), millis wouldn't tick up)
6. There is no rhyme nor reason to your organization of writes to the peripheral registers. But this is a place where that shouldn't be the case - you need to do that with some care to ensure that nothing is changed when it could cause a bunk interrupt: First disable the timer(s) you're working on. then set up everything except for CTRLA, and the INTCTRL and INTFLAGS. Then disable interrupts, set intctrl, set CTRLA (to the value you calculated already), and TCxn.INTFLAGS = TCxx.INTFLAGS (to clear all flags that may have gotten spuriously set. I think you only need 6 lines guarded with the cli/sei. 

7. This is the year 2023, and this is a modern AVR. Not the problem at hand, but.... you don't still need to default 9600 baud. You are more likely to cause problems than prevent them by using 9600 baud (due to the buffer filling and then the sketch execution speed being limited to the speed of it'sserial output). (I've been trying to set all my examples to 115200 baud which works at all supported speeds, even 1 MHz on a modern AVR. 

Modern and classic have the same maximum baud rate from a cursory glance at features - F_CPU/8 - but only modern AVRs can actually use normal baud rates anywhere near there. Classic AVR USART needed to know what integer to divide it's clock by to get a clock running at 1/8th or 1/16th the UART baud rate Well, when you're close to the maximum, you can use either the maximum baud rate or half of that, but nothing in the middle. Modern AVR s instead need the number of 64ths of clock cycles  to divide the clock by, which means that even the pessimal case, ideal value 64.5, baud rate error is limited to under 1%, and in realistic cases (lower baud to clock ratio, and, with accuracy improving as baud rate decreases (as you would expect) such that the baud calculation error is negligible. The USART was one of the most-improved peripherals on the first modern AVRs.

elektroboard, 2023-07-24:
Frankly, I'm a little confused. Could you add an example for what I want to do concretely?

SpenceKonde, 2023-07-24:
I don't have one, I never have wanted to measure a frequency - usually I'm running in event capture mode and getting each length.

hmeijdam, 2023-07-25:
Here is an example of Input Capture Frequency Measurement.

```
/*
Example for TCB0 Input Capture Frequency Measurement.
It is using pin PA2 as input and prints the frequency to the serial monitor.
If measuring low frequencies (like 50Hz), you have to select a low clock speed (e.g. 4MHz) or the 16 bit timer will overflow within one period.
Hans Meijdam, July 2023
*/ 
volatile uint16_t timer_capture;

void setup () {
  //Serial.swap();// use this to set the TXD*/RXD* to their alternate pins
  Serial.begin(19200);// Default Serial TX is on pin 0 (PA6)
  PORTA.PIN2CTRL = PORT_PULLUPEN_bm ; // pullup on the input pin

  // configure event channel
#if defined(__AVR_TINY_2__)// The 2 series parts have a different event system configuration
  EVSYS.CHANNEL0 = EVSYS_CHANNEL0_PORTA_PIN2_gc; //link event system ch0 to pin A2
  EVSYS.USERTCB0CAPT = EVSYS_USER_CHANNEL0_gc;// Connect TCB0 capture to event channel 0
#else// This is for the 0 and 1 series parts
  EVSYS.ASYNCCH0 = EVSYS_ASYNCCH0_PORTA_PIN2_gc; //link event system async ch0 to pin A2
  EVSYS.ASYNCUSER0 = EVSYS_ASYNCUSER0_ASYNCCH0_gc;// 0x01 Link TimerB (=AYNCUSER0) to event ch0
#endif

  // configure timer B
  TCB0.CTRLB |= TCB_CNTMODE_FRQ_gc; // (0x04<<0), Input Capture Frequency Measurement
  //  on rising edge, will capture into CCMP and reset counter
  TCB0.EVCTRL |= TCB_FILTER_bm ; // activate Input Capture Noise Cancellation Filter
  TCB0.EVCTRL |= TCB_CAPTEI_bm ; // Capture Event Input Enable
//  TCB0.EVCTRL |= TCB_EDGE_bm ; // uncomment if the falling edge needs to be used. Default rising edge
  TCB0.INTCTRL |= TCB_CAPT_bm ; // Capture Interrupt Enable
  TCB0.CTRLA |= (1 << 0); // enable TCB0
}

void loop () {
  uint16_t period = timer_capture / (F_CPU / 1000000L);
  Serial.print("Timer capture: ");
  Serial.print(timer_capture);
  Serial.print("   period: ");
  Serial.print(period);
  Serial.print(" microseconds.  Frequency: ");
  Serial.print(1000000L/period);
  Serial.println(" Hz");
}
ISR (TCB0_INT_vect) {
  /*The CAPT Interrupt flag is automatically cleared 
  after the low byte of the Compare/Capture (TCBn.CCMP) register has been read.
  datasheet chapter 21.3.3.1.5  */ 
  timer_capture = TCB0_CCMP;
}
```

closed 2023-07-25 by elektroboard