SpenceKonde/megaTinyCore #1002: MILLIS_USE_TIMERA0 issue on series 1 but OK on series 2

LaurentR59, 2023-09-17:
Hi

Like ever explained in DX CORE I met same kind of issue using MILLIS_USE_TIMERA0 to compile on ATTINY SERIES 1 parts

#error "wiring.c and timers.h want to set millis timer TCA0 to different divider"

It s ok with SERIES 2 with same config files (just sap to the good one CPU)

Context under PaltformIO ( latest version of it) 

[env:ATTINYX7]

;//FOR ATTINY SERIE 1 & 2:  ALL SERIE 1 & 2 PARTS
platform_packages = framework-arduino-megaavr-megatinycore
                    platformio/toolchain-atmelavr@^3.70300.220127
                    platformio/tool-avrdude-megaavr@^3.60300.220118

; 24PINS ATTINY SERIE 1 FAMILY:
platform = atmelmegaavr
framework = arduino

board = ATtiny3217
board_build.variant = txy7 
board_build.mcu = ATTINY3217

build_unflags = -D MILLIS_USE_TIMERB1

; 24PINS ATTINY SERIE 2 FAMILY:
;board = ATtiny3227
;board_build.variant = txy7 
;board_build.mcu = ATTINY3227

;build_unflags = -D MILLIS_USE_TIMERB1

build_flags = -v -D MILLIS_USE_TIMERA0

Folder framework-arduino-megaavr-megatinycore have received the latest files from 2.6.8 version. (current files version)

Any idea abouit this strange behavior? Why OK on SERIE 2and not SERIES1?

Laurent

LaurentR59, 2023-09-17:
Hi

A workaround is to set MILLIS_USE_TIMERD0 for parts that got it in SERIES 1 (ex 32 16 & 3217)

keep MILLIS_USER_TIMERA0 OK on any other parts for SERIES 2

Laurent

SpenceKonde, 2023-09-17:
Can you please post code that reproduces this??? it does not reproduce for me. See alkso my comments in the DxCore issue too, with some more detailed debugging info

commit 9e59a824 (2023-09-24): fix #1002
closed 2023-09-24 by SpenceKonde