SpenceKonde/megaTinyCore #1012: epprom retained ignored

med1306, 2023-10-04:
Hello

txs for this core
The epprom is erased each time i flash my attiny3216
With power cycle the epprom was ok

SpenceKonde, 2023-10-04:
Did you do "Burn Bootloader" like the text in the menu says is required to apply the changes? 

No, it does not matter if you are using a bootloader. The programming procedure is to set any fuse except one that if set incorrectly, could prevent further programming from being possible with the same programmer you used to set the fuse (we can't do logic like "Set/clear bit 1 and 2 of this fuse, but leave the rest unchanged "- that requires the program doing the upload know what the value of that fuse is now. We don't have that feature on serialupdi, and it wouldn't change things if it could because we couldn't change that behavior for just serial UPDI Unfortunately, to my great dismay, they placed the EEPROM retain bit  on the same fusebyte that UPDI disable bits are located at. So we don't set those on upload because a single casual upload if you forgot to reset the tools submenus to match the board you connected could brick the board. Dangerous fuses are only set upon "burn bootloader" which initializes *all* fuses, even ones we don't change normally (these are not written on upload, only non-dangerous, fuses that we use (reasoning being if the non-dangerous, unused fuses are set,you probably did that manually and would be unhappy if we undid that on every upload)

SpenceKonde, 2023-10-04:
See exact same question asked by someone else last week https://github.com/SpenceKonde/megaTinyCore/discussions/1007

med1306, 2023-10-05:
Sory for the redondant question it work perfectly now.

Again great txs to all your works on megatiny / dxcore and the amazing serialupdi way for programming.
For your information i flash a board far away from computer (50m) with SerialUpdi over RS-422 lines with galvanical isolation.
the usb/uart was a FT234x
the digital isolator was ISO7242
the RS-422 driver was MAX14789
a 74LVC2G3157 was use for auto UART or UPDI mode

closed 2023-10-05 by SpenceKonde