SpenceKonde/megaTinyCore #1014: Possible error around ADCPowerOptions() in Arduino.h

WebGH, 2023-10-06:
Megatinycore version 2.6.8
Arduino IDE Version 1.8.13
uC Attiny1616

I tried to use _AdcPowerOptios()_ but always ran ito the error that none of the options described in the dicumentation worked, they all were declared undefined.
So, I looked up the origin and find that the possible options are defined in "_Arduino.h_" inside the megatinycore sources folder.

BUT: The option "_ADC_DISABLE_" which should be available for all supported uCs is inside an _#ifdef_ block that is only valid for 2-series parts.
In General, all "generic" options seem to be only available for 2-series parts which is NOT what the documentation says.
I you use the hex values instead of the defined names, like "_0x30_" to disable the ADC, it compiles without any issue.

To me it seems the "_#endif_" of the "_#if MEGATINYCORE_SERIES < 2_" block (starting at line 91 in Arduino.h) seems to be too far down.
To my limited understanding this "_#endif_" should be behind line 183 ?
Or the generic options need to be copied to the region between line 91 and 132

SpenceKonde, 2023-10-08:
All ADC-related functions should have one definition in the MEGATINYCORE_SERIES < 2 block, and one in the else block. There is *no* code shared between them, there should be two copies of the ADCPowerOptions function, let's see what is going on

SpenceKonde, 2023-10-08:
Ah it was the constant defines only that were missing. 

The 4 valid options are now given named defines, and the rest (which don't exist on 0/1) are #defined as `badArg("This option is on 2-series tiny and AVR Ex-series only")` to try to chase people in the right direction.

Pity this just missed the release by a day.

closed 2023-10-08 by SpenceKonde

WebGH, 2023-10-10:
Dear Spence,

Thank you very much for the confirmation. I  also had issues when using the HEX values direct, instead of named defines.
Anyhow, I need some more time to investigate this.

With best regards,
   Wolfgang

SpenceKonde, 2023-10-10:
Well, the fix to the constant names is in 2.6.10 which will be available in board manager soon. Hex values (as far as I know) work in 2.6.9 and earlier and both should work in 2.6.11, if that's not the case, I don't know that. .

commit 23b1c90e (2023-10-08): Fix #1014: ADCPowerOptionsconstants missing on 0/1

The 4 valid options are now given named defines, and the rest are #defined as `badArg("This option is on 2-series tiny and AVR Ex-series only")`