SpenceKonde/megaTinyCore #1018: megatinycore on Micropchip Xplained Pro ATtiny3217

Albrechtidis01, 2023-10-09:
Hi Dr.Azzy,

I want to work with the Microchip Xplained Pro 3217 under the Arduino IDE 1.8.19.
From megatinycore Version 2.6.8 I selected 'Official Microchip Board/Xplained Pro ATtiny3217'.
If I compile, the following error is displayed:

fatal error: pins_arduino.h: No such file or directory

I checked the boards.txt file, but I can see no problem there. I assume, an other board is internally detected.
But I don't know really, how the board detection works internally...  
Can you help with some information here?
Thanks in advance

Albrecht Frick
Munich, Germany

PS:
the full output is here:

Arduino: 1.8.19 (Windows 10), Board: "Official Microchip Board, Xplained Pro ATtiny3217, 20 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (no reset pin), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), On all pins, with new implementation., Disabled (recommended), No delay before window "opens""

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\albre\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\albre\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\albre\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:microchip:board_name=xp3217,clock=20internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -ide-version=10819 -build-path C:\Users\albre\AppData\Local\Temp\arduino_build_761045 -warnings=none -build-cache C:\Users\albre\AppData\Local\Temp\arduino_cache_856989 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\albre\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\albre\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\albre\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\albre\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avrdude.path=C:\Users\albre\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\albre\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.python3.path=C:\Users\albre\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\albre\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -verbose C:\Users\albre\OneDrive\homegroup\ArduinoWork\_ARDUINO_TEMPLATES\Xpl3217\Xpl3217.Serial.MyLibs.V00\Xpl3217.Serial.MyLibs.V00.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\albre\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\albre\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\albre\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:microchip:board_name=xp3217,clock=20internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -ide-version=10819 -build-path C:\Users\albre\AppData\Local\Temp\arduino_build_761045 -warnings=none -build-cache C:\Users\albre\AppData\Local\Temp\arduino_cache_856989 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\albre\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\albre\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\albre\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\albre\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avrdude.path=C:\Users\albre\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\albre\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.python3.path=C:\Users\albre\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\albre\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -verbose C:\Users\albre\OneDrive\homegroup\ArduinoWork\_ARDUINO_TEMPLATES\Xpl3217\Xpl3217.Serial.MyLibs.V00\Xpl3217.Serial.MyLibs.V00.ino

Using board 'microchip' from platform in folder: C:\Users\albre\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.8

Using core 'megatinycore' from platform in folder: C:\Users\albre\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.8

Detecting libraries used...

"C:\\Users\\albre\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny3217 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10819 -DARDUINO_AVR_Xplained_Pro_3217 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.8.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=8UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -DLED_BUILTIN=PIN_PB4 -DPIN_BUTTON_BUILTIN=PIN_PB5 "-IC:\\Users\\albre\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\albre\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\cores\\megatinycore" "-IC:\\Users\\albre\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\variants\\atxy7" "C:\\Users\\albre\\AppData\\Local\\Temp\\arduino_build_761045\\sketch\\Xpl3217.Serial.MyLibs.V00.ino.cpp" -o nul

Alternatives for pins_arduino.h: []

ResolveLibrary(pins_arduino.h)

  -> candidates: []

In file included from C:\Users\albre\AppData\Local\Temp\arduino_build_761045\sketch\Xpl3217.Serial.MyLibs.V00.ino.cpp:1:0:

C:\Users\albre\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.8\cores\megatinycore/Arduino.h:686:10: fatal error: pins_arduino.h: No such file or directory

 #include "pins_arduino.h"

          ^~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

Fehler beim Kompilieren für das Board Official Microchip Board.

SpenceKonde, 2023-10-10:
This doesn't reproduce for me, does it still happen in the latest release from last night? 

Does it require the sketch to have that library included? If so I need to know where the library comes from (name of arduino libraries is meaningless. There does not exist a unique mapping between library names and actual library code, because the same name, if the library is popular, will map to dozens of publically available repos - this is the dark side of github and open source. The good news is anyone can make a fork and whoever wants to can use it. The bad news is that anyone can make a fork, and nobody bothers to update readme/docs. 

Everytime someone forks a library, modifies it to work for them (in github, so publically) and then links to it from their blog, 9 times out of 10, the readme is unchanged, so someone coming along doesn't realize that it's a modified version. Then someone else tries to use your code, or you try to rebuild your libraries folder after a crash from which only your code in github escaped (happened to me,though I eventually paid like half a grand to commercial data recovery because it had over 100 hours of PCB work on it that I couldn't get back and had to rev the board - but in that interim, I had the code for a project, but it had to be totally rewritten, because apparently when I started with that library I had picked a whacko version with a weird API, and couldn't find it or anything like it).

Albrechtidis01, 2023-10-11:
Hi Dr. Azzy,

 

*	I updated to the last megatinycore version 2.6.10
*	There are no libraries contained in the test sketch (XplPro3217.Minimal.ino see .zip)
*	In the Arduino IDE is displayed a message "Install package 'SAMD Boards (32bit ARM Cortex-M0)' to use your Board 'Arduino M0 Pro'. In my opinion this is a misleading info, I Ignored the message, did not install the package.

       Maybe it indicates some board conflict.

*	I found a workaround to work with the Xplained Pro ATtny3217: If I use the normal megatinycore entry Attiny3227/3217... with Chip 3217, then all works. 
*	Additional diag stuff I stored inside the megatinycore.diag.zip archive.
*	 

Best regards

Albrecht

SpenceKonde, 2023-10-11:
Did you send that file as an email attachment? Github only takes files you attach to comments made through the web interface,

Albrechtidis01, 2023-10-11:
sorry, now the 'attached' files...
[arduino.Attiny3217.ok.output.txt](https://github.com/SpenceKonde/megaTinyCore/files/12874988/arduino.Attiny3217.ok.output.txt)
[arduino.official.microboard.error.output.txt](https://github.com/SpenceKonde/megaTinyCore/files/12874989/arduino.official.microboard.error.output.txt)
[image]

Albrechtidis01, 2023-10-11:
and
[XplPro3217.Minimal.ino,txt](https://github.com/SpenceKonde/megaTinyCore/files/12875001/XplPro3217.Minimal.ino.txt)

SpenceKonde, 2023-10-12:
boards.txt had a typo in it that nobody had noticed (several actually), all relating to the microchip boards. I dont think they get much exercise.

closed 2023-10-12 by SpenceKonde

Albrechtidis01, 2023-10-12:
Thanks, fix works!

commit 3e3f8204 (2023-10-12): Fix #1018

Fix #1018

There were a number of gross problemsw wotj tje ,ocrpcjo[p bpards = 2 bpards had this issue, another also had a missing parameter, and none of the olptiboot ones actually enabled use of a bootloader when ou did burn bootloader (only the microchip boards, the generic ones were okay)