SpenceKonde/megaTinyCore #1021: ATtiny817 Xplained Mini pinouts

pwallner, 2023-10-17:
Hi,

maybe I misunderstand something, but the pinouts are not the same as on your picture [here](https://github.com/SpenceKonde/megaTinyCore/raw/master/megaavr/extras/ATtiny_x17.gif) and in the manual [here](https://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny817-Xplained-Mini-UserGuide-DS50002657B.pdf).

The user led is on PC0, which should be Pin17. But the LED blink sample is only working with Pin12.

The table in the manual shows:
| VQFN 24-Pin | VQFN 20-Pin | SOIC 20-Pin | Pin Name |
| ----------- | ----------- | ----------- |  ----------- |
| 17 | 15 | 12 | PC0 |

The 817 is a 24-Pin, but pin12 is for the 20-Pin chip?!

Here is my small test code:
```
void setup() {
  pinMode(12, OUTPUT);
}

void loop() {
  digitalWrite(12, HIGH);
  delay(1000);
  digitalWrite(12, LOW);
  delay(1000);
}
```

Thank you for your help.

pcfreak1201, 2023-10-18:
Year, that's why I don't like the Arduino-Pin-No.
It has nothing to do with the chip-pinout. Simply use PIN_PC0 instead of "12"

pwallner, 2023-10-18:
Thx for the hint! I think we can close the issue because I think it's more or less a cosmetical thing.

SpenceKonde, 2023-10-18:
Arduino pins are not physical pins. Arduino numbers are arbitrary core-defined numbers, which I happen to set systematically in order, counterclockwise from the power pins. (Hey, at least they're numbered in order! some classic cores... it's like they took pin number confetti, with numbers 0 to n-1, and shook it out over the pin configuration chartfor a part with n pins, and then counted how many of each number were in the box on the diagram, That seems like more work that just, you know, numbering them in the same order as port pin numbers, or numbering them in counterclockwidr or clovkwise direction.

The arduino pins are shown on the part specific documentation https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ATtiny_x17.md 

You can clearly see that the arduino pin number of PC0 is 12. 

But - why use the bare numbers? It causes exactly this sort of problem! We provide defines of the form PIN_PXn where X is a port letter and n is the bit number (Pxn #defines were taken). On mTC and DxC, regardless of how convoluted the pin numbering gets (Here, "not very" but on DxCore, there are things with pins 0, 1, 9, 10, 11, 16, 17, 18, 19, 20, 21 in my numbering, to preserve the correllation between port and bit and pin number, which happens to be incredibly convenient,

closed 2023-10-18 by SpenceKonde

pwallner, 2023-10-19:
@SpenceKonde thank you for your explanation! I am a beginner in attiny and was also confused by this numbering...

Btw, thank you for megatinycore, this is a great help for coding!!