SpenceKonde/megaTinyCore #1024: I2C master sends NACK

krasimirk, 2023-10-21:
Hello,
I have the following setup : Arduino ver. 2.2.1 / megaTinyCore ver. 2.68. ATtiny402 on breadboard without any other IC, 2 pull-up resistors with value 1K (I do not have higher value) to pin 4 SDA and pin 5 SCK. Running the following code, the ATtiny402 always sends NACK (logical 1) as 9th bit, instead to release the bus and to wait ACK/NACK from slave device:

#include <Wire.h>
void setup() {

  Wire.begin();
  Wire.beginTransmission(0x50);
  Wire.write(0);
  Wire.write(0);
  Wire.endTransmission();
}

You can see screenshot from scope. It shows the first transmission of the I2C packet and after that there is nothing.
I'm trying to connect external i2c eeprom 24C02 to the ATtiny402, but the 9th bit is always NACK (logical 1), when the device select command is send. Initially I though that the eeprom answer with NACK which is possible as according eeprom/i2c spec, but after removing the eeprom from the breadboard, seems the MCU sends the NACK.

[image]
[image]

Any idea what can be wrong?
Thanks in advance!
Kind Regards,
Krasimir Kostadinov

MX682X, 2023-10-21:
> without any other IC

The library terminates the transmission, if there is a NACK. As there is no other device connected, nothing will pull SCL low and there will be always a NACk

MX682X, 2023-10-21:
also, try the following, just to be sure
```
#include <Wire.h>
void setup() {

Wire.begin();
for (uint8_t i = 0; i < 8; i++) {

  Wire.beginTransmission(0x50+i);
  Wire.write(0); 
  Wire.write(0);
  if (Wire.endTransmission() == 0) {
    break;
  }
}
```
}

krasimirk, 2023-10-22:
Hello,
Here is picture from scope when EEPROM is connected to attiny and with your suggested code.
The address pins are connected to gnd.
The first picture from the scope is from addresses 0x50-0x55 and second one is from 0x56-0x57.  
[image]
[image]
Here is the breadboard:
[image]

I double check the i2c connection : pin 5 (SDA) from attiny is connected to pin5 (SDA) of the EEPROM, pin 4 (SCK) from attiny is connected to pin6 (SCK) of the EEPROM. I have another 24C02 and one 24LC512 and results are the same.
Kind Regards,
Krasimir Kostadinov

SpenceKonde, 2023-10-22:
1. You are missing the two required 0.1uF capacitiors, There should be one between power and ground of both the EEPROM and the AVR. Without decoupling capacitors, parts behave unpredictably. 
2. From the scope trace, it looks to me like there is no response whatsoever from the slave, implying a wiring problem. I would guess that 1 or both I2C lines aren't making an electrical connection.

krasimirk, 2023-10-22:
Thanks for the suggestions!
I added 1uF across Vdd/Gnd and checked pin to pin connection - result is the same as before.   
[image]

krasimirk, 2023-10-22:
Hello,
Finally I figured out what is going on.
So initially I made connection according Attiny documentation on page 14, Table 5-1 (PORT Function Multiplexing SOIC 8-pin). According this table :
PA1 pin5 SDA  I connected to the pin 5 of the EEPROM
PA2 pin4 SCL I connected to the pin 6 of the EEPROM
When I swap them 
PA2 pin4 which is SDA according my scope (but SCL according datasheet) is connected to the pin 5 of the EEPROM which is SDA
PA1 pin5 which is SCL according my scope (but SDA according datasheet) is connected to the pin 6 of the EEPROM which is SCL

[image]

everything is working correctly:

[image]
I do not have any explanation about this.
Thank you for your time, spent to this issue!
Kind Regards,
Krasimir Kostadinov

krasimirk, 2023-10-22:
Hi 
I found the error in the datasheet.
According the data in the table Table 5-1 mention in the previous comment, the PA1 is pin 5 and PA2 is pin 4. But in the pinout in the previous page, it is opposite:
[image]

krasimirk, 2023-10-23:
Hello,
I created a ticket in the Microchip site for the error in the datasheet.
This issue can be closed.
Kind Regards,
Krasimir Kostadinov

SpenceKonde, 2023-10-24:
Oh god what version of the datasheet are you using? 

I'm going to guess it's DS40001911A, which has TONS of mistakes in it because it was among the first modern AVRs. 
That document was superseded by Obsolete Publication DS40001911B in 07/2019. DS4001911 was one of the old tinyAVR 0/1-series datasheets, from before they had realized that these parts have so much errata and each die gets used on so many parts that the only hope for comprehension is and communication is to group the parts into datasheet+errata docs that correspond directly to which parts share a die. Then they realized that, and so that document was was superseded by DS40002287A in 12/2020. The initial version of the datasheet was so slapdash, it wasn't until the 7/19 revision that they **added the word preliminary to it**. That's how preliminary it was. Table 5-1 was the worst off.

I deduce that datasheet version since the farthest back I have is 40002287A, which refers to the prior revision as "Obsolete Publication DS40001911B - 07/2019", and one thing (of dozens) listed was correcting the I/O multiplexing table.

closed 2023-10-24 by SpenceKonde