SpenceKonde/megaTinyCore #1029: Serial baud rate setting differs from actual measurement on ATtiny1604.

7-rate, 2023-10-31:
On ATtiny1604, I set the baud rate to 300 like `Serial.begin(300);`, but I could not receive properly unless I set the value on the serial monitor (TearTerm) side to 900.

hmeijdam, 2023-10-31:
In a sketch with 
```
int main(void) {
  while (1)  {
  }
}
```
maybe or are you using Setup and Loop ?

SpenceKonde, 2023-11-01:
Per the documentation both here and in the datasheet, 300 baud cannot be generated by the hardware. 

https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/Ref_Serial.md#minimums

>300 baud will work only when F_CPU is no higher than 5 MHz.
600 baud will work only when F_CPU is no higher than 10 MHz.
1200 baud will work as long as F_CPU is no higher than 20 MHz.

closed 2023-11-01 by SpenceKonde