SpenceKonde/megaTinyCore #1030: Xplained Mini ATtiny817, unable to upload due to baudrate -b{upload.speed}

hmeijdam, 2023-10-31:
IDE: 1.8.13
mTC: Github version
Board: Xplained Mini  ATtiny817
Windows 11 64bit

I think I saw a similar issue [here](https://github.com/SpenceKonde/megaTinyCore/issues/1029) for the Attiny1604

When I want to upload something to the Xplained via the Atmega32U4 the below command line gets generated for Avrdude and fails because  ` -b{upload.speed} `  does not get parsed in a real baudrate.

```
E:\Arduino\arduino-1.8.13-megatinycore\hardware\tools\avr/bin/avrdude -CE:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr/avrdude.conf -v -pattiny817 -cxplainedmini_updi -Pusb -PCOM1 -b{upload.speed} -Uflash:w:C:\Users\Hans\AppData\Local\Temp\arduino_build_402701/Blink.ino.hex:i 
avrdude: invalid baud rate specified '{upload.speed}'
An error occurred while uploading the sketch

```

When I change it to 115200 as per the other issue, the error goes away. Unfortunately only to be replaced by another error, but that may be caused by me as I first changed the USB drivers with Zadig in my attempts to make the first error go away.

so this:
`E:\Arduino\arduino-1.8.13-megatinycore\hardware\tools\avr/bin/avrdude -CE:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr/avrdude.conf -v -pattiny817 -cxplainedmini_updi -Pusb -PCOM6 -b115200 -Uflash:w:C:\Users\Hans\AppData\Local\Temp\arduino_build_360913/Blink.ino.hex:i 
`

then results into this error
```

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr/avrdude.conf"

         Using Port                    : COM6
         Using Programmer              : xplainedmini_updi
         Overriding Baud Rate          : 115200
avrdude: jtag3_open_common(): JTAGICE3/EDBG port names must start with "usb"

avrdude done.  Thank you.
```

SpenceKonde, 2023-11-01:
I thought this was fixed in the most recent release...

hmeijdam, 2023-11-02:
Well, you are probably right, because now the parsing of the upload speed works. And I have no clue how that is possible, I tested now in the 2.6.10  boards manager release and after that in the Github version (both portable IDE installations) without pulling in any changes via Github Desktop. 

Both parse ok now and get a step further in the upload process.

Not that I am out of the wood yet.....my workaround no longer works

Identical error from 2.6.10 and Github version.
```
E:\Arduino\arduino-1.8.19\portable\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CE:\Arduino\arduino-1.8.19\portable\packages\megaTinyCore\hardware\megaavr\2.6.10/avrdude.conf -v -pattiny817 -carduino -D -PCOM9 -PCOM9 -b115200 -Uflash:w:C:\Users\Hans\AppData\Local\Temp\arduino_build_565215/sketch_nov02a.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "E:\Arduino\arduino-1.8.19\portable\packages\megaTinyCore\hardware\megaavr\2.6.10/avrdude.conf"

         Using Port                    : COM9
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x1d
avrdude: stk500_recv(): programmer is not responding
```

The only difference I see now between my Github version (IDE 1.8.13) and my 2.6.10 (IDE 1.8.19) version is, that in the latter my board is recognized as Uno WiFi rev2
[image]

And in the (IDE 1.8.13) Github version as "Atmel Atmega328p Xplained Mini"
[image]

to be continued (back to replacing USB drivers with Zadig)

SpenceKonde, 2023-12-05:
And you aresure it's not a 328p? 

I don't have much to suggest - though I don't know why tyou are specifying -P for the port twice, it should only be specified once, in this case USB.

hmeijdam, 2023-12-05:
Mystery about the Xplained Mini showing up as Atmega328 in my Github version of MTC I think I have solved.

[image]

While I selct the MTC board I also have another boards package for the legacy Xplained mini's (328P). 

After I removed that boards package via the boards manager. I only see "COM3" in my serial ports list. No more name behind it, so I think that's fine for me.

SpenceKonde, 2023-12-19:
Could you link me to that package? I would love to see how they made the xplained mini present as such in the menus. 

The problem is that -b{upload.speed} is being generated at all. It shouldn't be in your working command either, it's just getting ignored because it's irrelevant to non-serial ports. I think it's using the upload via a bootloader recipe to create that command, rather than the no bootloader please upload using a programmer recipe like it's supposed to be using... and of course there is no upload.speed if it's not uploading over a serial port; the programmer devices from Microchip are composite USB devices that present as a serial port and a programmer. The programmer is what it should be using for non-bootloader board selections.

hmeijdam, 2023-12-19:
The Atmel 328p Xplained package was this one. Present in the default board manager

[image]

Github [link ](https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/wiki)to their webpage

As soon as install it, I get the name of the 328P Xplained mini behind my COM3.

It must be the name in their boards.txt file, because if I change that name I get this.
[image]

In their boards.txt  changed this
`atmega328p_xplained_mini.name=Atmel atmega328p Xplained mini`
into 
`atmega328p_xplained_mini.name=Why the fuck is this here???`

I would be careful with copying their idea, as it is flawed in my case. The name from a different core's boards.txt get's activated, while I am selecting a board from another core.

Maybe it has to do that I am working with a:

- Portable IDE 1.8.13 installation
- Github installation of MegaTinyCore into that portable IDE
- Boards manager installation of the French Xplained Mini package

hmeijdam, 2023-12-19:
Because you said "It should be using the programmer of the composite USB device" 

I tried "upload using programmer"

It fails with a different JTAG error after connecting successfully and after that I have to power-cycle or It fails to connect to the programmer completely.

```
Arduino: 1.8.13 (Windows 10), Board: "Official Microchip Board, Xplained Mini ATtiny817, 20 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (no reset pin), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), On all pins, with new implementation., Disabled (recommended), No delay before window "opens""

E:\Arduino\arduino-1.8.13-megatinycore\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino\arduino-1.8.13-megatinycore\hardware -hardware E:\Arduino\arduino-1.8.13-megatinycore\portable\packages -hardware E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-megatinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-megatinycore\hardware\tools\avr -tools E:\Arduino\arduino-1.8.13-megatinycore\portable\packages -built-in-libraries E:\Arduino\arduino-1.8.13-megatinycore\libraries -libraries E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\libraries -fqbn=megaTinyCore:megaavr:microchip:board_name=xm817,clock=20internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=03EB_2145 -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_995342 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_193981 -prefs=build.warn_data_percentage=75 -verbose E:\Arduino\arduino-1.8.13-megatinycore\examples\01.Basics\Blink\Blink.ino

E:\Arduino\arduino-1.8.13-megatinycore\arduino-builder -compile -logger=machine -hardware E:\Arduino\arduino-1.8.13-megatinycore\hardware -hardware E:\Arduino\arduino-1.8.13-megatinycore\portable\packages -hardware E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-megatinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-megatinycore\hardware\tools\avr -tools E:\Arduino\arduino-1.8.13-megatinycore\portable\packages -built-in-libraries E:\Arduino\arduino-1.8.13-megatinycore\libraries -libraries E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\libraries -fqbn=megaTinyCore:megaavr:microchip:board_name=xm817,clock=20internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=03EB_2145 -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_995342 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_193981 -prefs=build.warn_data_percentage=75 -verbose E:\Arduino\arduino-1.8.13-megatinycore\examples\01.Basics\Blink\Blink.ino

Using board 'microchip' from platform in folder: E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr

Using core 'megatinycore' from platform in folder: E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr

Detecting libraries used...

"E:\\Arduino\\arduino-1.8.13-megatinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny817 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10813 -DARDUINO_AVR_XplainedMini817 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.10\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=10UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -DLED_BUILTIN=PIN_PC0 -DPIN_BUTTON_BUILTIN=PIN_PC5 "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\variants\\txy7" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_995342\\sketch\\Blink.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...

"E:\\Arduino\\arduino-1.8.13-megatinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny817 -DF_CPU=20000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10813 -DARDUINO_AVR_XplainedMini817 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.10\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=10UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 -DLED_BUILTIN=PIN_PC0 -DPIN_BUTTON_BUILTIN=PIN_PC5 "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IE:\\Arduino\\arduino-1.8.13-megatinycore\\portable\\sketchbook\\har [... cut ...]
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "E:\Arduino\arduino-1.8.13-megatinycore\portable\sketchbook\hardware\megaTinyCore\megaavr/avrdude.conf"

         Using Port                    : usb

         Using Programmer              : xplainedmini_updi

avrdude: Found CMSIS-DAP compliant device, using EDBG protocol

avrdude: jtag3_edbg_recv(): Inconsistent fragment number; expect 1, got 0

avrdude done.  Thank you.

An error occurred while uploading the sketch

```

SpenceKonde, 2023-12-20:
inconsistent fragment number error - that is either the same error, or the reverse of the error (the 0 and 1 swapped) that you would get trying to upload to a Dx via a pre-Dx *EDBG. I will investigate what is up with the boards.txt for those boards. I would suggest making sure there wasn't a required firmware update for the on-board debugger. That would be consistent with the Microchip MO (which at times has appeared to be "move slow and break stuff". I have two snap programmers that I wasted three days trying to make work with microchip studio before giving up. )

hmeijdam, 2023-12-20:
For what it's worth, I looked in Microchip Studio what firmware version is on the programmer of the Xplained Mini

[image]

SpenceKonde, 2023-12-21:
I'm sure that would mean something to a Microchip Support employee. I suggest you ask one of them. 

You got WAY farther with microchiop studio and an official programmer than I ever did. I was never able to make microchip studio detect that I had a programmer connected at all with days of work. You've managed to detect it and read out the firmware version!

I'm a certified incompetent at making Microchip's programming tools work. Like I said, I could *never* make *any* of the programming tools from microchip except curiosity nanos work. After the snap debacle though, it wa hard to motivate myself to work. Why waste more time farting around to get a slower programmer to work on an IDE I detest? Unless you're a dark sorcerer who can do hardware debugging. And if you can do that, wtf are you asking me questions for? I should be asking you questions in that case.

hmeijdam, 2023-12-21:
Well, this was more like a monkey pressing some buttons and reading what I saw on the screen. 
I can load the examples that come with Microchip Studio for my Xplained Mini, but  before making my first program from scratch with that tool, I need to become R2D2 first. 
I always program from within the Arduino IDE, with everything in one page. I don't even use multiple tabs.

SpenceKonde, 2024-12-18:
Well, at least when you pressed buttons on the IDE, you got a banana.... I couldn't even pull off that. (IE, as a monkey, you're outperforming me)

USB to me is a dark art. See that's why I'm starting on mTC instead of dXC ;-)

SpenceKonde, 2025-05-25:
Belive this to be a manifestation of long known IDE programmer selection issue when the old programmer is not correctly deselected

closed 2025-05-25 by SpenceKonde