SpenceKonde/megaTinyCore #1032: warning: "DEFAULT" redefined

pwallner, 2023-11-16:
I try to compile for a attiny817 (with [SparkFun_STUSB4500)](https://github.com/sparkfun/SparkFun_STUSB4500_Arduino_Library/issues/8) and get this warning, it disappears when I change DEFAULT to DEFAULT1 or something else (in file `stusb4500_register_map.h`), but don't know what will happen (I ordered the board "SparkFun Power Delivery Board - USB-C (Qwiic)", but will receive it in a couple of days, so I want to know it in advance):
```
In file included from sketch\SparkFun_STUSB4500.h:31:0,
                 from D:\pcb\SparkFun Power Delivery Board\Example1-ReadParameters\Example1-ReadParameters.ino:23:
sketch\stusb4500_register_map.h:1:0: warning: "DEFAULT" redefined
 #define DEFAULT                0xFF
 
In file included from sketch\Example1-ReadParameters.ino.cpp:1:0:
C:\Users\xxx\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\cores\megatinycore/Arduino.h:117:0: note: this is the location of the previous definition
   #define DEFAULT         ADC_REFSEL_VDDREF_gc
 
In file included from sketch\SparkFun_STUSB4500.h:31:0,
                 from sketch\SparkFun_STUSB4500.cpp:21:
sketch\stusb4500_register_map.h:1:0: warning: "DEFAULT" redefined
 #define DEFAULT                0xFF
 
In file included from sketch\SparkFun_STUSB4500.h:25:0,
                 from sketch\SparkFun_STUSB4500.cpp:21:
C:\Users\xxx\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10\cores\megatinycore/Arduino.h:117:0: note: this is the location of the previous definition
   #define DEFAULT         ADC_REFSEL_VDDREF_gc
```

SpenceKonde, 2023-12-05:
You should raise this issue with the author of the library, he is the one who has written a "arduino" library which tries to reuse a name used by the stock core and portions of the API thhat would break tons of code if changed. 
u
We can't change that "DEFAULT" define without breaking compatibility with countless existing arduino sketches, because DEFAULT is part of the API (namely, it's the define that you pass to analogReference() to set to the default I belive). If I had designed Arduino I would have prefixed or postfixed that with someting. But Arduino didn't do that, and I can't come up with any explanation that doesn't reflect poorly on them. So there are stupid named defines like DEFAULT and BUFFER_SIZE (default what? Well ADC reference.Size of what buffer? The I2C buffer of course). I routinely get requests to change these stupid names, but the fact is that doing so breaks compatibility. Library authors have a responsibility to not use macro names used in stock arduino!

closed 2023-12-05 by SpenceKonde

pwallner, 2023-12-05:
@SpenceKonde fully agree with you! Will contact the supplier if he is willing to correct. I wrote my own code which is doing more or less the same but without conflicts.
Thx for answering!