SpenceKonde/megaTinyCore #1033: Attiny3227 UART "TX:18 (PA1), RX:19 (PA2) (UART0 alt pins)" confusion.

ziggy90127, 2023-11-18:
Thank you for megaTinyCore!

For Attiny3227 with optiboot over serial, the screen shot shows "TX:18 (PA1), RX:19 (PA2) (UART0 alt pins)".  When I hookup to PA1 (pin 24) and PA2 (pin 1), it works.  What do the 18 and 19 numbers signify?

[image]

hmeijdam, 2023-11-19:
18 and 19 are Arduino pin numbers.
Did you look at the [pin diagram](https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ATtiny_x27.md)?

ziggy90127, 2023-11-19:
On the ATtiny_x27 diagram pins 18 and 19 are PC1 and PC2.  On the Atmega328p, pins 18 and 19 are AVCC and ADC6.  It's not important, I've got it working, I just thought it was a small numbering error.

SpenceKonde, 2023-11-19:
Copypaste error from the 20 pin parts. You're the first to notice, it's
been like that for years.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2023-12-05:
All of the pinouts I use, I will note, try to avoid the countless errors made in the classic AVR parts where they would have the same "number" pin have the same function. 

Of course, this is hopeless, because pins on different parts have different selections of special functions. This led to some of the most hellish pinouts imaginable in a few cases, and when mTC and DxC were being created I articulated a precise numbering scheme for all I/O pins. Note that we strongly discourage use of raw numbers to refer to pins. The PIN_Pxn constants provided by the core are what should be used, 

Furthermore, you seem to be confused about the difference between physical and logical pins. Unless specifically stated that physical pin numbers are being used, all pin numbers are logical pin numbers (power pins are skipped, and no matter what, UPDI (on modern AVR) or reset (on classic) is the last often inaccessible pin. This is because depending on the package the part is purchased in, even the same part, the physical pins can have different numbers from each other, while the PIN_Pxn constants always refer to that pin, and on all parts of the same generation, that pin will have the same special functions, allowing code to, in some cases, be ported to a different part by just selecting the desired board from the menu. I consider it to be one of Arduino's biggest mistakes to come up with things like "Analog Pins" (ya got 2 things called analog, and this only refers to one of them, how is this supposed to make it easier doe people to learn programming (which is Arduino's thing, they are HUGE in education.

Closing this as there does not appear to be a bug here.

closed 2023-12-05 by SpenceKonde

ziggy90127, 2023-12-05:
Thank you, sorry to waste your time on this.  My previous experience with attiny was before the existence of Arduino.  I thought that I had read somewhere that Arduino pin numbering started at the pin after VDD and went counter clockwise.  Using that logic I get this:

|attiny322x|PIN_Pxx value|physical pin|counter clockwise distance|
|---|---|---|---|
|PIN_PB2|9|14|5|
|PIN_PB3|8|13|5|
|PIN_PA1|18|24|6|
|PIN_PA2|19|1|6|
|PIN_PC2|14|19|5|
|PIN_PC1|13|18|5|

So at a distance of 6, I thought there may have been a problem since the others made sense at 5.  But yes, using PIN_Pxx is the way to go.  I just hand soldered my attiny3227 to a custom board, the only thing Arduino about it is the reset circuit, and of course the software.  In my case Arduino pin numbers can be ignored.