SpenceKonde/megaTinyCore #1044: Unexpected behaviour with timers

jonathanmlang, 2023-12-31:
``//pins

#define P_PUMP      PIN_PA1
#define P_CIRC      PIN_PA4
#define P_GLOW      PIN_PC1
#define P_TURB      PIN_PC2
#define P_RELAY     PIN_PC3
#define SER_TX      PIN_PB2
#define SER_RX      PIN_PB3
#define T_WTHERMA   PIN_PA7
#define T_WTHERMB   PIN_PB5
#define V_PUMP      PIN_PB4
#define V_CIRC      PIN_PA4
#define V_TURB      PIN_PA2
#define V_GLOW      PIN_PB0
#define H_SPD       PIN_PA6
#define V_SUP       PIN_PA3

//config variables
byte plugheat_t=30;   //time in seconds before glowplug is considered hot enough
byte fanramp=25;      //rpm per second up and down
byte pumpramp=5;      //pumprate*0.01 hz per second
byte littemp=100;     //temp in c for deeming burner lit
byte plugheat_t_end=30; //time to burn out chamber on poweroff
byte turnoff_t=50;    //temp in c to turn off heater on cooldown

int pump_rawpulse_off=1000;
int pump_rawpulse_on=5;

volatile int pump_rawpulse_incre=0;

//volatile bool onoff;
volatile byte time_correction=0;

void setup(){

  delay(2000);

  //outputtest
  pinMode(P_PUMP,1);
  pinMode(P_CIRC,1);
  pinMode(P_GLOW,1);
  pinMode(P_TURB,1);

  
  
  TCA0.SINGLE.CMP0 = 78; // Set compare threshold
  TCA0.SINGLE.CTRLA = TCA_SINGLE_ENABLE_bm | TCA_SINGLE_CLKSEL_DIV256_gc; // Enable timer with 1/256 prescaler
  TCA0.SINGLE.INTCTRL = TCA_SINGLE_CMP0_bm; // Enable compare

  
}

void loop(){

  
}

ISR(TCA0_CMP0_vect)
{
  /* Insert your TCA Compare 0 Interrupt handling code here */

 if (pump_rawpulse_incre > pump_rawpulse_off) 
    pump_rawpulse_incre = 0;
else 
    pump_rawpulse_incre++;

volatile bool onoff = (pump_rawpulse_incre <= pump_rawpulse_on) ? 1 : 0;

volatile bool time_leap;

if (time_correction >= 7) { //78.125 pulses in 10ms so leap pulse every 8
    time_correction = 0; 
    time_leap = 1;
} else {
    time_correction++; 
    time_leap = 0;
}

TCA0.SINGLE.CMP0 = TCA0.SINGLE.CMP0 - (time_leap ? 79 : 78);

digitalWriteFast(P_PUMP, onoff);

/* The interrupt flag has to be cleared manually */
TCA0.SINGLE.INTFLAGS = TCA_SINGLE_CMP0_bm;
}

``

In this code if I set ``TCA0.SINGLE.CMP0 = TCA0.SINGLE.CMP0 - 78`` the code works as expected. The led connected to P_PUMP flashes correctly.

TCA0.SINGLE.CMP0 = TCA0.SINGLE.CMP0 - (time_leap ? 79 : 78); flashed the led once and never again.
I have tried other ways of writing this such as

`TCA0.SINGLE.CMP0 -= 78 + 1; which also doesnt work but removing the +1 makes it work again.

Same applies for even doing it like this!
`if (time_leap) TCA0.SINGLE.CMP0 = TCA0.SINGLE.CMP0 - 79;
 else           TCA0.SINGLE.CMP0 = TCA0.SINGLE.CMP0 - 78;`
 
 Does anyone know whats going on here?
 
 Using attiny3226, arduino ide 1.8.13 and megatinycore 2.6.10

SpenceKonde, 2024-01-01:
Call takeOverTCA0() as documented, before writing to any registers in TCA directly if you are going to configure the TCA with a custom configuration - this does two things. 
First, it tells the core machinery that the timer doesn't exist for pwm purposes, preventing analog and digital write functions from scribbling over your configuration (the main issue is the 
Secondly, it reinitializes the timer tothe POR state; we significantly reconfigure the timer, mainly by putting it into SPLIT mode in order to get 6 8-bit PWM channels (since the API quite clearly states that it only accepts 8-bit values, we might as well use split mode to double the number of PWM channels right?)
Since unsetting that option is confusing to many users (and because of the other issue) we provide takeOverTCA0 (and takeOverTCD0 and takeOverTCA1 on parts that have those - though as there isn't a clean way to hard reset TCD, that unfortunately requires unpleasant manual reconfiguration - though a lot can be done to TCD0 (as documented in the core documentation - we have a whole document on TCD0 and what you can change without a takeover, since it's so unpleasant to work with). None of these functions can be used if that timer is the millis timer).

jonathanmlang, 2024-01-02:
Interesting, thanks for that. There is a lot of documents and I seem to keep missing many of them unintentionally.
Anyway in this program I need pwm on PA4 and PA5 only, and I also need millis() and micros(). Do you think I will hit any snags?

SpenceKonde, 2024-02-09:
On a 3226? Nope nothing jumps out at me. Those are PWM pins in the configuration we provide by default (it can be changed with the tools submenu, if you need PWM to be on different pins). The default millis timer on the 3226 is TCB1; on megaTinyCore (unlike DxCore), the core does not use the scarce, incredibly useful and entirely too rare on tinyAVRs TCBs in "generate one measly 8-bit crappy PWM channel" mode; We have 6 outputs from the TCA, plus 1-series often has 2 more from the TCD. And TCBs are the timer everyone wants to take over because they are so simple.

SpenceKonde, 2024-02-09:
Closing because at present there is no evidence here of any defect within the core.

closed 2024-02-09 by SpenceKonde